open-power/op-build #47: Use separate partitions for skiboot and skiroot

shenki, 2015-02-01:
I'd like to propose switching to the flash layout on habanero and palmetto to use separate partitions for skiboot and skiboot (aka the petitboot Linux + rootfs image).

This makes it easier for development, as it allows flashing of skiboot independently. In theory it will also speed booting, as we can load skiroot while skiboot is performing other hardware initialisation, although I have not implemented this yet.

The current skiboot build is 768KB on my machine, so I've given skiboot 1MB and the rest (15MB on palmetto, 21.5MB on habanero) is left for skiroot. The new partition is called `SKIROOT`.

The changes to do this are in the `skiroot` branches under my account:

 * skiboot https://github.com/shenki/skiboot/commit/9bac6301cd5c0a8c01874c23de0b2fc9f60b04b1
 * op-build https://github.com/shenki/op-build/commit/ea6eec719f037bb34ab170dd5beff66905d801cc
 * pnor https://github.com/shenki/pnor/commit/a27f4bbeb435603c1f2ee7722ca2ddafbd962878

I've done test builds for habanero and palmetto, and successfully booted it on a plametto. You can reproduce my build check out my branch of `op-build` and run the following:

```
op-build P8_PORE_BINUTILS_VERSION=dcf4f87c0d9e8c8092ace0eb696189ee1056da30 \
 SKIBOOT_SITE=https://github.com/shenki/skiboot \
 SKIBOOT_SITE_METHOD=git \
 OPENPOWER_PNOR_SITE=https://github.com/shenki/pnor \
 OPENPOWER_PNOR_SITE_METHOD=git \
 OPENPOWER_PNOR_VERSION=49c2ccc6aaf36bb1eccd512391612136baa1a629 \ 
 SKIBOOT_VERSION=9701378341152718600048a13f829f3fc2224614
```

Please review my patches, and if we're happy with these changes I'll create PRs for pnor and submit the patch to skiboot, and then create a PR for op-build change along with rolling the dependences.

jk-ozlabs, 2015-02-02:
Yep, this is definitely something I'd like to get done - we've already come up with a `pnor_load_resource` that mikey will be using for the CAPP µcode bits. Might be good to sync with him on changes to the resource loading bits.

Could we include the initramfs as a separate partiton too?

stewart-ibm, 2015-02-02:
I'm slightly concerned about only 15MB for palmetto... we're right on that boundary now. My kernel+initramfs build is 15116262 bytes. Can we have 20MB on Palmetto?

The initramfs on separate partition probably does make sense too. Maybe we need to investigate a bit further as to what the size should be with modules.

williamspatrick, 2015-02-02:
‎Keep in mind that 1mb only yields about 910k after ECC is applied. Is that really enough?                                                                                                                                                                                                                                          From: Joel StanleySent: Sunday, February 1, 2015 4:13 PMTo: open-power/op-buildReply To: open-power/op-buildSubject: [op-build] Use separate partitions for skiboot and skiroot (#47)I'd like to propose switching to the flash layout on habanero and palmetto to use separate partitions for skiboot and skiboot (aka the petitboot Linux + rootfs image).

This makes it easier for development, as it allows flashing of skiboot independently. In theory it will also speed booting, as we can load skiroot while skiboot is performing other hardware initialisation, although I have not implemented this yet.

The current skiboot build is 768KB on my machine, so I've given skiboot 1MB and the rest (15MB on palmetto, 21.5MB on habanero) is left for skiroot. The new partition is called SKIROOT.

The changes to do this are in the skiroot branches under my account:

skiboot shenki/skiboot@9bac630

op-build shenki@ea6eec7

pnor shenki/pnor@a27f4bb

I've done test builds for habanero and palmetto, and successfully booted it on a plametto. You can reproduce my build check out my branch of op-build and run the following:

op-build P8_PORE_BINUTILS_VERSION=dcf4f87c0d9e8c8092ace0eb696189ee1056da30 \
 SKIBOOT_SITE=https://github.com/shenki/skiboot \
 SKIBOOT_SITE_METHOD=git \
 OPENPOWER_PNOR_SITE=https://github.com/shenki/pnor \
 OPENPOWER_PNOR_SITE_METHOD=git \
 OPENPOWER_PNOR_VERSION=49c2ccc6aaf36bb1eccd512391612136baa1a629 \ 
 SKIBOOT_VERSION=9701378341152718600048a13f829f3fc2224614

Please review my patches, and if we're happy with these changes I'll create PRs for pnor and submit the patch to skiboot, and then create a PR for op-build change along with rolling the dependences.

williamspatrick, 2015-02-02:
‎I don't think it is entirely accurate to say Palmetto vs Habanero anyhow. We're currently in the process of refactoring these ‎files into single-sided and dual-sided for the Golden side support. I would think if dual side (Habanero) can fit 22.5 mb than so can single.                                                                                                                                                                                                                                         From: Stewart SmithSent: Sunday, February 1, 2015 6:54 PMTo: open-power/op-buildReply To: open-power/op-buildSubject: Re: [op-build] Use separate partitions for skiboot and skiroot (#47)I'm slightly concerned about only 15MB for palmetto... we're right on that boundary now. My kernel+initramfs build is 15116262 bytes. Can we have 20MB on Palmetto?

The initramfs on separate partition probably does make sense too. Maybe we need to investigate a bit further as to what the size should be with modules.

williamspatrick, 2015-02-02:
‎Jeremy, Ben seemed to have expressed some concern with determining a good size split between the two because of modules vs static kernel. Have you two discussed this and come to a conclusion?                                                                                                                                                                                                                                          From: Jeremy KerrSent: Sunday, February 1, 2015 6:21 PMTo: open-power/op-buildReply To: open-power/op-buildSubject: Re: [op-build] Use separate partitions for skiboot and skiroot (#47)Yep, this is definitely something I'd like to get done - we've already come up with a pnor_load_resource that mikey will be using for the CAPP µcode bits. Might be good to sync with him on changes to the resource loading bits.

Could we include the initramfs as a separate partiton too?

jk-ozlabs, 2015-02-02:
Just to confirm - once we've shipped a PNOR with a certain partition layout, we're not able to resize partitions in future updates, right? Is the BMC code update mechanism defined yet?

If not, then defining the partition sizes isn't nearly as critical.

jk-ozlabs, 2015-02-02:
In fact, I'd say we stick to the combined kernel+initramfs now, and we can do some experimentation on a split initramfs separately.

ozbenh, 2015-02-02:
> ‎Keep in mind that 1mb only yields about 910k after ECC is applied. Is
> that really enough?  

So far we don't have ECC on the skiboot partition... that's another
things we might want to look into.

williamspatrick, 2015-02-02:
> ‎Keep in mind that 1mb only yields about 910k after ECC is applied. Is
> that really enough?  

So far we don't have ECC on the skiboot partition... that's another
things we might want to look into.

shenki, 2015-02-11:
Skiboot https://github.com/open-power/skiboot/commit/698044a77116bff651da24b30a773f8b963839f0 now contains the bits we need, and I've opened a PR https://github.com/open-power/pnor/pull/19 for updating the pnor xml and script.

Once that's in we can update the version of the pnor repository and change the build to stop building the kernel into skiboot.

shenki, 2015-03-18:
This is now present in master-next

closed 2015-03-18 by shenki