open-power/op-build #63: Broken build

shenki, 2015-02-26:
As of https://github.com/open-power/op-build/commit/95e650d3e1fe450e92f51befe6f195bdecd86456 the build errors out with `hostboot_build_images//HABANERO_HB.targeting.bin': No such file or directory` in the openpower-pnor build step.

```
PATH="/var/local/jenkins/build/workspace/openpower-habanero/output/host/bin:/var/local/jenkins/build/workspace/openpower-habanero/output/host/sbin:/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/bin:/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/sbin:/usr/local/bin:/usr/bin:/bin:/usr/games" /var/local/jenkins/build/workspace/openpower-habanero/output/build/openpower-pnor-94389f62eabecf00ef129b4184bf4ba2bc9cd193/update_image.pl -op_target_dir /var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/hostboot_build_images/ -hb_image_dir /var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/hostboot_build_images/ -scratch_dir /var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/openpower_pnor_scratch/ -hb_binary_dir /var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/hostboot_binaries/ -targeting_binary_filename "HABANERO_HB.targeting.bin.ecc" -targeting_binary_source "HABANERO_HB.targeting.bin" -sbe_binary_filename "venice_sbe.img.ecc" -sbec_binary_filename "centaur_sbec_pad.img.ecc" -occ_binary_filename /var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/occ/"occ.bin"
dd if=/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/hostboot_build_images//HABANERO_HB.targeting.bin of=/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HABANERO_HB.targeting.bin ibs=4k conv=sync
dd: failed to open '/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/hostboot_build_images//HABANERO_HB.targeting.bin': No such file or directory
Error running command: dd if=/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/hostboot_build_images//HABANERO_HB.targeting.bin of=/var/local/jenkins/build/workspace/openpower-habanero/output/host/usr/powerpc64-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//HABANERO_HB.targeting.bin ibs=4k conv=sync. Nonzero return code of (256) returned.
package/pkg-generic.mk:214: recipe for target '/var/local/jenkins/build/workspace/openpower-habanero/output/build/openpower-pnor-94389f62eabecf00ef129b4184bf4ba2bc9cd193/.stamp_images_installed' failed
make: *** [/var/local/jenkins/build/workspace/openpower-habanero/output/build/openpower-pnor-94389f62eabecf00ef129b4184bf4ba2bc9cd193/.stamp_images_installed] Error 1
```

williamspatrick, 2015-02-26:
@ploetzma - please take a look.

ploetzma, 2015-02-26:
Found the bug, pull request is open

https://github.com/open-power/op-build/pull/64

ploetzma, 2015-02-26:
Fix got merged, can you verify the build now works @shenki ?

shenki, 2015-02-27:
@ploetzma Yep, all fixed! Cheers

closed 2015-02-27 by shenki