open-power/op-build #81: op-build is not compiling - Unable to checkout '49cbaa7e123e46f9f63e69883e29769b4dd0bc4a' in submodule path 'buildroot'

geissonator, 2015-03-03:
[andrewg@gfw169 op_0303-op-build]$ git clone --recursive git@github.com:open-power/op-build.git
Cloning into op-build...

remote: Counting objects: 844, done.
remote: Total 844 (delta 0), reused 0 (delta 0), pack-reused 843
Receiving objects: 100% (844/844), 270.07 KiB | 161 KiB/s, done.
Resolving deltas: 100% (481/481), done.
Submodule 'buildroot' (https://github.com/open-power/buildroot) registered for path 'buildroot'
Cloning into buildroot...
remote: Counting objects: 134955, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 134955 (delta 0), reused 0 (delta 0), pack-reused 134950
Receiving objects: 100% (134955/134955), 35.95 MiB | 494 KiB/s, done.
Resolving deltas: 100% (92439/92439), done.
fatal: reference is not a tree: 49cbaa7e123e46f9f63e69883e29769b4dd0bc4a
Unable to checkout '49cbaa7e123e46f9f63e69883e29769b4dd0bc4a' in submodule path 'buildroot'
[andrewg@gfw169 op_0303-op-build]$ cd op-build
[andrewg@gfw169 op-build]$ . op-build-env
[andrewg@gfw169 op-build]$ op-build palmetto_defconfig && op-build
make: Entering directory `/esw/san2/andrewg/op_0303-op-build/op-build/buildroot'
make: *** No rule to make target `palmetto_defconfig'.  Stop.

williamspatrick, 2015-03-03:
@jk-ozlabs - might have just forgot to do the push of his local repository on buildroot.

williamspatrick, 2015-03-06:
This is resolved now.

closed 2015-03-06 by williamspatrick