open-power/op-build #89: Support custom kernel patches and defconfig.

williamspatrick, 2015-03-06:
With open-power/op-build#79 , I added a subdirectory open-power/custom that allows a user to place additional packages, new configs, and patches for existing patches.  This commit sets the BR2_GLOBAL_PATCH_DIR to enable patch locations.

Please expand this for the kernel to support:
1.  Kernel patches via BR2_LINUX_KERNEL_PATCH
2. Custom kernel configs via BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG

As part of this work it might make sense to move the openpower_defconfig from the custom kernel repository and into op-build as a starting default.

This feature (and #79) were opened as a result of conversations with one of the OpenPower partners designing their own system and seeing how they were currently modifying op-build and buildroot content.

williamspatrick, 2015-03-06:
@jk-ozlabs - assigning to you for your comments (and hopefully someone on your team's implementation).

apopple, 2015-03-07:
@jk-ozlabs - We should move our kernel configurations out of the kernel tree and into something like openpower/configs/kernel/palmetto_config. That way the can be customised more easily. We can then point op-build at the appropriate kernel config using BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG & BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE.

If we're in agreement I can create a pull request with this change.

jk-ozlabs, 2015-03-09:
@apopple - Yes, that sounds good. I'd be happy to move the configs to the op-build tree.

shenki, 2015-09-24:
We brought in `BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG` in op-build v1.1 (https://github.com/open-power/op-build/commit/ff724fbe7b1045f4b7cd3bc9f4379afd86a5f670), and `BR2_LINUX_KERNEL_PATCH` will be part of v1.6 (https://github.com/open-power/op-build/commit/3e323ab7542f3bb3336cfcae0e3897425cecb9b0).

I think we can close this bug now.

closed 2015-09-24 by stewart-ibm