open-power/op-build #90: Large files in rootfs.

williamspatrick, 2015-03-06:
/etc/eudev/hwdb has large text files that are suspected to be unused.
/usr/locale/locale-archive is quite large considering only English is supported.

Can these files be reduced or removed?

jk-ozlabs, 2016-04-28:
@sammj is looking at the hwdb at present. We support 11 languages (including English), hence the size of the locale db.

sammj, 2016-05-05:
During my work with fbterm I was running with hwdb generation disabled (for decent space savings); I didn't test thoroughly but I did not notice any differences in general use with the hwdb missing. Are we aware of any specific devices that we'll require the hwdb for?

stewart-ibm, 2018-08-07:
So, we've attacked a few things over the years to address this, and we can't currently see any big ticket obvious items to go squash.

Idea is we close this specific issue and open a new one for any specific items we find in the future.

closed 2018-08-07 by stewart-ibm