open-power/op-build #100: Building on Palmetto: Ubuntu Packages

skachm, 2015-03-26:
During packages installation: gcc-multilib package is not available, and the g++-multilib package could not be found. (I don't know if these are required for native compilation or not.)

$ sudo apt-get install gcc-multilib g++-multilib
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'gcc-multilib' has no installation candidate
E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'
$

williamspatrick, 2015-03-26:
multilib shouldn't be needed on Power boxes.  Only needed currently for x86_64.  We're working on getting rid of that dependency as well.

williamspatrick, 2015-03-26:
@anoo1, you were working op-build on Power.  Can you comment on this?

anoo1, 2015-03-26:
Yes correct, the following are the only needed libraries to build op-build on Ubuntu on Power, like Patrick mentions we'll be updating the README:

sudo apt-get install cscope ctags libz-dev libexpat-dev \
build-essential g++ git bison flex unzip \
libxml-simple-perl libxml-sax-perl libxml2-dev libxml2-utils xsltproc

Also need this ffs change, you can change the "op-build/openpower/package/openpower-ffs/openpower-ffs.mk" to pick it up:
https://github.com/anoo1/ffs

skachm, 2015-03-31:
Those packages and the ffs fix seem to work (op-build built without any errors). How do I load the image onto the Palmetto to test now? 

Thanks!

williamspatrick, 2015-03-31:
The output/images directory should have palmetto.pnor.  There isn't a code update mechanism through the BMC http interface yet, so you have to flash the PNOR device from the BMC command line directly.  Power off the system, use 'scp' or 'rsync' to get the palmetto.pnor image onto your BMC (usually we place it in /tmp), and then use 'pflash' to write the image onto the PNOR device.

pflash -E -f -p /tmp/palmetto.pnor

If you want to save your current image off first, I think there is also a pflash command to read the entire device to a file.  See pflash --help.

skachm, 2015-04-05:
I get a 'command not found' error when trying to use pflash through the ipmitool connected via SOL ethernet. The TYAN documentation to access BMC suggests a serial-to-USB cable. Is this required? Or am I doing something wrong?

williamspatrick, 2015-04-06:
The SOL is to the Host OS.  This command needs to be ran on the BMC OS.  You can typically either use the serial port to the BMC or ssh to it.

@wghoffa - Any additional comments?

skachm, 2015-04-06:
I SSHed into the BMC OS with admin/admin, but I don't have write access and I get a permission denied trying to remount with write access. scp, sudo and su don't seem to be installed. rsync is installed, but still can't write.

matt@clarity01:~/home$ rsync -a ~/home admin@192.168.0.5:/home
admin@192.168.0.5's password: 
rsync: recv_generator: mkdir "/home/home" failed: Read-only file system (30)
*** Skipping any contents from this failed directory ***
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1183) [sender=3.1.0]

williamspatrick, 2015-04-06:
Correct, the flash of the BMC itself is mounted read-only.  This is why we rsync the flash image into /tmp and flash from there.

skachm, 2015-04-06:
/tmp is also mounted read only and rsync gives the same error.

skachm, 2015-04-06:
Still don't know why I can't write to /tmp, but I discovered I can write to /var so I used rsync to put palmetto.pnor there. Now pflash has permission problems:

$ pflash -E -f -p /var/palmetto.pnor     
can't open /dev/mem: Operation not permitted

wghoffa, 2015-04-06:
I think you want to ssh to the BMC with different credentials than
admin/admin. I use admin/admin for the webgui, but use sysadmin/superuser
when I SSH. I believe this should get you the permissions you need. I
assume it should also allow you to write to /tmp.

skachm, 2015-04-06:
That did it, thanks! Now booting a Palmetto with op-build v1.0 built natively on the Palmetto.

closed 2015-04-08 by skachm