open-power/op-build #101: Building on Palmetto: kmod, --enable-static

skachm, 2015-03-26:
kmod throws an error while configuring that '--enable-static" is not supported.

$ op-build palmetto_defconfig && op-build
...
...
>>> host-kmod 17 Configuring
...
...
checking whether to build static libraries... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
configure: error: --enable-static is not supported by kmod
make: *** [/home/clarity/op-build/output/build/host-kmod-17/.stamp_configured] Error 1
make: Leaving directory `/home/clarity/op-build/buildroot'
$

williamspatrick, 2015-03-26:
Jeremy, any ideas?  I don't think this is related to multilib support.  Maybe an old version of kmod?

skachm, 2015-03-29:
This seems to be a 14.04 problem, I upgraded our other Palmetto to 14.10 and it built.

closed 2015-03-29 by skachm

anoo1, 2015-04-30:
The README specifies that 14.04 is supported, so would make sense to provide a patch for 14.04 LE.

Workaround is to comment out line 12321 on file op-build/output/build/host-kmod-17/configure:

if test "x$enable_static" = "xyes"; then :
# as_fn_error $? "--enable-static is not supported by kmod" "$LINENO" 5
fi

anoo1, 2015-04-30:
Opened issue https://github.com/open-power/op-build/issues/162 to follow-up since this one is closed.