open-power/op-build #102: Tracker issue for op-build v1.1 release issues.

williamspatrick, 2015-03-27:
This is the tracking issue for any features and bugs we need to get into v1.1.  Tentative release is early in the week of 2015-03-30.

dcrowell77, 2015-03-27:
Support for pulling system information from OSYS record of planar vpd.  Fixed with Hostboot change Iaa9c4e00325f8fa6efb7a9fca1275bcea2759308

williamspatrick, 2015-03-27:
Iaa9c4 is open-power/hostboot@2114a66c76558f41cd305fc294ccdfaf6607c0e5 .

williamspatrick, 2015-03-27:
Created #105 for Dan's commit.

bofferdn, 2015-03-27:
Support for rebooting system automatically after SBE update.  Fixed with Hostboot change Id251307aa76259d8a290a92d9e9f3418b181bc3e

bofferdn, 2015-03-27:
There is a concern by Mike that Id251307aa76259d8a290a92d9e9f3418b181bc3e will break systems with older levels of AMI firmware; need to talk through that risk.

bofferdn, 2015-03-27:
Need fix to prevent Hostboot from powering off when it receives a stray IPMI message from BMC that matches the power off command op-code, but different netFn which Hostboot is not screening out.  Can't locate the change ID yet, need to follow up with Bill.

wghoffa, 2015-03-27:
http://gfw160.aus.stglabs.ibm.com:8080/gerrit/#/c/16671/ is the change.

williamspatrick, 2015-03-27:
IPMI change from Bill is I378e0763c5ca8c13830d0e6c057b24e6beb5dc8c.

bofferdn, 2015-03-27:
Ben also wants the commit a98bf456ff5d9f18dc2af1ffc59d67a302029a76 from habanero.xml in habanero-xml package to fix some sensor numbering.

bofferdn, 2015-03-28:
Placeholder for commits from Jeremy
- One to reset boot count to 2 after boot success
- One to trigger power cycle correctly

bofferdn, 2015-03-30:
Need a fix from Elizabeth to cap binary trace size to under a critical threshold, otherwise various tasks are crashing.  Change-id forthcoming.

wghoffa, 2015-03-30:
Elizabeth's change is: If8f08d1602b4437aa0e72e6d2e8d99ca615a2c77

wghoffa, 2015-03-30:
I have two also:
    1- Updating VPD fields used for populating FRU Inventory Records: Iaa4802e72e355acd03cb88e58c40d1069dd76ef5
    2- Add PNOR version information to FRU Inventory Record: Ib49fe67e9c6631b2b7ea0005e692c9aea6d84057

williamspatrick, 2015-03-31:
Processor team would like us to pull I245cd9947d5b7a05ff9a8f92ff18c658b1fe09d6 in as well.

williamspatrick, 2015-03-31:
Opened issues for all the outstanding comments listed in this tracker.  I'm going to start closing them out and pointing to Bill's pull request, etc.

bofferdn, 2015-04-02:
Please include 
I631dc758e03316a28d13f6d19032c1a76e3cffc1 - Dan's VPD cache fix
Ia4f6440551b4bb6d4a849d24290c19667f3a6c6b - Dan's  missing OSYS:SS data fix for Palmetto
Ie967d81917e9b942d5cb516856075ebdb5029487 - Mike's i2c arbitration loss fix

bofferdn, 2015-04-02:
Elizabeth's earlier change ID If8f08d1602b4437aa0e72e6d2e8d99ca615a2c77 doesn't appear to fix Palmetto, which means the same could be true for Habanero (but not seen).  Trying to get to the bottom of this tonight --I'll update this when we find out more.

williamspatrick, 2015-04-02:
Ia4f6 is already in as hostboot-0017-Do-not-fail-IPL-for-missing-OSYS-SS-data.patch (#119)

williamspatrick, 2015-04-02:
@bofferdn - Can you point to what the fix for If8f08 is not solving?  Is there a defect about it?

williamspatrick, 2015-04-02:
Opened #125 and #126 for I631 and Ie96 respectively.

bofferdn, 2015-04-03:
>>> Can you point to what the fix for If8f08 is not solving? Is there a defect about it?
Defect ==> http://w3.rchland.ibm.com/projects/bestquest/?defect=SW300117.  Hit issue on Palmetto w/ the fix in place.

williamspatrick, 2015-04-06:
Created tag for v1.1.

closed 2015-04-06 by williamspatrick