open-power/op-build #162: Ubuntu 14.04 LE: op-build fails with kmod --enable-static not supported

anoo1, 2015-04-30:
Opening this issue since https://github.com/open-power/op-build/issues/101 was closed.

Other teams have a requirement to run on 14.04 and they're unable to cleanly build op-build on Power systems. README specifies that 14.04 is supported.
Copying error msg from 101:

$ op-build palmetto_defconfig && op-build
...
...
>>> host-kmod 17 Configuring
...
...
checking whether to build static libraries... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
configure: error: --enable-static is not supported by kmod
make: *** [/home/clarity/op-build/output/build/host-kmod-17/.stamp_configured] Error 1
make: Leaving directory `/home/clarity/op-build/buildroot'
$

Current workaround is to comment out line 12321 on file op-build/output/build/host-kmod-17/configure:

if test "x$enable_static" = "xyes"; then :
# as_fn_error $? "--enable-static is not supported by kmod" "$LINENO" 5
fi

williamspatrick, 2015-05-01:
Seems like there is a problem with the host-kmod package in buildroot.

stewart-ibm, 2015-06-19:
I hit this on clean 14.04 too. Looks like 14.10 is our only supported Ubuntu version on ppc64el at least for building op-build.

stewart-ibm, 2015-09-28:
Can you try again with latest master-next? With the buildroot update I think this should be fixed.

anoo1, 2015-10-05:
Still same error with latest master-next.

$ git log
commit 71faa0f76f48f415125b9f2d663c617b4131c17c
Merge: b65cbe9 e4aa6fb
Author: Patrick Williams <patrick@stwcx.xyz>
Date:   Wed Sep 30 21:58:01 2015 -0500
    Merge pull request #281 from shenki/v4.2.1-openpower1
    linux: Update to v4.2.1-openpower1

$ op-build palmetto_defconfig && op-build
...
checking whether to enable maintainer-specific portions of Makefiles... yes
configure: error: --enable-static is not supported by kmod
make: *** [/home/tyan/op-build/output/build/host-kmod-17/.stamp_configured] Error 1
make: Leaving directory `/home/tyan/op-build/buildroot'

stewart-ibm, 2016-03-02:
I'm not hitting this on 14.04 in a Docker container on ppc64le on current master-next, is this still an issue? Please re-open if so.

closed 2016-03-02 by stewart-ibm