open-power/op-build #193: firestone_defconfig doesn't build in master-next

stewart-ibm, 2015-06-22:
As of fe9ad08866d98a865ac0927a02d6778c3155d888 (refs/remotes/origin/master-next) at least, firestone_defconfig fails to build:

cp -f ../../../obj/genfiles/p8.npu.scom.if ../../../img/p8.npu.scom.if
    DEP        fsidd.C
    MAKE       attn CODE
    CXX        vfsrp.C
    DEP        mbox_dma_buffer.C
    DEP        attrsync.C
    MAKE       fapi CODE
    DEP        mdiasm.C
    MAKE       hwp CODE
In file included from ../../../src/include/usr/hwpf/fapi/fapiReturnCode.H:76:0,
                 from ../../../src/include/usr/hwpf/fapi/fapiUtil.H:63,
                 from ecmdDataBufferBase.C:41:
../../../obj/genfiles/fapiHwpErrorInfo.H:20634:5: error: redefinition of 'REG_FFDC_TEST_X_EX_REGISTERS'
     REG_FFDC_TEST_X_EX_REGISTERS = 0xb225f1ca,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:20215:5: error: 'REG_FFDC_TEST_X_EX_REGISTERS' previously defined here
     REG_FFDC_TEST_X_EX_REGISTERS = 0xb225f1ca,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:20639:5: error: redefinition of 'RC_MSS_MAINT_PUT_MARK_STORE_BAD_INPUT_RANK'
     RC_MSS_MAINT_PUT_MARK_STORE_BAD_INPUT_RANK = 0xff72b733,
     ^
../../../obj/genfiles/fapiHwpErrorInfo.H:20575:5: error: 'RC_MSS_MAINT_PUT_MARK_STORE_BAD_INPUT_RANK' previously defined here
     RC_MSS_MAINT_PUT_MARK_STORE_BAD_INPUT_RANK = 0xff72b733,
     ^

williamspatrick, 2015-06-22:
Hmm. This has the appearance of a parallel build issue. 

@jazurin - can you take a look?  I think you indicated Firestone built for you in the pull request.

stewart-ibm, 2015-06-23:
This would be a parallel build, building in 160 vCPU guest.

stewart-ibm, 2015-08-14:
Seems to be fixed.. hasn't reproduced since.

closed 2015-08-14 by stewart-ibm