open-power/op-build #194: Ubuntu 15.05 fails compile on Power server  'libfakeroot.so' not found

causten, 2015-06-23:
if [ ! -e /home/causten/gitrepos/op-build/output/target/init ]; then /usr/bin/install -m 0755 fs/cpio/                                       init /home/causten/gitrepos/op-build/output/target/init; fi
rm -f /home/causten/gitrepos/op-build/output/build/_fakeroot.fs
rm -f /home/causten/gitrepos/op-build/output/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
rm -f /home/causten/gitrepos/op-build/output/build/_users_table.txt
echo "chown -h -R 0:0 /home/causten/gitrepos/op-build/output/target" >> /home/causten/gitrepos/op-buil                                       d/output/build/_fakeroot.fs
cat system/device_table.txt > /home/causten/gitrepos/op-build/output/build/_device_table.txt
printf '/bin/busybox                     f 4755 0 0 - - - - -\n/usr/share/udhcpc/default.script f 755                                         0 0 - - - - -\n /dev/console c 622 0 0 5 1 - - -\n' >> /home/causten/gitrepos/op-build/output/build/_                                       device_table.txt
echo "/home/causten/gitrepos/op-build/output/host/usr/bin/makedevs -d /home/causten/gitrepos/op-build/                                       output/build/_device_table.txt /home/causten/gitrepos/op-build/output/target" >> /home/causten/gitrepo                                       s/op-build/output/build/_fakeroot.fs
printf '' >> /home/causten/gitrepos/op-build/output/build/_users_table.txt
PATH="/home/causten/gitrepos/op-build/output/host/bin:/home/causten/gitrepos/op-build/output/host/sbin                                       :/home/causten/gitrepos/op-build/output/host/usr/bin:/home/causten/gitrepos/op-build/output/host/usr/s                                       bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games" /home/ca                                       usten/gitrepos/op-build/buildroot/support/scripts/mkusers /home/causten/gitrepos/op-build/output/build                                       /_users_table.txt /home/causten/gitrepos/op-build/output/target >> /home/causten/gitrepos/op-build/out                                       put/build/_fakeroot.fs
echo "  cd /home/causten/gitrepos/op-build/output/target && find . | cpio --quiet -o -H newc > /home/c                                       austen/gitrepos/op-build/output/images/rootfs.cpio" >> /home/causten/gitrepos/op-build/output/build/_f                                       akeroot.fs
chmod a+x /home/causten/gitrepos/op-build/output/build/_fakeroot.fs
PATH="/home/causten/gitrepos/op-build/output/host/bin:/home/causten/gitrepos/op-build/output/host/sbin                                       :/home/causten/gitrepos/op-build/output/host/usr/bin:/home/causten/gitrepos/op-build/output/host/usr/s                                       bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games" /home/ca                                       usten/gitrepos/op-build/output/host/usr/bin/fakeroot -- /home/causten/gitrepos/op-build/output/build/_                                       fakeroot.fs
fakeroot: preload library `libfakeroot.so' not found, aborting.
make: *** [/home/causten/gitrepos/op-build/output/images/rootfs.cpio] Error 1
make: Leaving directory `/home/causten/gitrepos/op-build/buildroot'

williamspatrick, 2015-06-23:
This looks like a general buildroot problem.

stewart-ibm, 2016-03-03:
Still with master-next on Ubuntu 14.04.

I think I have a fix for that, the bug is that host-fakeroot is being built without the libfakeroot.so library, due to a failure in the (ancient) libtool to properly detect that we can in fact build shared libs for our host platform, just not for elf64ppc.

Patch incoming, will get into our buildroot and upstream.

stewart-ibm, 2016-03-04:
Now fixed in master-next

closed 2016-03-04 by stewart-ibm

munroesj52, 2018-05-01:
Hi Stewart!. I am on a quixotic mission to make my PMAC G5 quad usefull for a little longer.

The Ubuntu 14.04 PPC64 install work mostly but the 64-bit user land is incomplete. Should not be a issue as I know people and the AT team opensourced the AT build scripts (https://github.com/advancetoolchain/advance-toolchain) for those brave (or foolish) enough to try.

After fighting this fish for 2 days, it fails at the last step with: 
ERROR: ld.so: object 'libfakeroot-sysv.so' from LD_PRELOAD cannot be preloaded:

So missing the 64-bit libfakeroot-sysv.so. Now I have a (actually several) 64-bit toolchains and build this myself, But I suspect I will need your patch? It is included in https://github.com/open-power/op-build?

I just need to build the 64-bit libfakeroot-sysv.so ...