open-power/op-build #204: Build fails on Fedora 22

irarayjenkins, 2015-07-16:
Machine: Linux gemini 4.0.7-300.fc22.x86_64 #1 SMP Mon Jun 29 22:15:06 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

Seems to be an issue between ncurses and gcc: http://lists.gnu.org/archive/html/bug-ncurses/2014-07/msg00020.html 

GCC version: gcc (GCC) 5.1.1 20150618 (Red Hat 5.1.1-4)

Error:
In file included from ../ncurses/curses.priv.h:283:0,
                 from ../ncurses/lib_gen.c:19:
_25759.c:835:15: error: expected ‘)’ before ‘int’
../include/curses.h:1594:56: note: in definition of macro ‘mouse_trafo’
 #define mouse_trafo(y,x,to_screen) wmouse_trafo(stdscr,y,x,to_screen)
                                                        ^
/usr/bin/gcc -DHAVE_CONFIG_H -I../ncurses -I. -I/home/ijenkins/dev/op-build/output/host/usr/include -D_GNU_SOURCE -DNDEBUG -I. -I../include -I/home/ijenkins/dev/op-build/output/host/usr/include/ncurses -O2 -I/home/ijenkins/dev/op-build/output/host/usr/include  --param max-inline-insns-single=1200 -fPIC -c ../ncurses/./base/lib_inchstr.c -o ../obj_s/lib_inchstr.o
Makefile:795: recipe for target '../obj_s/lib_gen.o' failed
make[2]: *** [../obj_s/lib_gen.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/ijenkins/dev/op-build/output/build/host-ncurses-5.9/ncurses'
Makefile:307: recipe for target '../lib/libncurses.so' failed
make[1]: *** [../lib/libncurses.so] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/ijenkins/dev/op-build/output/build/host-ncurses-5.9/progs'
package/pkg-generic.mk:177: recipe for target '/home/ijenkins/dev/op-build/output/build/host-ncurses-5.9/.stamp_built' failed
make: *** [/home/ijenkins/dev/op-build/output/build/host-ncurses-5.9/.stamp_built] Error 2
make: Leaving directory '/home/ijenkins/dev/op-build/buildroot'

williamspatrick, 2015-07-16:
@jk-ozlabs - I assume we need an update to buildroot to tackle this.

irarayjenkins, 2015-07-17:
This may be relevant: http://trac.sagemath.org/ticket/18301

williamspatrick, 2015-08-13:
@jk-ozlabs - Any update?

naynajain, 2015-09-02:
ncurses has released their new package ncurses-6.0 which has the fix to work fine with gcc > 5.
And it looks like latest buildroot released on 2015-08 has this patch which works around the gcc > 5 incompatibility issue. It still pulls ncurses-5.9 but has the patch to fix gcc > 5 issue.
Buildroot which is currently used in open-power is still of 2014-05.  So, looks like upgrading buildroot might fix this issue and probably few other related to gcc > 5 incompatibility.

Apart from this there is also gcc > 5 issue which compiling localedef package. And it gets fixed by adding 
HOST_LOCALEDEF_CONF_ENV = CFLAGS="$(HOST_CFLAGS) -fgnu89-inline"
to file buildroot/package/localedef/localedef.mk

stewart-ibm, 2015-09-02:
Ahh, excellent. I have pull requests open to go to 2015-05, and it should be simple to (from there) go to 2015-08. @shenki is also working on getting our remaining buildroot patches upstream, which means hopefully by end of year we can just use upstream buildroot.

naynajain, 2015-09-03:
Ok.. That's great !! It seems the localedef issue, I mentioned in previous comment,  is still not yet fixed in 2015-08 version of buildroot. So, for open-power buildroot to have this fix, I think will have to be explicitly added.

stewart-ibm, 2015-09-15:
We've merged in buildroot 2015-05 for v1.7, looking at 2015-08 for v1.8

However... if you get a pull request up to our buildroot that'll fix that issue, I'd happily look at it.

stewart-ibm, 2016-01-12:
Current op-build master-next builds for me on Fedora 23, so I think we can close this now.

closed 2016-01-12 by stewart-ibm