open-power/op-build #206: zlib fails to compile because sourceforge is down

irarayjenkins, 2015-07-17:
Error:
--2015-07-17 10:33:13--  http://downloads.sourceforge.net/project/libpng/zlib/1.2.8/zlib-1.2.8.tar.xz
Resolving downloads.sourceforge.net (downloads.sourceforge.net)... 54.240.188.66, 54.230.70.184, 54.230.70.69, ...
Connecting to downloads.sourceforge.net (downloads.sourceforge.net)|54.240.188.66|:80... connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://downloads.sourceforge.net/#!/project/libpng/zlib/1.2.8/zlib-1.2.8.tar.xz [following]
--2015-07-17 10:33:13--  http://downloads.sourceforge.net/
Reusing existing connection to downloads.sourceforge.net:80.
HTTP request sent, awaiting response... 200 OK
Length: 639 [text/html]
Saving to: ‘/home/ijenkins/dev/op-build/dl/zlib-1.2.8.tar.xz.tmp’

100%[============================================================>] 639         --.-K/s   in 0s      

2015-07-17 10:33:13 (48.2 MB/s) - ‘/home/ijenkins/dev/op-build/dl/zlib-1.2.8.tar.xz.tmp’ saved [639/639]

>>> zlib 1.2.8 Extracting
xzcat /home/ijenkins/dev/op-build/dl/zlib-1.2.8.tar.xz | tar --strip-components=1 -C /home/ijenkins/dev/op-build/output/build/zlib-1.2.8  -xf -
xzcat: /home/ijenkins/dev/op-build/dl/zlib-1.2.8.tar.xz: File format not recognized
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make: *** [/home/ijenkins/dev/op-build/output/build/zlib-1.2.8/.stamp_extracted] Error 2
make: Leaving directory `/home/ijenkins/dev/op-build/buildroot'

If you look at the zlib file:
ijenkins@ares:~/dev/op-build$ cat /home/ijenkins/dev/op-build/dl/zlib-1.2.8.tar.xz 
<html><head>
<title>SourceForge</title>

<script src="//code.jquery.com/jquery-1.11.0.min.js"></script>
<script src="//sourceforge.net/js/mirrors.js"></script>
<script src="/js/sf.js"></script>
<script>
var DR_loc = DR_parse_hash_url();
if (DR_loc) {
    DR_sf_main(DR_loc);
} else {
    window.location.href = 'http://sourceforge.net/home.html';
}
</script>
</head><body>
<noscript>
We're sorry -- the Sourceforge site is currently in Disaster Recovery mode, and currently requires
the use of javascript to function.  Please check back later.
</noscript>
</body></html>

Maybe change the op-build to pull zlib from http://zlib.net/ instead of Sourceforge?

irarayjenkins, 2015-07-17:
Maybe we don't want to depend on Sourceforge, in general? 

http://www.pcworld.com/article/2938017/why-big-open-source-projects-are-fleeing-sourceforges-free-software-hub.html

williamspatrick, 2015-07-17:
We can't really help it. Buildroot, as the Linux distribution we are basing things on, is pointing at sf because that is the hosting location for that library.

stewart-ibm, 2015-07-18:
Agree with both. We should probably poke upstream to not rely on sourceforge :)

ozbenh, 2015-07-18:
On Fri, 2015-07-17 at 19:14 -0700, Stewart Smith wrote:
> Agree with both. We should probably poke upstream to not rely on
> sourceforge :)

We should have our own mirror of everything we need for a point release
and have the "release" op-build pull from that.

Cheers,
Ben.

closed 2015-08-01 by williamspatrick