open-power/op-build #216: Unable to build master-next

nacc, 2015-07-31:
Attempting to build garrison_defconfig, I get:

>>> linux d0ff068a9962ba96d4e6ddfba2a4e9fd8bc9a763 Building
PATH="/home/nacc/op-build/output/host/bin:/home/nacc/op-build/output/host/sbin:/home/nacc/op-build/output/host/usr/bin:/home/nacc/op-build/output/host/usr/sbin:/home/nacc/bin:/home/nacc/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games" /usr/bin/make -j9 HOSTCC="/usr/bin/gcc" HOSTCFLAGS="" ARCH=powerpc INSTALL_MOD_PATH=/home/nacc/op-build/output/target CROSS_COMPILE=" /home/nacc/op-build/output/host/usr/bin/powerpc64-buildroot-linux-gnu-" DEPMOD=/home/nacc/op-build/output/host/sbin/depmod -C /home/nacc/op-build/output/build/linux-d0ff068a9962ba96d4e6ddfba2a4e9fd8bc9a763 zImage.epapr
make[1]: Entering directory '/home/nacc/op-build/output/build/linux-d0ff068a9962ba96d4e6ddfba2a4e9fd8bc9a763'
scripts/kconfig/conf --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  WRAP    arch/powerpc/include/generated/asm/clkdev.h
  WRAP    arch/powerpc/include/generated/asm/hash.h
  WRAP    arch/powerpc/include/generated/asm/irq_work.h
  WRAP    arch/powerpc/include/generated/asm/mcs_spinlock.h
  WRAP    arch/powerpc/include/generated/asm/preempt.h
  WRAP    arch/powerpc/include/generated/asm/scatterlist.h
  WRAP    arch/powerpc/include/generated/asm/rwsem.h
  WRAP    arch/powerpc/include/generated/asm/trace_clock.h
  WRAP    arch/powerpc/include/generated/asm/vtime.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  UPD     include/config/kernel.release
  HOSTCC  scripts/basic/bin2c
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  CC      kernel/bounds.s
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/pnmtologo
  HOSTCC  scripts/conmakehash
  SHIPPED scripts/dtc/dtc-lexer.lex.c
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/dtc/checks.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/mod/mk_elfconfig
  SHIPPED scripts/dtc/dtc-parser.tab.h
  SHIPPED scripts/dtc/dtc-parser.tab.c
  GEN     scripts/mod/devicetable-offsets.h
  GEN     include/generated/bounds.h
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  CC      arch/powerpc/kernel/asm-offsets.s
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/livetree.o
  MKELF   scripts/mod/elfconfig.h
  HOSTCC  scripts/dtc/srcpos.o
  HOSTCC  scripts/mod/file2alias.o
  HOSTCC  scripts/mod/modpost.o
  HOSTCC  scripts/dtc/treesource.o
  HOSTCC  scripts/dtc/util.o
  HOSTCC  scripts/dtc/dtc-lexer.lex.o
  HOSTCC  scripts/mod/sumversion.o
  HOSTCC  scripts/dtc/dtc-parser.tab.o
  GEN     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  HOSTLD  scripts/dtc/dtc
  HOSTLD  scripts/mod/modpost
  CC      init/main.o
  CHK     include/generated/compile.h
  CC      init/do_mounts.o
  CC      init/do_mounts_md.o
  CC      init/do_mounts_initrd.o
  CC      init/initramfs.o
  CC      init/do_mounts_rd.o
  HOSTCC  usr/gen_init_cpio
  CC      arch/powerpc/kernel/cputable.o
  UPD     include/generated/compile.h
  CC      init/init_task.o
  GEN     usr/initramfs_data.cpio.gz
  CC      arch/powerpc/kernel/ptrace.o
  CC      init/version.o
  CC      arch/powerpc/kernel/syscalls.o
  AS      usr/initramfs_data.o
  CC      arch/powerpc/mm/fault.o
  CC      arch/powerpc/mm/mem.o
  CC      arch/powerpc/kernel/irq.o
  LD      usr/built-in.o
/home/nacc/op-build/output/host/usr/bin/powerpc64-buildroot-linux-gnu-ld: unrecognised emulation mode: elf64lppc
Supported emulations: elf64ppc elf32ppclinux elf32ppc elf32ppcsim
scripts/Makefile.build:336: recipe for target 'usr/built-in.o' failed
make[2]: *** [usr/built-in.o] Error 1
Makefile:929: recipe for target 'usr' failed
make[1]: *** [usr] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC      arch/powerpc/mm/pgtable.o
  CC      arch/powerpc/kernel/align.o
  CC      arch/powerpc/mm/gup.o
  LD      init/mounts.o
/home/nacc/op-build/output/host/usr/bin/powerpc64-buildroot-linux-gnu-ld: unrecognised emulation mode: elf64lppc
Supported emulations: elf64ppc elf32ppclinux elf32ppc elf32ppcsim
scripts/Makefile.build:390: recipe for target 'init/mounts.o' failed
make[2]: *** [init/mounts.o] Error 1
Makefile:929: recipe for target 'init' failed
make[1]: *** [init] Error 2
  CC      arch/powerpc/mm/mmap.o
  CC      arch/powerpc/kernel/signal_32.o
  CC      arch/powerpc/mm/init_64.o
  CC      arch/powerpc/kernel/pmc.o
  CC      arch/powerpc/mm/pgtable_64.o
  CC      arch/powerpc/mm/hash_utils_64.o
  CC      arch/powerpc/kernel/vdso.o
  CC      arch/powerpc/kernel/process.o
  AS      arch/powerpc/mm/slb_low.o
  CPP     arch/powerpc/kernel/systbl_chk.i
  CC      arch/powerpc/kernel/idle.o
  CC      arch/powerpc/mm/slb.o
  CC      arch/powerpc/kernel/signal.o
  CC      arch/powerpc/mm/hash_native_64.o
  CC      arch/powerpc/kernel/sysfs.o
  AS      arch/powerpc/mm/hash_low_64.o
  CC      arch/powerpc/kernel/cacheinfo.o
  CC      arch/powerpc/kernel/time.o
  CC      arch/powerpc/mm/tlb_hash64.o
  CC      arch/powerpc/kernel/prom.o
  CC      arch/powerpc/mm/mmu_context_hash64.o
  CC      arch/powerpc/mm/numa.o
  CC      arch/powerpc/kernel/traps.o
  CC      arch/powerpc/mm/slice.o
  CC      arch/powerpc/kernel/setup-common.o
  CC      arch/powerpc/mm/hugetlbpage.o
  CC      arch/powerpc/kernel/udbg.o
  AS      arch/powerpc/kernel/misc.o
  CC      arch/powerpc/kernel/io.o
  CC      arch/powerpc/kernel/dma.o
  AS      arch/powerpc/kernel/misc_64.o
  CC      arch/powerpc/kernel/setup_64.o
  CC      arch/powerpc/kernel/sys_ppc32.o
  LDS     arch/powerpc/kernel/vdso64/vdso64.lds
  VDSO64A arch/powerpc/kernel/vdso64/sigtramp.o
  LD      arch/powerpc/mm/built-in.o
  VDSO64A arch/powerpc/kernel/vdso64/gettimeofday.o
  VDSO64A arch/powerpc/kernel/vdso64/datapage.o
  LDS     arch/powerpc/kernel/vdso32/vdso32.lds
/home/nacc/op-build/output/host/usr/bin/powerpc64-buildroot-linux-gnu-ld: unrecognised emulation mode: elf64lppc
Supported emulations: elf64ppc elf32ppclinux elf32ppc elf32ppcsim
scripts/Makefile.build:336: recipe for target 'arch/powerpc/mm/built-in.o' failed
make[2]: *** [arch/powerpc/mm/built-in.o] Error 1
Makefile:929: recipe for target 'arch/powerpc/mm' failed
make[1]: *** [arch/powerpc/mm] Error 2
  VDSO64A arch/powerpc/kernel/vdso64/cacheflush.o
  CC      arch/powerpc/kernel/signal_64.o
  VDSO32A arch/powerpc/kernel/vdso32/sigtramp.o
  CC      arch/powerpc/kernel/ptrace32.o
  VDSO32A arch/powerpc/kernel/vdso32/gettimeofday.o
  VDSO64A arch/powerpc/kernel/vdso64/note.o
  VDSO32A arch/powerpc/kernel/vdso32/datapage.o
  VDSO64A arch/powerpc/kernel/vdso64/getcpu.o
  CC      arch/powerpc/kernel/paca.o
  VDSO32A arch/powerpc/kernel/vdso32/cacheflush.o
  CC      arch/powerpc/kernel/nvram_64.o
  VDSO64L arch/powerpc/kernel/vdso64/vdso64.so.dbg
  VDSO32A arch/powerpc/kernel/vdso32/note.o
/home/nacc/op-build/output/host/usr/lib/gcc/powerpc64-buildroot-linux-gnu/4.8.3/../../../../powerpc64-buildroot-linux-gnu/bin/ld: unrecognised emulation mode: elf64lppc
Supported emulations: elf64ppc elf32ppclinux elf32ppc elf32ppcsim
collect2: error: ld returned 1 exit status
arch/powerpc/kernel/vdso64/Makefile:26: recipe for target 'arch/powerpc/kernel/vdso64/vdso64.so.dbg' failed
make[3]: *** [arch/powerpc/kernel/vdso64/vdso64.so.dbg] Error 1
scripts/Makefile.build:404: recipe for target 'arch/powerpc/kernel/vdso64' failed
make[2]: *** [arch/powerpc/kernel/vdso64] Error 2
make[2]: *** Waiting for unfinished jobs....
  VDSO32A arch/powerpc/kernel/vdso32/getcpu.o
  VDSO32L arch/powerpc/kernel/vdso32/vdso32.so.dbg
/home/nacc/op-build/output/host/usr/lib/gcc/powerpc64-buildroot-linux-gnu/4.8.3/../../../../powerpc64-buildroot-linux-gnu/bin/ld: unrecognised emulation mode: elf32lppclinux
Supported emulations: elf64ppc elf32ppclinux elf32ppc elf32ppcsim
collect2: error: ld returned 1 exit status
arch/powerpc/kernel/vdso32/Makefile:33: recipe for target 'arch/powerpc/kernel/vdso32/vdso32.so.dbg' failed
make[3]: *** [arch/powerpc/kernel/vdso32/vdso32.so.dbg] Error 1
scripts/Makefile.build:404: recipe for target 'arch/powerpc/kernel/vdso32' failed
make[2]: *** [arch/powerpc/kernel/vdso32] Error 2
Makefile:929: recipe for target 'arch/powerpc/kernel' failed
make[1]: *** [arch/powerpc/kernel] Error 2
make[1]: Leaving directory '/home/nacc/op-build/output/build/linux-d0ff068a9962ba96d4e6ddfba2a4e9fd8bc9a763'
package/pkg-generic.mk:177: recipe for target '/home/nacc/op-build/output/build/linux-d0ff068a9962ba96d4e6ddfba2a4e9fd8bc9a763/.stamp_built' failed
make: *** [/home/nacc/op-build/output/build/linux-d0ff068a9962ba96d4e6ddfba2a4e9fd8bc9a763/.stamp_built] Error 2
make: Leaving directory '/home/nacc/op-build/buildroot'

I've tried `op-build clean`, because it seems like the toolchain isn't getting rebuilt to support LE?

This is on an Ubuntu 15.04 x86_64 installation...

jk-ozlabs, 2015-07-31:
@nacc: duplicate of https://github.com/open-power/op-build/issues/210 - you want this fix: https://github.com/open-power/op-build/pull/213, or update your .config to be little-endian: https://github.com/stewart-ibm/op-build/commit/6e66114a6d15a04a1760058ed22f613b1530dbaf

jk-ozlabs, 2015-07-31:
Ah, but that fix doesn't cover garrison. Reopening.

closed 2015-07-31 by williamspatrick

touchqvfb, 2015-12-09:
Hi spatrick,
I still get the error blow,when I complie op-build, I try openpower_mambo_defconfig or firestone_defconfig or palmetto_defconfig.it is the same error.I use the op-build default compiler,and change nothing.the project is LE. How can I solve the error. Many thanks.
BR2_powerpc64le=y
BR2_powerpc_power8=y

BOOTCC arch/powerpc/boot/fdt_ro.o
BOOTCC arch/powerpc/boot/fdt_rw.o
BOOTCC arch/powerpc/boot/fdt_strerror.o
BOOTCC arch/powerpc/boot/fdt_sw.o
BOOTCC arch/powerpc/boot/fdt_wip.o
BOOTCC arch/powerpc/boot/inffast.o
BOOTCC arch/powerpc/boot/inflate.o
BOOTCC arch/powerpc/boot/inftrees.o
BOOTAR arch/powerpc/boot/wrapper.a
WRAP arch/powerpc/boot/zImage.epapr
/home/dev/project/bios_github/a/op-build/output/host/usr/bin/powerpc64le-buildroot-linux-gnu-ld: unknown similate mode: -T
support similate： elf64lppc elf32lppc elf32lppclinux elf32lppcsim elf64ppc elf32ppc elf32ppclinux elf32ppcsim
make[2]: *** [arch/powerpc/boot/zImage.epapr] error 1
make[1]: *** [zImage.epapr] error 2
make[1]:leave dir `/home/dev/project/bios_github/a/op-build/output/build/linux-8adc30a30b88c793c2058e85426f4f94158797ad'
make: *** [/home/dev/project/bios_github/a/op-build/output/build/linux-8adc30a30b88c793c2058e85426f4f94158797ad/.stamp_built] error 2
make:leave dir“/home/dev/project/bios_github/a/op-build/buildroot”

williamspatrick, 2015-12-09:
Can you let us know what distribution / version and architecture you are building on?  This is not an error I recognize. 

@stewart-ibm - probably need your or Joel's help on this. 

Patrick

touchqvfb, 2015-12-14:
Hi spatrick,
          I reinstall my ubuntu14.04 OS,And compile op-build,there are no the error below:
/home/dev/project/bios_github/a/op-build/output/host/usr/bin/powerpc64le-buildroot-linux-gnu-ld: unknown similate mode: -T
support similate： elf64lppc elf32lppc elf32lppclinux elf32lppcsim elf64ppc elf32ppc elf32ppclinux elf32ppcsim
          Maybe It's my OS env issue. Thanks for help.