open-power/op-build #227: https://github.com/open-power/op-build/commit/9ec3164e342b9ec9bfd2b98416085ffe7b60dbf3 broke garisson

stewart-ibm, 2015-08-14:
As of https://github.com/open-power/op-build/commit/9ec3164e342b9ec9bfd2b98416085ffe7b60dbf3 garisson_defconfig no longer builds. It did before this was merged

# merge in any system specific attributes, hostboot attributes
/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/mergexml.sh /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"GARRISON_hb.system.xml" /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types.xml /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/attribute_types_hb.xml /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_merged.xml /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/target_types_hb.xml /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"GARRISON_hb.mrw.xml" > /home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml;
# creating the targeting binary
/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/xmltohb.pl --hb-xml-file=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/temporary_hb.hb.xml --fapi-attributes-xml-file=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/fapiattrs.xml --src-output-dir=none --img-output-dir=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/ --vmm-consts-file=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/vmmconst.h --noshort-enums --bios-xml-file=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/"GARRISON_bios.xml" --bios-schema-file=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/bios.xsd --bios-output-file=/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images/"garrison"_bios_metadata.xml
[FATAL!] Target instance "sys0" cannot override attribute "MSS_INTERLEAVE_ENABLE" unless the attribute has already been defined in the target type inheritance chain.
     1: main::generateTargetingImage(5295)
     2: (369)
package/pkg-generic.mk:177: recipe for target '/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/build/garrison-xml-4410e63e97a388e156082d5bcb0325c0bbf670ad/.stamp_built' failed
make: *** [/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/output/build/garrison-xml-4410e63e97a388e156082d5bcb0325c0bbf670ad/.stamp_built] Error 1
make: Leaving directory '/home/jenkins/workspace/op-build-master-next/defconfig/garrison/label_exp/utopic/buildroot'
Build step 'Execute shell' marked build as failure
Archiving artifacts
Finished: FAILURE

williamspatrick, 2015-08-14:
> As of 9ec3164 garisson_defconfig no longer builds. It did before this was merged

Garrison needs an mrw update.  We had to make changes for memory interleaving and garrison must have gotten missed.

williamspatrick, 2015-08-14:
@rjknight @jazurin @Erich-Hauptli - Not sure who's problem this is.  Can one of you three take it on?

williamspatrick, 2015-08-18:
I believe this is fixed now with #241

closed 2015-08-18 by williamspatrick