open-power/op-build #236: Update Linux kernel tree for petitboot

shenki, 2015-08-17:
As part of the 1.7 release we will track a newer version of Linux, probably 4.1

This is a tracking bug. Please assign it to me

williamspatrick, 2015-08-26:
@shenki - The current version of Github ACLs doesn't allow me to assign you because you aren't in a group that has write-permission to this repository.

williamspatrick, 2015-09-10:
Any update?  I'm thinking 1.8 on this one.

shenki, 2015-09-10:
Yes, let's defer

jk-ozlabs, 2015-09-10:
We are waiting on an IO driver here, which will likely be submitted against a more recent kernel version than 3.17; this could be good reason to update to a 4.x kernel for v1.7...

shenki, 2015-09-10:
The following patches are remaining after rebasing on 4.2.

Patches we carry
-----------------------
drm/ast: Default to 8bpp on big endian
xhci: Use xhci_pci_remove for xhci device shutdown
xhci: do not halt the secondary HCD
net/bnx2x: Add shutdown handler

Upstream in 4.3
---------------------
powerpc/kexec: Reset HILE before kexec_sequence
powerpc/kexec: Reset secondary cpu endianess before kexec

Defconfigs
--------------
powerpc: Add openpower_defconfig
powerpc: Add skiroot defconfig

stewart-ibm, 2015-09-11:
After discussion on drivers and patches needed from them as well as a couple of other open bugs it would be good to have fixed (fixed upstream), we're thinking 4.2 for op-build 1.7 is the best way

shenki, 2015-09-14:
With PR #264 this has now been merged into master-next. Thanks!

closed 2015-09-14 by shenki