open-power/op-build #243: problems building master-next garrison PNOR

jazurin, 2015-08-19:
I'm running this command to build:

op-build garrison_defconfig && op-build hostboot-dirclean hostboot-rebuild openpower-pnor-rebuild

I checked that garrison_defconfig has the correct name for BR2_HOSTBOOT_BINARY_WINK_FILENAME (n1.ref_image.hdr.bin.ecc) I also checked that the file exists in the hostbinaries package (output/build/hostboot-binaries-22470e51b6342f9c307e28b3ea7cdb194ff954d4/) .  

Can you please take a look? Thank you.

Snippet:
8192 bytes (8.2 kB) copied, 3.4171e-05 s, 240 MB/s
ecc --inject /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//hostboot.temp.bin --output /esw/san2/jmazu
rin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//firdata.bin.ecc --p8
dd if=/esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/openpower-pnor.version.txt of=/esw/san2/jmazurin/opbuil
d_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//openpower_version.temp ibs=4K conv=sync
0+1 records in
8+0 records out
4096 bytes (4.1 kB) copied, 0.000602447 s, 6.8 MB/s
cp /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//openpower_version.temp /esw/san2/jmazurin/opbuild_up
date/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_version/openpower-pnor.version.txt
cp /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries//centaur_sbec_pad.img.ecc /esw/san2/jmazurin/opbuild_updat
e/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//
cp /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries//venice_sbe.img.ecc /esw/san2/jmazurin/opbuild_update/op-b
uild/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//
cp /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries//n1.ref_image.hdr.bin.ecc /esw/san2/jmazurin/opbuild_updat
e/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//
**cp: cannot stat `/esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries//n1.ref_image.hdr.bin.ecc': No such file or
directory**
Error running command: cp /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries//n1.ref_image.hdr.bin.ecc /esw/san2
/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_pnor_scratch//. Nonzero return code of (256) returned.
make: *** [/esw/san2/jmazurin/opbuild_update/op-build/output/build/openpower-pnor-ed1682e10526ebd85825427fbf397361bb0e34aa/.stamp_images_installed] Error 1
make: Leaving directory `/esw/san2/jmazurin/opbuild_update/op-build/buildroot'

dcrowell77, 2015-08-19:
Do you have commit 7b13538469e0a66106a17a65cf7230e96673b8d0 from hostboot-binaries?  That is where the n1 data was added.

ploetzma, 2015-08-19:
He states he has commit https://github.com/open-power/hostboot-binaries/commit/22470e51b6342f9c307e28b3ea7cdb194ff954d4 , which is the actual merge of the n1 data.

It's clearly not in the hostboot binaries dir that its trying to grab from
$ ls /esw/san2/jmazurin/opbuild_update/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries
centaur_sbec_pad.img.ecc  cvpd.bin  p8.ref_image.hdr.bin.ecc  venice_sbe.img.ecc

but it is very much so in the output/build/hostboot-binaries dir
$ ls /esw/san2/jmazurin/opbuild_update/op-build/output/build/hostboot-binaries-22470e51b6342f9c307e28b3ea7cdb194ff954d4/
centaur_sbec_pad.img.ecc  cvpd.bin  LICENSE  n1.ref_image.hdr.bin.ecc  naples_sbe.img.ecc  NOTICE  p8.ref_image.hdr.bin.ecc  venice_sbe.img.ecc

Can you try to remove the offending directory and try to rebuild?

rm -rf output/build/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_binaries

There is an issue open from forever ago that these don't get updated properly all the time
https://github.com/open-power/op-build/issues/65

jazurin, 2015-08-19:
Thanks for the responses.   Patrick recommended adding 'hostboot-binaries-rebuild' to the build command, and that did the trick.  The build command I used:
`op-build garrison_defconfig && op-build hostboot-binaries-rebuild hostboot-dirclean hostboot-rebuild openpower-pnor-rebuild`
didn't seem to increase the build time by much, and probably will use it by the default for other systems.   The explanation Patrick gave is this: *"You also need to 'hostboot-binaries-rebuild' because we copy a different WINK file between P8 and P8+ machines."*

closed 2015-08-19 by jazurin

stewart-ibm, 2015-08-19:
As of 049a6c09440672bb061c97ce1d7918e840a47202 for master-next, a totally clean op-build is building for Garrison for me and has done since around 2bd56afb0529b76c06f55e9a34d13a226a72114b (and 26f8ce9ae2ee0ee0a40f7f89f5b79df8e21b0147 most certainly failed)