open-power/op-build #244: V1.5: libz error while building host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4

weiwangudel, 2015-08-19:
Hi, 

I am building v1.5 and encountered the following error while building host-p8-pore-binutils. 

/bin/sh ./libtool --tag=CC   --mode=link gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2  -all-static -o size size.o bucomm.o version.o filemode.o ../bfd/libbfd.la ../libiberty/libiberty.a  -lz 
libtool: link: gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2 -static -o size size.o bucomm.o version.o filemode.o  ../bfd/.libs/libbfd.a ../libiberty/libiberty.a -lz
/usr/bin/ld: cannot find -lz
collect2: ld returned 1 exit status

The LD_LIBRARY_PATH shows that the zlib-1.2.8 is in there. 
echo $LD_LIBRARY_PATH
/g/g99/wang87/zlib-1.2.8:/g/g99/wang87/sblim-sfccommon-install/lib:/g/g99/wang87/zlib-1.2.8

I can fix the above command by MANUALLY adding  -L/g/g99/wang87/zlib-1.2.8 (where libz.so is located) 
/bin/sh ./libtool --tag=CC   --mode=link gcc -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -Werror -g -O2  -all-static -o size size.o bucomm.o version.o filemode.o ../bfd/libbfd.la ../libiberty/libiberty.a  -lz  -L/g/g99/wang87/zlib-1.2.8

How could I fix this without changing the op-build script? How could I fix the issue by changing the op-build script?

I don't remember how I successfully built v1.3, I must have encountered this before. But I am stuck at building v1.5.

Thank you for your help!

Wei

williamspatrick, 2015-08-21:
Wei,

The linker doesn't honor LD_LIBRARY_PATH, only the loader does.

You might be able to modify that package to pass LDFLAGS +=
-L/g/g99/wang87/zlib-1.2.8 but I'm not certain it will work.

I'm really surprised your distro does not have libz installed.  Can you
just install it?

You could also try to add host-zlib to this line in
openpower/package/p8-pore-binutils/p8-pore-binutils.mk:

HOST_P8_PORE_BINUTILS_DEPENDENCIES = host-binutils

Patrick

weiwangudel, 2015-08-21:
Hi Patrick,

The DEPENDENCIES line is already in p8-pore-binutils.mk while I encounter this issue. 

I am working on a locally customized operating system at LLNL. I don't have permission to install zlib into the default install directory.  As you can tell, I installed it to my home directory the zlib-1.2.8. 

I found that I can do "make"  and even "make install" without -lz error, under the directory:
host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4/binutils

However, whenever I do an op-build from the root directory of op-build, the Makefile of binutils that I changed (e.g. adding -L/g/g99/wang87/zlib-1.2.8) is overwritten. I am thinking if the Makefile is not overwritten, the build might proceed. 

In addition, I noticed that zlib-1.2.8 is automatically downloaded, can the libz.so in output/build/zlib-1.2.8 be integrated automatically? 

Thank you!

Wei

williamspatrick, 2015-08-21:
The dependency line is there but did you add host-zlib to it?  This tells op-build to create a host (as opposed to target) version of the package for linking against.

williamspatrick, 2015-08-21:
If host-zlib doesn't work then you could create a patch file for the makefile change you did and add it to openpower/packages/p8-pore-binutils/.  See the packages/hostboot subdirectory for an example if you need.

weiwangudel, 2015-08-21:
I indeed did not add host-zlib. However, after adding host-zlib, I still encounter -lz.
I have written the following patch
commit 49c6339a040d2f9e4529db1fc944b3ed83dc6cb0
Author: Wei Wang <wang87@impetus.llnl.gov>
Date:   Fri Aug 21 09:52:17 2015 -0700

    after patch

diff --git a/output/build/host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4/binutils/Makefile b/output/build/host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4/binutils/Makefile
index 7cb01d8..35cd014 100644
--- a/output/build/host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4/binutils/Makefile
+++ b/output/build/host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4/binutils/Makefile
@@ -296,7 +296,7 @@ INSTALL_SCRIPT = ${INSTALL}
 INSTALL_STRIP_PROGRAM = $(install_sh) -c -s
 INSTOBJEXT = .mo
 LD = ld -m elf_x86_64
-LDFLAGS +=  
+LDFLAGS +=  -L/g/g99/wang87/zlib-1.2.8
 LEX = `if [ -f ../flex/flex ]; then echo ../flex/flex; else echo flex; fi`
 LEXLIB = 
 LEX_OUTPUT_ROOT = lex.yy

Thanks!

weiwangudel, 2015-08-21:
Patch keeps complaining about not being able to find the Makefile...
Close issue: zlib not supported on the customized OS. Should switch to another 64-bit machine to do op-build.

closed 2015-08-21 by weiwangudel

weiwangudel, 2015-08-21:
Successfully builds V1.5 on another 64-bit OS with zlib installed in standard directory /usr/lib64/libz.so. 
Only encountered a few unzip errors (complaining about not gzip type).

weiwangudel, 2015-08-21:
Correction, the machine failed to build v1.5 has libz.so located at /usr/lib64/libz.so as well. Don't know why it complains about "cannot find -lz" in the link stage while building binutils for host-p8-pore-binutils. 
Maybe the presence of a user installed zlib directory confused the building process.

weiwangudel, 2015-12-14:
Hi Patrick!

I finally avoided this "-lz" error when building host-p8-pore-binutils-1d6a41d5a5104ce9bc8b6ad59f8e58c3a50824a4 by manually adding libz path to almost every Makefile.in file of its subdirectories (like binutils, gprof) -- on this particular machine. 

However, I am now caught in the following error related to hostboot:

make[8]: *** [../../../obj/genfiles/plugins/fapiPlatHwpErrParser.H] Segmentation fault (core dumped)
make[8]: *** Deleting file `../../../obj/genfiles/plugins/fapiPlatHwpErrParser.H'
make[8]: *** Waiting for unfinished jobs....
    MAKE       rule GEN
    MAKE       test GEN
make[8]: *** [../../../obj/genfiles/fapiAttributeService.C] Segmentation fault (core dumped)
make[8]: *** Deleting file `../../../obj/genfiles/fapiAttributeService.C'
make[7]: *** [_BUILD/PASSES/GEN/BODY] Error 2
    MAKE       test GEN
make[6]: *** [_BUILD/SUBDIR/GEN/hwpf] Error 2
make[6]: *** Waiting for unfinished jobs....
make[5]: *** [_BUILD/PASSES/GEN/POST] Error 2
make[4]: *** [_BUILD/SUBDIR/GEN/usr] Error 2
make[3]: *** [_BUILD/PASSES/GEN/POST] Error 2
make[2]: *** [_BUILD/SUBDIR/GEN/src] Error 2
make[1]: *** [_BUILD/PASSES/GEN/POST] Error 2
make[1]: Leaving directory `/g/g99/wang87/op-build/output/build/hostboot-c646754e720b5cd21534425ca90bb414a4a3ff12'
make: *** [/g/g99/wang87/op-build/output/build/hostboot-c646754e720b5cd21534425ca90bb414a4a3ff12/.stamp_built] Error 2

I was able to compile successfully only once on this machine. Thereafter, I am faced with -lz problem for a long time (and I switched to other machines to build). And now I am stuck in this hostboot building phase. 

Do you know how to fix this?  

Thank you!

Wei

weiwangudel, 2015-12-14:
Hi Patrick,

Please disregard. It was just due to "Disk Space Quota exceeded". 

Thanks!

Wei