open-power/op-build #248: Update buildroot base

stewart-ibm, 2015-08-20:
Update our base version of buildroot, attempt to get as close to upstream as possible.

May/may not be ready for 1.7.

Current status: boots to petitboot.

stewart-ibm, 2015-09-09:
It looks like we're going to go with 201505 rather than 201508 this time around, with a possible update in v1.8 (we may be having toolchain issues in 201508, not enough time to debug).

I'll work with @shenki and @jk-ozlabs over the next few days to merge the buildroot bits and ensure we can merge for op-build v1.7

williamspatrick, 2015-09-10:
We wanted to release 1.7 on Friday and it is intended to be the
lock-down release for some initial system releases.  I'm not really keen
on completely changing the underlying buildroot two days before.

Prefer to defer to 1.8 entirely unless we have some really important
reasons to do otherwise.

stewart-ibm, 2015-09-10:
Looking at the diff of the package versions that buildroot pulls in between 201405 (current base) and 201505 (proposed) it's relatively minor for the most part:

automake-1.14.1.tar.xz to automake-1.15.tar.xz

Bison from 3.0.2 to 3.04
busybox 1.22.1 to 1.23.1 (misc bug fixes in a variety of tools)
dropbear-2014.63.tar.bz2 to dropbear-2015.67.tar.bz2 (we probably really want this)
ethtool 3.14 to 3.18
eudev 1.7 to 3.0 (Currently testing, but initial signs are that this is fine)
gcc 4.9.0 to 4.9.2 (no doubt a good idea)
glibc 2.19 to 2.20
kmod 17 to 20
libtool 2.4.2 to 2.4.6
lzo 2.06 to 2.08
mpc 1.0.2 to 1.0.3
powerpc-utils 1.4 to 1.2.24 (which looks like a regression but isn't, in this case, 1.2.24 > 1.4)
rsync 3.1.0 to 3.1.1
util-linux 2.24.2 to 2.26.2

So the visible impacts are likely only going to be around petitboot itself (which we're testing pretty hard) and bug fix releases of some of the utilities.

So it's likely a lot safer than it sounds, and jumping now rather than after we freeze or as part of a service pack is likely going to give us a better base to maintain.

stewart-ibm, 2015-09-11:
Pull request is here: https://github.com/open-power/op-build/pull/255

We're good to merge that in and are confident in it.

stewart-ibm, 2015-09-14:
This is done, closing.

closed 2015-09-14 by stewart-ibm