open-power/op-build #307: petitboot reboot prevents ubuntu from booting

jazurin, 2015-11-04:
This has been seen 2 times during OP HW CI since November 2nd (2 times out of 24 boots).  We are booting using op-build master-next code on a habanero. 
It looks like an exception causes petitboot to shutdown. 
For reference here are the SOL logs of each failed test run:
https://gist.github.com/jazurin/82b49415d88842a1e105
https://gist.github.com/jazurin/de1893a90d950a3997fd

The system is going down NOW!

Sent SIGTERM to all processes

Sent SIGKILL to all processes
cpu 0x0: Vector: 300 (Data Access) at [c000000fd13936e0]
    pc: c0000000009cbe74: xhci_irq+0x254/0x17cc
    lr: c0000000009cbc5c: xhci_irq+0x3c/0x17cc
    sp: c000000fd1393960
   msr: 9000000000009033
   dar: 28
 dsisr: 40000000
  current = 0xc000000fc82236c0
  paca    = 0xc00000000fe80000	 softe: 0	 irq_happened: 0x01
    pid   = 4039, comm = kexec
enter ? for help
[c000000fd1393960] 00003fffd38f2758 (unreliable)
[c000000fd1393af0] c00000000099922c usb_hcd_irq+0x44/0x68
[c000000fd1393b10] c0000000009aa4c8 usb_hcd_pci_remove+0x44/0x140
[c000000fd1393b50] c0000000009afeac xhci_pci_remove+0x64/0x8c
[c000000fd1393b80] c0000000003d4bc0 pci_device_shutdown+0x50/0x88
[c000000fd1393bb0] c00000000059ada4 device_shutdown+0x170/0x1e8
[c000000fd1393c50] c000000000072390 kernel_restart_prepare+0x4c/0x60
[c000000fd1393c70] c0000000000b98bc kernel_kexec+0x68/0xbc
[c000000fd1393ce0] c0000000000728f4 SyS_reboot+0x1b4/0x1f4
[c000000fd1393e30] c000000000009198 system_call+0x38/0xd0
--- Exception: c01 (System Call) at 00003fffa1ce1084
SP (3fffd38f1f80) is in userspace
0:mon>

jazurin, 2015-11-12:
We hit this problem again.  Any news on this issue?

Thanks.

 Info: Booting in 8 sec: Ubuntu[24;19H7[24;32H[24;19H6[24;32H[24;19H5[24;32H[24;19H4[24;32H[24;19H3[24;32H[24;19H2[24;32H[24;19H1[24;32H[24;16HUbuntu.         [24;23H[15Drunning boot hooks[18Dperforming kexec_load[5D reboot
The system is going down NOW!

Sent SIGTERM to all processes

Sent SIGKILL to all processes
cpu 0x0: Vector: 300 (Data Access) at [c000000fe6ed76e0]
    pc: c0000000009cbe74: xhci_irq+0x254/0x17cc
    lr: c0000000009cbc5c: xhci_irq+0x3c/0x17cc
    sp: c000000fe6ed7960
   msr: 9000000000009033
   dar: 28
 dsisr: 40000000
  current = 0xc000000fe5221b60
  paca    = 0xc00000000fe80000	 softe: 0	 irq_happened: 0x01
    pid   = 4036, comm = kexec
enter ? for help
[c000000fe6ed7960] 00003fffe4da5ee8 (unreliable)
[c000000fe6ed7af0] c00000000099922c usb_hcd_irq+0x44/0x68
[c000000fe6ed7b10] c0000000009aa4c8 usb_hcd_pci_remove+0x44/0x140
[c000000fe6ed7b50] c0000000009afeac xhci_pci_remove+0x64/0x8c
[c000000fe6ed7b80] c0000000003d4bc0 pci_device_shutdown+0x50/0x88
[c000000fe6ed7bb0] c00000000059ada4 device_shutdown+0x170/0x1e8
[c000000fe6ed7c50] c000000000072390 kernel_restart_prepare+0x4c/0x60
[c000000fe6ed7c70] c0000000000b98bc kernel_kexec+0x68/0xbc
[c000000fe6ed7ce0] c0000000000728f4 SyS_reboot+0x1b4/0x1f4
[c000000fe6ed7e30] c000000000009198 system_call+0x38/0xd0
--- Exception: c01 (System Call) at 00003fff9f631084
SP (3fffe4da5710) is in userspace

shenki, 2015-11-13:
This is running 4.2.5-openpower1. There has been some change in the xhci driver with the stable patches brought in, I suspect that is causing us issues.

ozbenh, 2015-11-13:
Does look like a NULL dereference in xhci... definitely something to report upstream

jazurin, 2015-12-02:
@stewart-ibm what is the latest status on this issue?  We are still seeing this problem.   Are there any workarounds?  Thanks.

stewart-ibm, 2015-12-02:
Are you able to bisect the kernel between the last good revision and this? If it's a regression, we should report that upstream and get it fixed in stable too.

shenki, 2015-12-07:
@jazurin Have you tested a firmware that has the v4.2.6-openpower1 kernel in it? That is our latest.

If you can reproduce there, then we can take another look

zoomtechjwq, 2015-12-08:
I got this error when I compile v1.7 op-build   palmetto_defconfig.What is the reason?
BOOTCC  arch/powerpc/boot/fdt_strerror.o
  BOOTCC  arch/powerpc/boot/fdt_sw.o
  BOOTCC  arch/powerpc/boot/fdt_wip.o
  BOOTCC  arch/powerpc/boot/inffast.o
  BOOTCC  arch/powerpc/boot/inflate.o
  BOOTCC  arch/powerpc/boot/inftrees.o
  BOOTAR  arch/powerpc/boot/wrapper.a
  WRAP    arch/powerpc/boot/zImage.epapr
/home/dev/project/bios_github/c/op-build/output/host/usr/bin/powerpc64le-buildroot-linux-gnu-ld: unknown Simulation mode: -T
support Simulation mode： elf64lppc elf32lppc elf32lppclinux elf32lppcsim elf64ppc elf32ppc elf32ppclinux elf32ppcsim
make[2]: *** [arch/powerpc/boot/zImage.epapr]  error 1
make[1]: *** [zImage.epapr] error 2
make[1]:leave `/home/dev/project/bios_github/c/op-build/output/build/linux-4.2.1'
make: *** [/home/dev/project/bios_github/c/op-build/output/build/linux-4.2.1/.stamp_built] error 2
make:leave “/home/dev/project/bios_github/c/op-build/buildroot”

jazurin, 2015-12-08:
@shenki Here is a fail log with Linux version 4.2.6-openpower1:

https://gist.github.com/jazurin/23ffc974d6e394792de0

jazurin, 2015-12-08:
@stewart-ibm I did not run a specific regression to isolate the kernel that introduced this issue, but we started seeing the problem in early in November, around the time when aaafb39e6b2c18ce665272605d8b9b9fde942d74 was merged.

ozbenh, 2015-12-08:
That's a NULL reference in the USB XHCI driver during kexec. Bloody
thing has been a source of pain forever, that driver sucks.

Stewart / Joel, can you look at what XHCI related changes went in ?

Cheers,
Ben.

shenki, 2016-01-19:
I booted 4.4 (with our patch changing .shutdown in the xhci driver) on a habanero and got this:

```
[   16.989755] xhci_hcd 0001:06:00.0: xHCI Host Controller
[   16.989816] xhci_hcd 0001:06:00.0: new USB bus registered, assigned bus number 1
[   17.020062] xhci_hcd 0001:06:00.0: Host not halted after 16000 microseconds.
[   17.020113] xhci_hcd 0001:06:00.0: can't setup: -110
[   17.020147] xhci_hcd 0001:06:00.0: USB bus 1 deregistered
[   17.020231] xhci_hcd 0001:06:00.0: init 0001:06:00.0 fail, -110
[   17.020276] xhci_hcd: probe of 0001:06:00.0 failed with error -110
```

This is the same as the error that we worked around with http://www.spinics.net/lists/linux-usb/msg104581.html

https://github.com/open-power/linux/commit/9495fb487a5c58fec056ede804ff5036353acce0

We dropped this patch in https://github.com/open-power/linux/commits/v4.2.4-openpower1 as the stable release pulled in some xhci changes that removed the function we were patching.

It looks like we need to add back the "do not halt the secondary HCD" code, but in a way that works with the current driver.

stewart-ibm, 2016-01-21:
We should likely poke the XHCI maintainer about this, perhaps we can improve the situation upstream and stop having such issues...

That being said, I hear we have someone with some deeper USB knowledge inside the LTC, so I'm currently finding who they are and enlisting their help :)

krisman, 2016-01-27:
Hi,

I'm no xhci expert, so I could be terribly wrong, but I think the problem may be the change in the moment when we halt the HCD, introduced by  8c24d6d7b09 (" xhci: stop everything on the first call to xhci_stop"), which removed Thadeu's fix from the OpenPower tree. Thadeu's patch only halted the hardware later, when invoking xhci_stop for the primary HCD, after calling the interrupt handler in usb_hcd_pci_remove. On the other hand, 8c24d6d7b09 stops everything at the first call to xhci_stop, which is always for the secondary HCD when executing the .remove handler.  This problem is strange, since apparently it doesn't oops any other time we poke the .remove handler on stock kernels, but still, it's worth investigating.  I'm trying to reproduce it elsewhere.

With this in mind if, for any reason, we step over the  STS_EINT check of the controller status register, in xhci_irq, it would explain both the failure and the Oops, depending on the actual value of the status register.  For instance, with the current patch, if for any reason we get STS_FATAL  during the xhci_irq, we'd try to halt the device again, provoking the timeout, since it was already halted before.  I don't think this can happen with Thadeu's version of the fix.

I'm working on reviewing the behavior of the status register to confirm this hypotesis.  It would be very helpful if we could enable dynamic debug in xhci-hcd,  add some tracing code in the xhci_irq and attempt to reproduce, to confirm whether the status register is giving us something unexpected during the call to xhci_irq.  Also, dump_stack() in xhci_halt would allow us to confirm that the timeout is coming from the xhci_irq call itself.  Since I'm having a hard time reproducing it myself without a fw rebuild, can I get some help from you guys to confirm this ideas?  Or point me to instructions on building and flashing a palmetto system? :)

If I am correct, reverting 8c24d6d7b09 and readding Thadeu's fix might be the simplest workaround for this, until we properly fix it upstream.

shenki, 2016-01-28:
Thanks for the suggestions! I verified that reverting 8c24d6d7b09 and re-adding Thadeu's fix works around the issue for now.

I built a kernel without the fixes but with dump_stack() in xhci_halt as you suggested. This is what I see when kexecing from that kernel back into itself again on a habanero:

```
[   17.515864] xhci_hcd 0001:06:00.0: xHCI Host Controller
[   17.515931] xhci_hcd 0001:06:00.0: new USB bus registered, assigned bus number 1
[   17.516005] CPU: 52 PID: 1 Comm: swapper/0 Not tainted 4.4.0-openpower2unreleased1 #2
[   17.516058] Call Trace:
[   17.516079] [c000000ffc503640] [c000000000b73210] dump_stack+0x90/0xb0 (unreliable)
[   17.516140] [c000000ffc503670] [c0000000009e3b98] xhci_halt+0x44/0xa8
[   17.516190] [c000000ffc5036e0] [c0000000009e4c80] xhci_gen_setup+0x354/0x47c
[   17.516241] [c000000ffc503770] [c0000000009f3dc0] xhci_pci_setup+0x70/0xa8
[   17.516295] [c000000ffc5037a0] [c0000000009be990] usb_add_hcd+0x2dc/0x8a8
[   17.516346] [c000000ffc503860] [c0000000009ce1ec] usb_hcd_pci_probe+0x2c8/0x3cc
[   17.516408] [c000000ffc503900] [c0000000009f3958] xhci_pci_probe+0x20/0xfc
[   17.516460] [c000000ffc503940] [c0000000003e1390] pci_device_probe+0x8c/0xfc
[   17.516514] [c000000ffc5039c0] [c0000000005aa65c] driver_probe_device+0x134/0x2b8
[   17.516575] [c000000ffc503a50] [c0000000005aa870] __driver_attach+0x90/0xc0
[   17.516629] [c000000ffc503a90] [c0000000005a8644] bus_for_each_dev+0xa4/0xb8
[   17.516681] [c000000ffc503ae0] [c0000000005aa124] driver_attach+0x2c/0x40
[   17.516731] [c000000ffc503b00] [c0000000005a9c20] bus_add_driver+0x100/0x240
[   17.516784] [c000000ffc503b90] [c0000000005ab10c] driver_register+0xc0/0x110
[   17.516835] [c000000ffc503c10] [c0000000003e0aa8] __pci_register_driver+0x4c/0x60
[   17.516897] [c000000ffc503c30] [c00000000108d79c] xhci_pci_init+0x44/0x58
[   17.516949] [c000000ffc503c50] [c00000000000ade8] do_one_initcall+0x138/0x1d4
[   17.517000] [c000000ffc503d20] [c000000001063d50] kernel_init_freeable+0x190/0x24c
[   17.517059] [c000000ffc503dc0] [c00000000000b510] kernel_init+0x24/0x114
[   17.517110] [c000000ffc503e30] [c0000000000094b4] ret_from_kernel_thread+0x5c/0xa8
[   17.547386] xhci_hcd 0001:06:00.0: Host not halted after 16000 microseconds.
[   17.547437] xhci_hcd 0001:06:00.0: can't setup: -110
[   17.547473] xhci_hcd 0001:06:00.0: USB bus 1 deregistered
[   17.547556] xhci_hcd 0001:06:00.0: init 0001:06:00.0 fail, -110
[   17.547602] xhci_hcd: probe of 0001:06:00.0 failed with error -110
```

shenki, 2016-01-28:
If you want to play with kernels on a machine yourself, try the instructions on @jk-ozlabs's blog:

 http://jk.ozlabs.org/blog/post/160/custom-kernels-openpower-firmware/

There are some scripts that automate transferring to the BMC here: 

 https://github.com/open-power/skiboot/tree/master/external/boot-tests

You don't need to use them, but they do have some nice hints for what tools to use, etc.

shenki, 2016-02-19:
@krisman hey, have you had a chance to take a look at this?

We've gone ahead and made the changes I mentioned in our kernel, but we would like to find a solution that is likely to be accepted by upstream

krisman, 2016-02-29:
Hi @shenki I am a bit overloaded right now with other releases.  This is in my todo list for the next weeks but I don't think I'll be able to get back to it before 3/15 or so.

shenki, 2016-03-15:
This is resolved with the xhci patches we carry.

We can move this to 1.9 to track the creation of a solution suitable for upstreaming.

shenki, 2016-04-18:
Hey @krisman, any chance you can find some time to take a look at this?

krisman, 2016-04-18:
Hi @shenki, I took some time to work on this at the end of last week. I have a patch ready for upstream, which should be sent out to the list tomorrow morning.  Can you help me verify this on OPAL?  I'll put you in Cc.

shenki, 2016-04-18:
On Tue, Apr 19, 2016 at 7:42 AM, krisman <notifications@github.com> wrote:

> Hi @shenki <https://github.com/shenki>, I took some time to work on this
> at the end of last week. I have a patch ready for upstream, which should be
> sent out to the list tomorrow morning. Can you help me verify this on
> OPAL? I'll put you in Cc.
>
>
Great news! Thanks for your help.

Please do cc me, and I'll do some testing on the machines we have in our
lab. joel@jms.id.au

shenki, 2016-04-27:
@krisman has posted a patch, and there has been some feedback on the list. We should review and test the patches

http://www.spinics.net/lists/linux-usb/msg139687.html
http://www.spinics.net/lists/linux-usb/msg139616.html

stewart-ibm, 2016-04-27:
From the target date for op-bulid v1.9, it looks like we'll be sticking with our own patches for 1.9 and looking at the upstream ones for v1.10, so changing milestone accordingly.

krisman, 2016-05-02:
xhci maintainer has accepted the patch upstream and queued it up for 4.7
<http://www.spinics.net/lists/linux-usb/msg139687.html>

@shenki, did you get a chance to try it out?

shenki, 2016-05-05:
@krisman I did test it, and I thought that it resolved the issue. But I've given it another go and I don't see any usb in the host after kexec. 

It looks like I was wrong.

```
[    0.937375] xhci_hcd 0001:05:00.0: xHCI Host Controller
[    0.937434] xhci_hcd 0001:05:00.0: new USB bus registered, assigned bus number 1
[    0.967769] xhci_hcd 0001:05:00.0: Host not halted after 16000 microseconds.
[    0.967923] xhci_hcd 0001:05:00.0: can't setup: -110
[    0.968029] xhci_hcd 0001:05:00.0: USB bus 1 deregistered
[    0.968148] xhci_hcd 0001:05:00.0: init 0001:05:00.0 fail, -110
[    0.968268] xhci_hcd: probe of 0001:05:00.0 failed with error -110
```

This is with your patch applied to 4.4.9 in petitboot, and then kexecing into Ubuntu 3.19.0-24-generic.

krisman, 2016-05-05:
Hmm, I see. This  is another issue, not related to the oops itself.  The problem is another call to xhci_halt in the probe path, which Thadeu's patch also avoided.    I think the best way to work around this is a specific quirk for the TI device, since they are the one's that can't handle duplicated halted commands, apparently.  I think I can reproduce this with a distro kernel.  Let me give it a try.

shenki, 2016-05-09:
@krisman I made the mistake of merging a broken kernel into our op-build release, and I need to provide a fix so we can tag master.

Do you have a suggested patch that I could try?

If not, we will go back to Thadeu's patch and try again for the next release. Thanks for your help!

krisman, 2016-05-10:
@shenki, not yet.  I'll work on it tomorrow morning.

sammj, 2016-05-26:
Copying a comment from an IBMer who has seen a similar but not quite the same issue on both FSP- and BMC-based machines:

> This is the box that looses its USB connections after a single boot.
> After a boot you have to move the USB device to a different slot to get it
> to recognize.
> If you try to move an existing USB device from slot 1 to slot 2 while the
> system is up it will not recognize the  move.

stewart-ibm, 2016-07-08:
How are we looking on this? Is this likely to be fixed in the next week?

stewart-ibm, 2016-07-08:
(i.e. should this be v1.10 or v1.11 ?)

shenki, 2016-07-08:
@krisman let me know you have time to look at a fix

stewart-ibm, 2016-08-30:
@krisman ping?
It's been nearly *FOUR* months since your last update.

shenki, 2016-09-19:
Started a chat with upstream about this: https://lkml.org/lkml/2016/9/19/68

shenki, 2016-12-02:
Recent debugging of the xhci issue revealed that our current behaviour is buggy.
```
[  942.272194] xhci_hcd 0003:09:00.0: remove, state 1
[  942.272346] usb usb2: USB disconnect, device number 1
[  942.272535] xhci_hcd 0003:09:00.0: USB bus 2 deregistered
[  942.272683] xhci_hcd 0003:09:00.0: remove, state 1
[  942.272719] usb usb1: USB disconnect, device number 1
[  942.272753] usb 1-3: USB disconnect, device number 2
[  942.272786] usb 1-3.1: USB disconnect, device number 3
[  942.430268] usb 1-3.2: USB disconnect, device number 4
[  942.610280] usb 1-3.3: USB disconnect, device number 5
[  942.830258] usb 1-3.4: USB disconnect, device number 6
[  943.109942] xhci_hcd 0003:09:00.0: Host not halted after 16000 microseconds.
[  943.110176] xhci_hcd 0003:09:00.0: Host controller not halted, aborting reset.
[  943.110439] xhci_hcd 0003:09:00.0: USB bus 1 deregistered
```

stewart-ibm, 2016-12-22:
Do we have something sane in the tree for 1.13 or do we need something else?

shenki, 2016-12-22:
It works as well as it always has. In fact, we have the patch from https://github.com/open-power/op-build/issues/775 which makes it work slightly better.

We still don't have a solution upstream though.

stewart-ibm, 2017-07-10:
Moving to 1.20... maybe we'll have something upstream by then :(

shenki, 2017-07-10:
This is fixed in 4.12

stewart-ibm, 2017-07-10:
AWESOME

PR #339 (2016-02-08): Release 4.4-openpower2 kernel
 - Updated defconfig for 4.4
 - Adds EEH PE patch
 - Restores xhci driver workarounds for USB bugs
 - aacraid driver is now usptream
 - Powerpc kexec patches are upstream

Tested on palmetto and habanero systems.

This fixes #307 and #329.
closed 2017-07-10 by stewart-ibm