open-power/op-build #325: some error about fsi and cvpd

touchqvfb, 2015-12-25:
Hi  all,
        I boot op-build 1.7 palmetto,then got error message below. there is some issue with the fsi and cvpd,Can you give me some detail info about the function and thory of fsi. I boot op-build1.6 is all ok on our mathine.
5.40831|================================================
  5.40832|Error reported by fsi (0x0A00)
  5.42393|  presenceDetect> FSI and CVPD do not agree
  5.42394|  ModuleId   0x21 FSI::MOD_FSIPRES_MEMBPRESENCEDETECT
  5.42394|  ReasonCode 0x0a10 FSI::RC_FSI_CVPD_MISMATCH
  5.42395|  UserData1  HUID of membuffer : 0x0000000000040000
  5.42395|  UserData2  FSI Presence : 0x0000000100000000
  5.42395|User Data Section 0, type UD
  5.42396|  Subsection type 0x04
  5.42396|  ComponentId errl (0x0100)
  5.42397|User Data Section 1, type UD
  5.42397|  Subsection type 0x06
  5.42397|  ComponentId errl (0x0100)
  5.42397|  CALLOUT
  5.42398|  HW CALLOUT
  5.42398|  Reporting CPU ID: 8
  5.42398|  Called out entity:
  5.42399|User Data Section 2, type UD
  5.42399|  Subsection type 0x02
  5.42399|  ComponentId errl (0x0100)
  5.42400|  TARGET
  5.42400|  HUID: 00040000
  5.42402|  Phys path: Physical:/Sys0/Node0/Membuf1
  5.42403|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS1/Membuf1
  5.42403|User Data Section 3, type UD
  5.42403|  Subsection type 0x01
  5.42404|  ComponentId fsi (0x0a00)
  5.42404|  STRING
  5.42404|  
  5.42405|User Data Section 4, type UD
  5.42405|  Subsection type 0x15
  5.42406|  ComponentId hb-trace (0x3100)
  5.42406|User Data Section 5, type UD
  5.42406|  Subsection type 0x15
  5.42407|  ComponentId hb-trace (0x3100)
  5.42407|User Data Section 6, type UD
  5.42408|  Subsection type 0x03
  5.42408|  ComponentId errl (0x0100)
  5.42408|User Data Section 7, type UD
  5.42409|  Subsection type 0x01
  5.42409|  ComponentId errl (0x0100)
  5.42409|  STRING
  5.42410|  Hostboot Build ID: 
  5.42410|User Data Section 8, type UD
  5.42410|  Subsection type 0x04
  5.42411|  ComponentId errl (0x0100)
  5.42411|================================================
 11.75839|HWAS|PRESENT> Membuf[04]=4000000000000000
 11.75839|HWAS|PRESENT> Proc[05]=8000000000000000
 18.38766|ISTEP  6. 6
 18.46377|System shutting down with error status 0xC04
 18.47266|================================================
 18.47267|Error reported by hwas (0x0C00)
 18.48215|  checkMinimumHardware found no functional dimm cards.
 18.48216|  ModuleId   0x03 MOD_CHECK_MIN_HW
 18.48216|  ReasonCode 0x0c06 RC_SYSAVAIL_NO_MEMORY_FUNC
 18.48216|  UserData1  HUID of node : 0x0002000000000000
 18.48217|  UserData2  number of present, non-functional dimms : 0x0000000000000000
 18.48217|User Data Section 0, type UD
 18.48217|  Subsection type 0x15
 18.48218|  ComponentId hb-trace (0x3100)
 18.48218|User Data Section 1, type UD
 18.48218|  Subsection type 0x06
 18.48219|  ComponentId errl (0x0100)
 18.48219|  CALLOUT
 18.48219|  PROCEDURE ERROR
 18.48219|  Procedure: 1
 18.48219|User Data Section 2, type UD
 18.48220|  Subsection type 0x03
 18.48220|  ComponentId errl (0x0100)
 18.48220|User Data Section 3, type UD
 18.48221|  Subsection type 0x01
 18.48221|  ComponentId errl (0x0100)
 18.48221|  STRING
 18.48221|  Hostboot Build ID: 
 18.48222|================================================
 18.89063|================================================
 18.89063|Error reported by hwas (0x0C00)
 18.89063|  checkMinimumHardware found no functional membufs
 18.89063|  ModuleId   0x03 MOD_CHECK_MIN_HW
 18.89064|  ReasonCode 0x0c09 RC_SYSAVAIL_NO_MEMBUFS_FUNC
 18.89064|  UserData1  HUID of node : 0x0002000000000000
 18.89065|  UserData2  number of present nonfunctional membufs : 0x0000000100000000
 18.89065|User Data Section 0, type UD
 18.89065|  Subsection type 0x15
 18.89065|  ComponentId hb-trace (0x3100)
 18.89066|User Data Section 1, type UD
 18.89066|  Subsection type 0x06
 18.89066|  ComponentId errl (0x0100)
 18.89066|  CALLOUT
 18.89067|  PROCEDURE ERROR
 18.89067|  Procedure: 1
 18.89067|User Data Section 2, type UD
 18.89067|  Subsection type 0x03
 18.89068|  ComponentId errl (0x0100)
 18.89068|User Data Section 3, type UD
 18.89068|  Subsection type 0x01
 18.89068|  ComponentId errl (0x0100)
 18.89069|  STRING
 18.89069|  Hostboot Build ID: 
 18.89069|================================================
 19.19235|================================================
 19.19236|Error reported by hwas (0x0C00)
 19.19236|  Insufficient hardware to continue.
 19.19236|  ModuleId   0x03 MOD_CHECK_MIN_HW
 19.19236|  ReasonCode 0x0c04 RC_SYSAVAIL_INSUFFICIENT_HW
 19.19237|  UserData1   : 0x0000000000000000
 19.19237|  UserData2   : 0x0000000000000000
 19.19237|User Data Section 0, type UD
 19.19237|  Subsection type 0x15
 19.19238|  ComponentId hb-trace (0x3100)
 19.19238|User Data Section 1, type UD
 19.19238|  Subsection type 0x06
 19.19239|  ComponentId errl (0x0100)
 19.19239|  CALLOUT
 19.19239|  PROCEDURE ERROR
 19.19239|  Procedure: 1
 19.19240|User Data Section 2, type UD
 19.19240|  Subsection type 0x01
 19.19240|  ComponentId errl (0x0100)
 19.19241|  STRING
 19.19241|  host_gard
 19.19241|User Data Section 3, type UD
 19.19242|  Subsection type 0x03
 19.19242|  ComponentId errl (0x0100)
 19.19243|User Data Section 4, type UD
 19.19243|  Subsection type 0x01
 19.19243|  ComponentId errl (0x0100)
 19.19244|  STRING
 19.19244|  Hostboot Build ID: 
 19.19245|================================================

dcrowell77, 2016-01-04:
FSI is a custom support interface created by IBM to provide sideband access to the chips.  I think there is a spec out in the openpower.org site somewhere if you care about the hardware details.  

However, FSI is not the issue in this case.  What I see is that we are detecting the Centaur chip just fine (via FSI) but we cannot find any VPD for it.  The VPD for the Centaur is part of the planar vpd image and it gets cached into the CVPD section of PNOR during the first boot after a code update or hardware replacement.  If there are no errors above this related to vpd or i2c, then you may just have a bad planar vpd image on your board.  I'm a bit surprised that v1.6 would work though.  How old is your system?  We haven't seen any issues on our Palmettos with this level.

williamspatrick, 2016-01-05:
The "pass 1" Palmetto systems we had early on didn't have the system / centaur
VPD chip, but none of those were shipped to customers from Tyan.  We
should still support building an image for those systems but it requires
making changes to the Hostboot config in op-build.

I believe changing these to 'unset' in
openpower/configs/hostboot/palmetto.config was the setting:

    set CVPD_READ_FROM_HW
    set CVPD_WRITE_TO_HW
    set PVPD_READ_FROM_HW
    set PVPD_WRITE_TO_HW

touchqvfb, 2016-01-06:
Hi spatrick ,
        I 'unset' below option.Then it show the error below,after about 15s it reboot again.why?Thanks very must for reply.   4.67258|HWAS|PRESENT> Membuf[04]=4000000000000000 It mean hostboot cannot detact my centaur and dimm.

    unset CVPD_READ_FROM_HW
    unset CVPD_WRITE_TO_HW
    un set PVPD_READ_FROM_HW
    unset PVPD_WRITE_TO_HW

1.94108|ERRL|Dumping errors reported prior to registration
  3.71698|Ignoring boot flags, incorrect version 0x0
  4.06823|ISTEP  6. 3
  4.50249|ISTEP  6. 4
  4.50305|ISTEP  6. 5
  4.67258|HWAS|PRESENT> Membuf[04]=4000000000000000
  4.67259|HWAS|PRESENT> Proc[05]=8000000000000000
  4.74912|================================================
  4.74913|Error reported by vpd (0x1D00)
  4.75885|  The requested record was not found in the VPD TOC.
  4.75886|  ModuleId   0x22 VPD::VPD_IPVPD_FIND_RECORD_OFFSET
  4.75886|  ReasonCode 0x1d03 VPD::VPD_RECORD_NOT_FOUND
  4.75886|  UserData1  Requested Record : 0x0000000000000002
  4.75887|  UserData2  Requested Keyword : 0x0000000000000015
  4.75887|User Data Section 0, type UD
  4.75887|  Subsection type 0x06
  4.75887|  ComponentId errl (0x0100)
  4.75888|  CALLOUT
  4.75888|  HW CALLOUT
  4.75888|  Reporting CPU ID: 8
  4.75888|  Called out entity:
  4.75889|User Data Section 1, type UD
  4.75889|  Subsection type 0x06
  4.75889|  ComponentId errl (0x0100)
  4.75889|  CALLOUT
  4.75890|  PROCEDURE ERROR
  4.75890|  Procedure: 4
  4.75890|User Data Section 2, type UD
  4.75890|  Subsection type 0x06
  4.75891|  ComponentId errl (0x0100)
  4.75891|  CALLOUT
  4.75891|  PROCEDURE ERROR
  4.75891|  Procedure: 85
  4.75892|User Data Section 3, type UD
  4.75892|  Subsection type 0x15
  4.75892|  ComponentId hb-trace (0x3100)
  4.75893|User Data Section 4, type UD
  4.75893|  Subsection type 0x01
  4.75893|  ComponentId vpd (0x1d00)
  4.75894|  STRING
  4.75894|  
  4.75894|User Data Section 5, type UD
  4.75894|  Subsection type 0x15
  4.75895|  ComponentId hb-trace (0x3100)
  4.75895|User Data Section 6, type UD
  4.75895|  Subsection type 0x03
  4.75895|  ComponentId errl (0x0100)
  4.75896|User Data Section 7, type UD
  4.75896|  Subsection type 0x01
  4.75896|  ComponentId errl (0x0100)
  4.75897|  STRING
  4.75897|  Hostboot Build ID: 
  4.75897|================================================
 11.65139|System Shutting Down In 15 Seconds To Perform Reconfiguration

dcrowell77, 2016-01-06:
Can you confirm what pass of hardware you are using?  The new error is coming from the fact that there is no data in the pnor for the CVPD.  It would really be better if you weren't using down-level hardware.

touchqvfb, 2016-01-07:
I insert  2 POWER8 CPU,8 centaur and 32 dimm,  or  1 CPU,4centaur,16 dimm  into Our machine  .op-build can only recongnize 1cpu,1centaur,2dimm. From the IPL log,op-build 1.7 (palmetto_defconfig) cannot recognize all of the centaur and dimm. Why there is no data in the pnor for the CVPD? How can I put the VPD data into PNOR? What is VPD_RECORD_NOT_FOUND mean?TKS!
  6.00773|HWAS|PRESENT> DIMM[03]=00AA000000000000
  6.00774|HWAS|PRESENT> Membuf[04]=4000000000000000
  6.00774|HWAS|PRESENT> Proc[05]=8000000000000000

1.94986|ERRL|Dumping errors reported prior to registration
  3.74927|Ignoring boot flags, incorrect version 0x0
  4.29299|ISTEP  6. 3
  4.72718|ISTEP  6. 4
  4.72768|ISTEP  6. 5
  6.00773|HWAS|PRESENT> DIMM[03]=00AA000000000000
  6.00774|HWAS|PRESENT> Membuf[04]=4000000000000000
  6.00774|HWAS|PRESENT> Proc[05]=8000000000000000
  9.98319|ISTEP  6. 6
 10.06810|ISTEP  6. 7
...
 17.88480|ISTEP 12. 1
 17.96897|ISTEP 12. 2
 18.03453|ISTEP 12. 3
 18.06290|ISTEP 12. 4
 18.14881|System shutting down with error status 0x9000002A
 18.17661|================================================
 18.17661|Error reported by vpd (0x1D00)
 18.18613|  The requested record was not found in the VPD TOC.
 18.18613|  ModuleId   0x22 VPD::VPD_IPVPD_FIND_RECORD_OFFSET
 18.18613|  ReasonCode 0x1d03 VPD::VPD_RECORD_NOT_FOUND
 18.18613|  UserData1  Requested Record : 0x0000000000000007
 18.18614|  UserData2  Requested Keyword : 0x000000000000001b
 18.18614|User Data Section 0, type UD
 18.18614|  Subsection type 0x04
 18.18615|  ComponentId errl (0x0100)
 18.18615|User Data Section 1, type UD
 18.18615|  Subsection type 0x06
 18.18615|  ComponentId errl (0x0100)
 18.18616|  CALLOUT
 18.18616|  HW CALLOUT
 18.18616|  Reporting CPU ID: 15
 18.18616|  Called out entity:
 18.18617|User Data Section 2, type UD
 18.18617|  Subsection type 0x06
 18.18617|  ComponentId errl (0x0100)
 18.18617|  CALLOUT
 18.18618|  PROCEDURE ERROR
 18.18618|  Procedure: 4
 18.18618|User Data Section 3, type UD
 18.18618|  Subsection type 0x06
 18.18619|  ComponentId errl (0x0100)
 18.18619|  CALLOUT
 18.18619|  PROCEDURE ERROR
 18.18619|  Procedure: 85
 18.18620|User Data Section 4, type UD
 18.18620|  Subsection type 0x15
 18.18620|  ComponentId hb-trace (0x3100)
 18.18620|User Data Section 5, type UD
 18.18621|  Subsection type 0x01
 18.18621|  ComponentId vpd (0x1d00)
 18.18621|  STRING
 18.18621|  
 18.18622|User Data Section 6, type UD
 18.18622|  Subsection type 0x15
 18.18622|  ComponentId hb-trace (0x3100)
 18.18622|User Data Section 7, type UD
 18.18623|  Subsection type 0x15
 18.18623|  ComponentId hb-trace (0x3100)
 18.18623|User Data Section 8, type UD
 18.18624|  Subsection type 0x15
 18.18624|  ComponentId hb-trace (0x3100)
 18.18624|User Data Section 9, type UD
 18.18624|  Subsection type 0x02
 18.18625|  ComponentId errl (0x0100)
[... 95 lines cut ...]
 24.27308|  Subsection type 0x03
 24.27308|  ComponentId hwpf (0x0900)
 24.27309|User Data Section 1, type UD
 24.27309|  Subsection type 0x03
 24.27309|  ComponentId hwpf (0x0900)
 24.27309|User Data Section 2, type UD
 24.27310|  Subsection type 0x01
 24.27310|  ComponentId errl (0x0100)
 24.27310|  STRING
 24.27310|  mss_eff_config
 24.27310|User Data Section 3, type UD
 24.27311|  Subsection type 0x06
 24.27311|  ComponentId errl (0x0100)
 24.27311|  CALLOUT
 24.27311|  PROCEDURE ERROR
 24.27312|  Procedure: 85
 24.27312|User Data Section 4, type UD
 24.27312|  Subsection type 0x03
 24.27312|  ComponentId errl (0x0100)
 24.27313|User Data Section 5, type UD
 24.27313|  Subsection type 0x01
 24.27313|  ComponentId errl (0x0100)
 24.27313|  STRING
 24.27314|  Hostboot Build ID: 
 24.27314|================================================

williamspatrick, 2016-01-07:
What machine is this?  You said palmetto_defconfig but then 2 CPUs. Palmetto is a single socket config. 

My previous remarks about the palmetto.config were maybe an incorrect assumption that you had a palmetto machine, because the symptoms you had were the same as a pass1 palmetto with the standard palmetto.config file. 

Patrick

touchqvfb, 2016-01-08:
Our machine have two socket, It can use 1 power8 CPU, 4centuar,16dimm Or 2 CPU,8centuar,32dimm.I compile firestone_defconfig,but It cannot run,and no any IPL info print.So I try palmetto_defconfig ,and use single CPU. But it seem that it cannot recognize all of centuar and dimm(the HW is OK).

williamspatrick, 2016-01-08:
When you say "our machine", does this mean a custom machine that is not palmetto, habanero, or Firestone?  If so, you won't be able to use this repository as is; there is customization that the ODM is responsible for. One of the repositories used to build the PNOR is, for example, Firestone-xml. This repository tells Hostboot information like where to find each chip and other board wiring information. If you give the wrong information the code won't work. 

Sorry, I don't recognize which company you are working with, but the IBM AE team in China has been doing most of the work to assist ODMs with this customization effort. If you do not already have a contact, please send me an email and I will get you in touch with the right people.  

Patrick

cooktail, 2016-03-01:
I rebuild the op-build1.7 for new board.Then I  fix the issue.thanks all.

williamspatrick, 2016-08-31:
No response from originator in months.

closed 2016-08-31 by williamspatrick