open-power/op-build #327: aboud 32MB firestone PNOR

touchqvfb, 2016-01-06:
Hi all,
       How can I use  firestone_defconfig  to compile a 32MB output/images/firestone.pnor,The pnor flash of Our mathine is 32MB.thanks!

ploetzma, 2016-01-06:
Anytime you build, you should get a 32mb pnor in the op-build/output/images directory 

-rwxr-xr-x 1 maploetz maploetz  67108864 Dec  3 16:40 firestone.pnor
-rwxr-xr-x 1 maploetz maploetz  33554432 Dec  3 16:40 firestone_update.pnor

The firestone_update.pnor is a 32MB image, that does not contain a golden side.

williamspatrick, 2016-01-06:
How did you end up with a 32MB pnor?  None of the Firestone machines were built that way and it is not a configuration we have tested at all. Can you get a 64MB chip to replace?

Patrick

dcrowell77, 2016-01-06:
Just using firestone_update.pnor won't be enough, the Hostboot code needs to know the flash size as well.  Look at the palmetto files for our only example of a 32MB flash.  There is a unique pnor layout xml and changes to the hostboot config file.

touchqvfb, 2016-01-07:
I cannot boot firestone_update.pnor in 32MB pnor flash,there is no any log message print.

dcrowell77, 2016-01-07:
See my previous message, on its own firestone_update.pnor will not work if you truly have a 32MB pnor chip.  It seems like you are running on an unreleased board, is that true?  If so, you need to create your own tree in op-build with configuration info specific to your hardware.  If you have a 32MB chip, use the palmetto information as a reference for settings related to pnor.  However, if you have multiple sockets you'll need to use habanero or firestone as a starting point for the system layout.

stewart-ibm, 2016-01-08:
If your machine isn't a firestone, you must *not* pretend it is - *especially* in the device tree.

touchqvfb, 2016-01-11:
I use defaultPnorLayoutSingleSide.xml,then get a 32MB firestone.pnor.  How can I modify  and get FIRESTONE_HB.targeting.bin ?

dcrowell77, 2016-01-11:
The XXX_HB.targeting.bin files are generated from the XXX*.xml files from <op-build>/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openpower_mrw_scratch/.  Those files in turn come from the xxx-xml package associated with your machine.  If you are running a custom machine, you need to start from the appropriate reference platform (palmetto, habanero, firestone) and make adjustments as needed using the serverwiz2 tool out in github.  As mentioned elsewhere, please work with our AEs to get you on the right track.

Once you have a valid set of xml/targeting data for your machine, the only settings you need to change for our code to support a 32MB pnor image is to modify these entries in your xxx.config file.
   set PNOR_IS_32MB
   unset PNOR_TWO_SIDE_SUPPORT

You also need to make sure your openpower/configs/xxx_defconfig is pointing to the single sided pnor layout.
BR2_OPENPOWER_PNOR_XML_LAYOUT_FILENAME="defaultPnorLayoutSingleSide.xml"

touchqvfb, 2016-01-13:
I  unset SET_NOMINAL_PSTATE in openpower/configs/firestone.config,and change nothing else.Then firestone_defconfig have IPL log print. Can I unset SET_NOMINAL_PSTATE (palmetto_defconfig is unset)? thanks!

dcrowell77, 2016-01-13:
SET_NOMINAL_PSTATE has nothing to do with PNOR so I have no idea why that would have helped your situation.  That flag affects how we configure the processor cores before we start the payload.  There is no issue with unsetting it for boot, but it will affect runtime performance.

touchqvfb, 2016-01-14:
Hmm，I don't know too.The op-build1.7 can run to ISTEP 12. 4,then show VPD_RECORD_NOT_FOUND and stop.I don't insert the VPD flash,How can I skip checking VPD?I don't need VPD,when I try to bring up op-build1.7.
18.03453|ISTEP 12. 3
18.06290|ISTEP 12. 4
23.24408|Error reported by vpd (0x1D00)
23.24408| The requested record was not found in the VPD TOC.
23.24409| ModuleId 0x22 VPD::VPD_IPVPD_FIND_RECORD_OFFSET
23.24409| ReasonCode 0x1d03 VPD::VPD_RECORD_NOT_FOUND
23.24409| UserData1 Requested Record : 0x0000000000000007
23.24410| UserData2 Requested Keyword : 0x000000000000001b
23.24410|User Data Section 0, type UD
23.24410| Subsection type 0x04
23.24410| ComponentId errl (0x0100)
23.24411|User Data Section 1, type UD
23.24411| Subsection type 0x06
23.24411| ComponentId errl (0x0100)
23.24411| CALLOUT
23.24411| HW CALLOUT
23.24411| Reporting CPU ID: 15
23.24412| Called out entity:
23.24412|User Data Section 2, type UD
23.24412| Subsection type 0x06
23.24413| ComponentId errl (0x0100)
23.24413| CALLOUT
23.24413| PROCEDURE ERROR
23.24413| Procedure: 4
23.24413|User Data Section 3, type UD
23.24414| Subsection type 0x06
23.24414| ComponentId errl (0x0100)
23.24414| CALLOUT
23.24414| PROCEDURE ERROR
23.24415| Procedure: 85
23.24415|User Data Section 4, type UD
23.24415| Subsection type 0x15
23.24415| ComponentId hb-trace (0x3100)
23.24416|User Data Section 5, type UD
23.24416| Subsection type 0x01
23.24416| ComponentId vpd (0x1d00)
23.24416| STRING

williamspatrick, 2016-01-14:
Can you upload a copy of your defconfig and hostboot config file?

Hostboot needs VPD for all of the chips. For Palmetto, we have a way to use a file VPD for the Centaur, but all other systems / chips we use a VPD chip. 

Patrick

touchqvfb, 2016-01-14:
Because firestone_defconfig have log print,I bring up op-build1.7 base it now. Because VPD of every machine  is defference,it will perplex the Workers on the production line.So can you tell me the way of  firestone to use a file VPD for the Centaur.

touchqvfb, 2016-01-14:
I change 
unset SET_NOMINAL_PSTATE, in firestone.config
BR2_OPENPOWER_PNOR_XML_LAYOUT_FILENAME="defaultPnorLayoutSingleSide.xml"   in firestone_defconfig.
nothing else.

dcrowell77, 2016-01-14:
We need some clarification on what you are trying to do here.  Are you actually running on Firestone hardware or not?  Using file-mode VPD is not recommended, though it can be done.  However, even if you use file-mode you still need the actual VPD data to boot.  Do you have that?

cooktail, 2016-03-01:
firestone_defconfig also can run on 32MB flash.

stewart-ibm, 2016-06-17:
Closing this out as invalid, this is obviously not a supported machine.

closed 2016-06-17 by stewart-ibm