open-power/op-build #349: Version must contain op-build and kernel details

shenki, 2016-02-15:
Currently we lack any information about the op-build tag/sha/date if the vendor choses to override product version.

```
open-power-palmetto-156807c-dirty
	hostboot-390b72b-opdirty-992a9bd
	occ-1093bf9-opdirty
	skiboot-5.1.12-opdirty
	hostboot-binaries-43d5a59-opdirty
	palmetto-xml-8281868-opdirty-caaa70c
	capp-ucode-105cb8f-opdirty
```

We never put any information about petitboot versions nor kernel versions.

I have started work on this here: https://github.com/shenki/op-build/commit/81150befe82332daa63f1113bb04bfca554305e5

We want it to look like this:

```
open-power-palmetto-156807c-dirty
	hostboot-390b72b-opdirty-992a9bd
	occ-1093bf9-opdirty
	skiboot-5.1.12-opdirty
	hostboot-binaries-43d5a59-opdirty
	palmetto-xml-8281868-opdirty-caaa70c
	capp-ucode-105cb8f-opdirty
	linux-4.2.8-ooenpower1-opdirty
        petitboot-v1-417fsdj-opdirty
```

shenki, 2016-03-09:
11:38 < joel> sammj: yeah. that doesn't do what we need for 'linux' tho
11:38 < sammj> joel: right, just noticed it's missing the patch versions
11:38 < joel> sammj: also, we want to add the op-build and buildroot shas
11:39 < joel> sammj: so that when "vendors" (aka IBM) change the top level 
              string to op810something, we still have a clue without reading the 
              magic webpage

shenki, 2016-03-09:
@sammj is working on this

williamspatrick, 2016-03-09:
In theory it is as simple as updating this line:

https://github.com/open-power/op-build/blob/master/openpower/package/openpower-pnor/openpower-pnor.mk#L36

sammj, 2016-03-11:
Latest version of addressing this is here: https://github.com/open-power/op-build/pull/382

stewart-ibm, 2016-03-15:
Since #382 was merged, closing this issue.

closed 2016-03-15 by stewart-ibm