open-power/op-build #355: Move to GCC 5.3

stewart-ibm, 2016-02-22:
Newer buildroot supports gcc 5.3.

we plan to switch toolchain up to 5.3 with op-build v1.9, giving enough time for projects to solve any GCC 5.3 build issues they may have.

shenki, 2016-02-23:
Buildroot tree is at https://github.com/shenki/buildroot/tree/2016.02-op-build

Can be built for Habanero using the config here https://github.com/shenki/op-build/tree/buildroot-2016.02

Note that OCC and Hostboot do not build with the newer GCC

stewart-ibm, 2016-02-26:
We'll want to verify that we're not affected by https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69969

shenki, 2016-02-29:
Given we're planning on tagging v1.9 in July, we should target GCC 6 for this release, which should be out around April.

dcrowell77, 2016-02-29:
What is the compelling reason to move forward to a new gcc at this point?  It seems like being on the bleeding edge doesn't serve us any purpose when we have a working environment now.

stewart-ibm, 2016-03-03:
5.3 gets us P9 (probably, if there's backports), moving to GCC 6 when it's available certainly gets us P9. It also gets us better warnings, codegen, a toolchain matching modern linux distros that people will be using for development and the more recent we are, the more help we can get from our toolchain team.

stewart-ibm, 2016-04-27:
Moving to v1.10 target, v1.9 is too close to make the change.

shenki, 2016-05-05:
Move to GCC 6.1?

stewart-ibm, 2016-05-05:
where's buildroot going? It'd be great to just follow that lead.

shenki, 2016-05-05:
They're on 6.1, so their 2016.05 release will contain the same:

https://plus.google.com/100084591154899263420/posts/hDvZrD7XCvT

stewart-ibm, 2016-08-30:
I have posted https://github.com/open-power/hostboot/pull/62 which fixes Hostboot for GCC6.

I'm waiting on Hostboot to merge this and then I think we go for it - we have a workable PNOR build then.

stewart-ibm, 2016-11-17:
So, we're now there for everything but Witherspoon. I'll open an issue for that.

closed 2016-11-17 by stewart-ibm