open-power/op-build #385: More configuration options for petitboot

KennethWilke, 2016-03-10:
Hello,

I have a feature request. I would like to be able to set some default options for petitboot, such as the default timeout and boot options via the menuconfig in op-build. From what I understand these settings are in the nvram partition. Would this be possible to add?

Thanks!

stewart-ibm, 2016-03-11:
The easiest way would likely be to have some way to set the initial contents of the NVRAM partition... although that then puts it as "preset preferences" rather than "defaults" - so if you erased NVRAM to reset to default, you'd get a different result.

@sammj - your thoughts?

sammj, 2016-03-11:
Yep that sounds like the neatest way, although as you say if NVRAM is cleared at some point later on you will get the Petitboot defaults, not what was specified in op-build.
_Hypothetically_ we could also supply default values to Petitboot at compile time, but that could get a bit too messy.

shenki, 2016-03-15:
I like the idea of initialising nvram at build time. Even if the user has not chosen any defaults, this would save us time on the first boot where we currently initialise nvram.

@KennethWilke do you have someone from your team who would be interested in implementing this?