open-power/op-build #388: Move to buildroot 2016.02

stewart-ibm, 2016-03-11:


stewart-ibm, 2016-03-31:
I've mailed the list with requests to look at these trees:

https://github.com/stewart-ibm/buildroot/tree/2016.02-op-build
https://github.com/stewart-ibm/op-build/tree/buildroot-2016.02

shenki, 2016-03-31:
Hey @stewart-ibm, looking good! I've got some additional commits in my branch here:

https://github.com/shenki/buildroot/commits/2016.02-op-build

The summary:
* busybox version bump: security updates
* powerpc-utils: version bump, this went into master yesterday so you hadn't seen it
* dropbear: version bump: security updates, systemd init so we drop @sammj's 'prevent install of init script' patch 

We need to ensure that dropbear does not start at boot. This could be achieved by setting `DROPBEAR_INSTALL_INIT_SYSTEMD`, but I assume this would stop us from being able to do `service start dropbear`. We could remove `/etc/systemd/system/multi-user.target.wants/dropbear.service` from the rootfs? Is there a better practice here?

stewart-ibm, 2016-04-01:
Doesn't https://github.com/stewart-ibm/buildroot/commit/1951655fb0c45dab90a21829c77b8347df665e88 get rid of the dropbear install?

We can have a script that runs at the end of everything and cleans things up.... but we could also just add an option to install teh init script somewhere where it's disabled by default.

stewart-ibm, 2016-04-01:
After an on-IRC hint from @shenki , I see we're talking newer dropbear that only has systemd activation...

we should likely just wipe out *all* systemd activation scripts as a final step of image generation.

closed 2016-04-26 by stewart-ibm

shenki, 2016-04-27:
Was this closed-fixed? Perhaps a we should link to the commit when closing bugs

shenki, 2016-04-27:
Found it: https://github.com/open-power/op-build/compare/911dc99fa7...10be651e90

stewart-ibm, 2016-04-27:
I'm still so *incredibly* used to Launchpad doing that auto-magically. At some point, around 2024, I'll probably lose that habit :)