open-power/op-build #408: op-build alias gone without README update

scprek, 2016-03-28:
README indicates to do the following, which no longer works.

```bash
git clone --recursive git@github.com:open-power/op-build.git
cd op-build
. op-build-env
op-build palmetto_defconfig && op-build
```
What are developers supposed to do now?
Current error
```
> . op-build-env
-bash: op-build-env: line 11: syntax error near unexpected token `('
-bash: op-build-env: line 11: `op-build () {'
```

williamspatrick, 2016-03-28:
What version of bash are you running?  There have been a few changes lately to that alias, which were suppose to make it more widely supported and not less. 

There is no intended change here that requires README to be updated. Let's figure out why the alias is no longer working for you. 

Patrick

scprek, 2016-03-28:
Hoping this was just a nested shell issue. Austin pool machines are GNU bash, version 4.1.2(1)-release (x86_64-redhat-linux-gnu). I was in an ssh session and did a pull to get the latest, then ran into this problem, but now after exiting and trying again it work.

```
smcprek@gfw160: /esw/san2/smcprek/open-power/forks/op-build
> . op-build-env 
-bash: op-build-env: line 11: syntax error near unexpected token `('
-bash: op-build-env: line 11: `op-build () {'

smcprek@gfw160: /esw/san2/smcprek/open-power/forks/op-build
> bash --version
GNU bash, version 4.1.2(1)-release (x86_64-redhat-linux-gnu)
Copyright (C) 2009 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

smcprek@gfw160: /esw/san2/smcprek/open-power/forks/op-build
> exit
logout
Connection to gfw160 closed.

smcprek@mac: ~
> ssh gfw160

smcprek@gfw160: /esw/san2/smcprek/open-power/forks/op-build
> bash --version
GNU bash, version 4.1.2(1)-release (x86_64-redhat-linux-gnu)
Copyright (C) 2009 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

smcprek@gfw160: /esw/san2/smcprek/open-power/forks/op-build
> . op-build-env 
```

closed 2016-03-28 by scprek

shenki, 2016-03-28:
I've done this before; the problem is caused by sourcing op-build-env twice in the one shell