open-power/op-build #427: Build fail with enabling debug trace to console

johnhihi, 2016-04-01:
Hello,

I am trying to build a pnor with "**set CONSOLE_OUTPUT_TRACE**" in **barreleye.config** file.
But I got build error during openpower-pnor installing phase.

Below is the cut of error message. Seems the size is exceed the space....

> TRACE: A-D-B: side:A HBB:32899072, primaryTOC:0, backupTOC:33521664, golden: no
TRACE: A-D-B: side:B HBB:66482176, primaryTOC:33554432, backupTOC:67076096, golden: yes
ERROR: main::checkSpaceConstraints: Image provided (/home/dev/Source_fxn_for_release/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/openp
ower_pnor_scratch//hostboot_extended.header.bin.ecc) has size (11796480) which is greater than allocated space (5898240) for section=HBI.  Aborting! at /ho
me/dev/Source_fxn_for_release/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu/sysroot/hostboot_build_images//buildpnor.pl line 715.

dcrowell77, 2016-04-01:
This is an unfortunate situation that we're seeing as well.  You can play some tricks with the pnor layout in your environment to get more space into the HBI section, that's about the only recourse available.  Adding @ploetzma to add some details.

We do have a feature in development that will support outputing our machine-readable (and much more compact) trace format to the console.  This will allow the full output of the messages in the same amount of space that we take up without console tracing enabled.  This does require post-processing of the output though.  This support is still a little ways out though so you can't count on it for any immediate work.

stewart-ibm, 2016-04-11:
Dan <notifications@github.com> writes:
> This is an unfortunate situation that we're seeing as well.  You can
> play some tricks with the pnor layout in your environment to get more
> space into the HBI section, that's about the only recourse [...]

xz compress the strings?

Actually... there's probably a bunch of other things that can be
attacked on size - I've had a solid go at skiboot a few times - maybe in
a week or so (when I'm back from vacation) we could sit down and poke at
HB.

-- 
Stewart Smith
OPAL Architect, IBM.

dcrowell77, 2016-04-11:
Re: the trace strings - Plan is to just output the hashed traces that we already have going to our in-memory buffers to the console, no need to do any extra compression.

I have no doubt there are places for improvement but we are limited somewhat in what we can remove as we (the hostboot team) doesn't own a large percentage of the code, the hardware team does.  We welcome any attempts from people with the time. :-)

williamspatrick, 2016-08-31:
Duplicate of open-power/hostboot#63 .  More appropriate to track the problem there since it is a Hostboot code problem.

closed 2016-08-31 by williamspatrick