open-power/op-build #457: Refactor defconfigs to have a generic base

ruscur, 2016-05-02:
We have a defconfig for every machine with a lot of shared content, with any changes across the board needing to be made to each file.  This becomes more of a hassle to update as more and more machines get added, it'd be pretty easy to have a generic defconfig for every OpenPOWER machine type, and just have machine-specific details in each machine defconfig.

That helps people to understand specific nuances about each machine, as well as being more maintainable.

williamspatrick, 2016-05-02:
Agreed it would be desirable. I've previously looked into it myself and I don't see it as possible without a change to buildroot. Buildroot doesn't support config fragments and it doesn't have a mechanism for the "external" layer (op-build in this case) to add make recipes at defconfig time.

Someone would need to propose changes to buildroot to support this and get them accepted upstream.

shenki, 2016-05-03:
I did some googling and found this:

http://elinux.org/Buildroot:DeveloperDaysFOSDEM2014#How_to_handle_.27target.27_defconfigs_versus_.27development.27_defconfigs

Also, from http://free-electrons.com/doc/training/buildroot/buildroot-slides.pdf

> Build a release system for platform1
> $ ./support/kconfig/merge_config.sh platform1.frag packages.frag > .config
> $ make olddefconfig
> $ make
>
> Build a debug system for platform2
> $ ./support/kconfig/merge_config.sh platform2.frag packages.frag debug.frag > .config
> $ make olddefconfig
> $ make

It would be nice if buildroot let us put something like this in `$BR2_EXTERNAL/external.mk`:
```
%_defconfig: $(BR2_EXTERNAL)/config/%_frag $(BR2_EXTERNAL)/config/common_frag
        $(TOPDIR)/support/kconfig/merge_config.sh $^ > $@
```
but as Patrick said this does not work.

We could chose to do this at commit time and have both the fragments and the combined defconfigs checked into the tree. That way developers get the benefit of having easy to maintain configurations, and users don't have to perform any extra steps at build time.

shenki, 2016-05-04:
The work to be done:

0. Generate .configs for all of our platforms
1. Work out a common subset from all the defconfigs
2. Create a common_frag with that subset
3. Remove those options from all the defconfigs
4. Generate merged defconfigs (They should be the same as before?)
4. Generate .configs with the merged defconfigs, and compare them to step 1

sammj, 2016-05-09:
Sam Bobroff has a buildroot patch that allows defconfigs to be specified as the combination of several partial defconfigs which suits our usecase very well.

http://lists.busybox.net/pipermail/buildroot/2016-May/160745.html

sammj, 2016-05-11:
I've put a first-shot at splitting up the platform defconfigs up over here, based on the above patch: https://github.com/sammj/op-build/commit/6a334b864cd06f6ac9bcbdf2505d7f7d47dee8dc
Bikeshed away :)

sammj, 2016-06-29:
Sam has submitted an improved version of his mergeconfig changes as part of this patchset - http://patchwork.ozlabs.org/patch/638448/
A V2 is incoming, if we like the look of that perhaps we should consider including it in our tree (the buildroot list is pretty busy so it will probably be a while before it is merged)

sammj, 2018-08-08:
This didn't end up making it upstream as Buildroot decided that building configs is something that should be done by the user; Buildroot just consumes them.
For context: http://lists.busybox.net/pipermail/buildroot/2016-July/165623.html

We can still look into making a generic base somehow, but it's going to mean having our own mechanism to build defconfigs.