open-power/op-build #470: Switching from golden side pnor boot to primary side pnor boot is not happening for habanero machine.

pridhiviraj, 2016-05-09:
Problem statement:
Install a habanero system with OpenPower Firmware, and do a golden side pnor boot. After further reboots system still booting always from golden side pnor firmware, and it is not booting from primary side pnor firmware.

How to Reproduce:
1.Install a Habanero system with latest OpenPower Firmware.

ipmitool sensor list | grep -i "boot count"
Boot Count | 0x0 | discrete | 0x0280| na | na | na | na | na | na

initially boot count =0x02

  1.Do a Power ON followed by immediate power off(To make sure boot fail early)
    check for boot count sensor value, it should be decremented by 1
    ipmitool sensor list | grep -i "boot count"
    Boot Count | 0x0 | discrete | 0x0180| na | na | na | na | na | na

  2.Do a Power ON followed by immediate power off for 2nd time(To make sure boot fail early)
    check for boot count sensor value, it should be decremented by 1
    ipmitool sensor list | grep -i "boot count"
    Boot Count | 0x0 | discrete | 0x0080| na | na | na | na | na | na

  3.Do a Power ON Operation to boot the system from golden side PNOR(Because boot count =0x00)

ipmitool sensor list | grep -i golden
BIOS Golden Side | 0x0 | discrete | 0x0180| na | na | na | na | na | na

BMC Golden Side | 0x0 | discrete | 0x0080| na | na | na | na | na | na

cd /sys/firmware/devicetree/base/xscom@3fc0000000000/isa@b0020/pnor@fc000000
lsprop *-tocs
active-image-tocs
02000000 03ff8000
alternate-image-tocs
00000000 01ff8000
golden-image-tocs
02000000 03ff8000
Here we can observe at this stage system booted from golden side

   1. Reboot the sytem again. reboot [ OK ] Started Show Plymouth Reboot Screen. [ OK ] Stopped Virtualization daemon. Stopping D-Bus System Message Bus...

After completion of reboot check again system boot side still it is in golden side of pnor.
cd /sys/firmware/devicetree/base/xscom@3fc0000000000/isa@b0020/pnor@fc000000
lsprop *-tocs
active-image-tocs
02000000 03ff8000
alternate-image-tocs
00000000 01ff8000
golden-image-tocs
02000000 03ff8000

ipmitool sensor list | grep -i golden
BIOS Golden Side | 0x0 | discrete | 0x0180| na | na | na | na | na | na

BMC Golden Side | 0x0 | discrete | 0x0080| na | na | na | na | na | na

Further subsequent reboots also system is not booting from primary side of pnor, and it is always booting from golden side of pnor.

How to switch from golden side pnor to primary side pnor manually:

   1.On BMC Busy box execute below command.
    /etc/init.d/boot_into_primary

   2.Do a ipmi hard power reset to boot the system into primary.
    ipmitool power reset.

Firmware levels:
When system is in primary side:
open-power-habanero-6ec4e33
buildroot-d78ccc1
skiboot-5.2.1
hostboot-e189edd-9824b9f
linux-4.4.8-openpower1-bd9161a
petitboot-v1.0.1-8f13d5f
habanero-xml-5a2a541-cb4e251
occ-19a3ebc
hostboot-binaries-7f593a3
capp-ucode-105cb8f

When system is in golden side:
Product Name : OpenPOWER Firmware
Product Version : TN71-BP012-810.1611.20160406a
Product Extra : hostboot-bc98d0b-3f66050
Product Extra : occ-0362706-805fe5a
Product Extra : skiboot-5.1.13
Product Extra : hostboot-binaries-43d5a59
Product Extra : habanero-xml-5a2a541-228a675
Product Extra : capp-ucode-105cb8f

dcrowell77, 2016-05-09:
You can force the sensors (Golden Side BIOS and Boot Count) to get back to primary, but this seems like a valid issue somewhere up in skiboot.  Once we successfully boot to runtime the Boot Count should be set back to 2.  Can you read that value from the runtime state?

pridhiviraj, 2016-05-10:
Once system boots from golden side pnor, below is the value of  boot count sensor read at runtime. skiboot is setting back to 2, But still system is always is booting from golden side pnor after further subsequent reboots.
Boot Count       | 0x0        | discrete   | 0x0280| na        | na        | na        | na        | na        | na 

BIOS Golden Side | 0x0        | discrete   | 0x0180| na        | na        | na        | na        | na        | na        
BMC Golden Side  | 0x0        | discrete   | 0x0080| na        | na        | na        | na        | na        | na     

The above values are same, after further reboots, skiboot is setting back to 2, But actually system is booting from golden side PNOR.

pridhiviraj, 2016-05-10:
And also collected logs from BMC, there also it is mentioning same.

# cat /var/bmc_collectlogs.log | grep -i boot   
root=/dev/mtdblock3 ro ip=none console=ttyS4,38400 rootfstype=cramfs bigphysarea=8192 imagebooted=1
May 10 10:03:10 AMIA0423F3062AE compmanager: [1275 CRITICAL][compmngrhostboot.c:929]OEM extension loaded successfully#012 
May 10 10:19:54 AMIA0423F3062AE IPMIMain: [697 CRITICAL][OEMHooksHandler.c:1321]Primary side PNOR has failed to boot on repeated attempts. Trying to boot from Golden Side sensor now..
<5>Kernel command line: root=/dev/mtdblock3 ro ip=none console=ttyS4,38400 rootfstype=cramfs bigphysarea=8192 imagebooted=1
May 10 10:03:05 AMIA0423F3062AE /usr/sbin/cron[1299]: (CRON) INFO (Running @reboot jobs)
May 10 10:09:46 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Primary side of PNOR...
May 10 10:10:38 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Primary side of PNOR...
May 10 10:16:08 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Primary side of PNOR...
May 10 10:17:20 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Primary side of PNOR...
May 10 10:19:54 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Golden side of PNOR...
May 10 10:25:34 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Golden side of PNOR...
May 10 10:32:26 AMIA0423F3062AE IPMIMain: [697 INFO]Booting from Golden side of PNOR...
May 10 10:02:40 AMIA0423F3062AE kernel: Kernel command line: root=/dev/mtdblock3 ro ip=none console=ttyS4,38400 rootfstype=cramfs bigphysarea=8192 imagebooted=1
May 10 10:09:46 AMIA0423F3062AE kernel: Performing the Normal side Full Power ON + Boot Sequence
May 10 10:09:47 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 0 
May 10 10:09:47 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence
May 10 10:10:38 AMIA0423F3062AE kernel: Performing the Normal side Full Power ON + Boot Sequence
May 10 10:10:39 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 0 
May 10 10:10:39 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence
May 10 10:16:08 AMIA0423F3062AE kernel: Performing the Normal side Full Power ON + Boot Sequence
May 10 10:16:09 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 0 
May 10 10:16:09 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence
May 10 10:17:20 AMIA0423F3062AE kernel: Performing the Normal side Full Power ON + Boot Sequence
May 10 10:17:21 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 0 
May 10 10:17:21 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence
May 10 10:19:54 AMIA0423F3062AE kernel: Performing the Golden side Full Power ON + Boot Sequence
May 10 10:19:55 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 1 
May 10 10:19:55 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence
May 10 10:25:34 AMIA0423F3062AE kernel: Performing the Golden side Full Power ON + Boot Sequence
May 10 10:25:35 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 1 
May 10 10:25:35 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence
May 10 10:32:26 AMIA0423F3062AE kernel: Performing the Golden side Full Power ON + Boot Sequence
May 10 10:32:27 AMIA0423F3062AE kernel: Performing the Boot Sequence after Power ON routine boot side = 1 
May 10 10:32:27 AMIA0423F3062AE kernel: Performing the FSI Magic Sequence as part of Boot Sequence

FW Level of BMC:
# cat /proc/ractrends/Helper/FwInfo 
FW_VERSION=2.16.91818
FW_DATE=Mar 7 2016
FW_BUILDTIME=13:37:36 CDT
FW_DESC=8348 P2 SRC RR9 03072016
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X
# 
# uname -a
Linux AMIA0423F3062AE 2.6.28.10-ami #1 Mon Mar 7 13:26:51 CST 2016 armv5tejl unknown
#

dcrowell77, 2016-05-10:
These seems like an issue with the BMC.  Those issues are outside the scope of this repo.

williamspatrick, 2016-05-13:
@stewart-ibm , @spinler : Any ideas on how this is suppose to work?  My guess would be that when we get up far enough, the host sends down an IPMI progress indicator to say we have made it up that far, and then the BMC turns off golden mode (because there has been a successful boot).  I would expect that this is part of the same code the resets the boot count on the BMC.

@spinler - Do you agree with my guess of how it is suppose to work above?  Any idea what the BMC keys off of?

@stewart-ibm - Does skiboot, petitboot, or the final Linux send the "appropriate" progress code for this?  Can you confirm that the test case above would have gotten far enough to reset the boot count / golden-side?

williamspatrick, 2016-05-13:
@stewart-ibm: I'm going to assign to you for now.

williamspatrick, 2016-05-13:
@stewart-ibm: I guess it would also be good to confirm that 'reboot' does a full reIPL and not a fast kexec reboot.  Obviously a kexec is not going to switch us off the golden side since the BMC doesn't see that as a full reIPL.

spinler, 2016-05-13:
Looks like this is also tracked under SW352606.  This is all controlled by the BMC, so you can close this issue if you want and we can track it in CQ where we can bridge to AMI if necessary.

stewart-ibm, 2016-11-03:
(copied from internal bug tracker here)

AMI's response:

Jayesh,

This is the intended behavior, and working as designed. This is how it has been ever since it has been implemented in Habanero timeframe.

The reason is that, for the BMC, 'Boot Count' and the 'BIOS Golden Side' sensor is the tracker to identify which side of PNOR should be booted.

Here, by doing a immediate power off after a power on, you are simulating the behavior as if host didn't complete booting. (If the host had booted, it would reset the Boot Count sensor back to 2).

So, after 2 attempts, when Boot Count sensor goes to 0, according to BMC's state-machine, it has attempted 2 boots to the Primary side, and BMC would try the Golden side of PNOR. That is why the Boot Count sensor is 0, and the BIOS Golden side sensor is 1. When the Golden side PNOR boots, it would reset the Boot Count sensor back to 2. If you do a power off again when the Golden PNOR is booting, then BMC would think that Golden PNOR also failed to boot and BMC would give up.

Now, the BMC knows that, the primary side failed and hence BMC booted the Golden side (as indicated by BIOS Golden Side) sensor. Hence, even on repeated boots, until the "BIOS Golden side" sensor is 1, BMC would continue to boot only the Golden PNOR, irrespective of how many times you try.

If the user wants to manually boot into the Primary side of PNOR, user would have to set the "BIOS Golden Side" sensor to 0 (using set sensor reading command), and then do a power reset or power cycle, and host will boot from Primary side of PNOR.

Thus for the BMC, if the "BIOS Golden side" sensor is 1, then Golden PNOR will be booted. If the "BIOS Golden side" sensor is 0, then Primary PNOR will be booted. This gives the user, the control to boot either sides of PNOR whenever user wants to, in addition to automated fail-over recovery or boot into golden PNOR when primary PNOR fails to boot.

closed 2016-11-03 by stewart-ibm