open-power/op-build #471: ipmitool fru PNOR version info missing capp-ucode- version.

pridhiviraj, 2016-05-09:
ipmi fru info for pnor version, missing capp-ucode version.

How to reproduce:
Build a habanero_update.pnor image using op-build master-next branch, Flash this image on OpenPower habanero machine.
Read ipmitool fru print 43 for PNOR version info.
ipmitool fru print 43
Product Name : OpenPOWER Firmware
Product Version : open-power-habanero-6ec4e33
Product Extra : buildroot-d78ccc1
Product Extra : skiboot-5.2.1
Product Extra : hostboot-e189edd-9824b9f
Product Extra : linux-4.4.8-openpower1-bd9161a
Product Extra : petitboot-v1.0.1-8f13d5f
Product Extra : habanero-xml-5a2a541-cb4e251
Product Extra : occ-19a3ebc
Product Extra : hostboot-binaries-7f593a3
Product Extra : c

Try reading PNOR VERSION partition using ipmitool.
ipmitool raw 0x3a 0x0b 0x56 0x45 0x52 0x53 0x49 0x4f 0x4e 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 | xxd -r -p
open-power-habanero-6ec4e33
buildroot-d78ccc1
skiboot-5.2.1
hostboot-e189edd-9824b9f
linux-4.4.8-openpower1-bd9161a
petitboot-v1.0.1-8f13d5f
habanero-xml-5a2a541-cb4e251
occ-19a3ebc
hostboot-binaries-7f593a3
capp-ucode-105cb8f

We can observe from above two outputs, ipmitool fru print 43 is missing app-ucode-105cb8f which is a version of capp-ucode.

contact: ppaidipe@linux.vnet.ibm.com

Comment from @williamspatrick 
Now that the Linux and buildroot version was added, we do not have the space in the ipmi sensor to hold all of the version information, so it gets truncated. We made a decision on the order of the packages so that higher priority packages appeared first and capp-ucode was considered lower priority. We need to either revert to the old version behavior, which did not have Linux, buildroot, and Petitboot, or send this to the BMC to see if the sensor size can be increased. This is not a skiboot defect. 

Patrick 
Sent from my iPhone

sammj, 2016-05-09:
This is not necessarily an op-build defect either. If it is a hard requirement (why?) that we see the capp-ucode version over IPMI then either

a) We need to decide which version should be dropped to fit it, or
b) Ask the BMC vendor to increase the size.

Personally, I think we should just not worry about it unless it is easy enough to fix on the BMC.

williamspatrick, 2016-05-13:
@pridhiviraj: Please respond to @sammj's comments above.  I mostly agree with him.  Is there are hard requirement on specific version fields showing up (and where are they documented?) or is this, as my understanding, a best-effort sensor?

@wghoffa: You originally worked with AMI and wrote the Hostboot code for this sensor.  Would appreciate any insight you have as well.

wghoffa, 2016-05-13:
We did have talks of increasing the size of sections of the FRU Inventory in the past and instead decided on using additional records for the additional data. Obviously it is less desirable, but it was the simpler solution at the time and could be used here as well. If there are more desirable fw levels we could figure out how to make sure those are displayed as well.

pridhiviraj, 2016-05-14:
Hi
There is no specific hard requirements, But generally all users use ipmitool fru print for any of the inventory info, so better to display all the FW levels either by increasing the sensor size or can use additional records as mentioned by @wghoffa .

dcrowell77, 2016-05-16:
Unfortunately I think that increasing the sensor size requires changes in the BMC code stack (though I could be wrong).

closed 2016-08-31 by williamspatrick