open-power/op-build #474: mc reset cold making the overall system to boot instead of only BMC.

pridhiviraj, 2016-05-12:
Install a OpenPower firmware on habanero machine, which is built from master-next branch of op-build.
Problem statement:
1.Power of the system
2.Issue a ipmitool mc reset cold 
3.At this stage we expect only BMC to reset, But Overall system is booting.

Seen this issue at below commit level of op-build, 
3b7310db3c507c1058daf41bcfc1798145909457

ipmitool -H <BMC-IP> -I lanplus -U ADMIN -P admin power off
Chassis Power Control: Down/Off
ipmitool -H <BMC-IP> -I lanplus -U ADMIN -P admin power status
Chassis Power is off
ipmitool -H <BMC-IP> -I lanplus -U ADMIN -P admin mc reset cold
Sent cold reset command to MC
ipmitool -H <BMC-IP> -I lanplus -U ADMIN -P admin sol activate
[SOL Session operational.  Use ~? for help]
 30.67927|ISTEP 16. 2
 30.92453|ISTEP 16. 3
 30.95228|ISTEP 16. 4
 31.19216|ISTEP 18.13
 31.34792|ISTEP 18.14
 31.32799|ISTEP 20. 1
 33.40771|ISTEP 21. 1
 39.54485|htmgt|OCCs are now running in ACTIVE state

Firmware levels:
PNOR:
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-3b7310d
 Product Extra         : 	buildroot-bc248fa
 Product Extra         : 	skiboot-5.2.2
 Product Extra         : 	hostboot-15bb028-9824b9f
 Product Extra         : 	linux-4.4.9-openpower1-cbbfd90
 Product Extra         : 	petitboot-v1.0.1-8f13d5f
 Product Extra         : 	habanero-xml-5a2a541-cb4e251
 Product Extra         : 	occ-19a3ebc
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	c

BMC Level:
FW_VERSION=2.16.91818
FW_DATE=Mar 7 2016
FW_BUILDTIME=13:37:36 CDT
FW_DESC=8348 P2 SRC RR9 03072016
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X
# 

This issue is not seen on OP810.30 Builds.

dcrowell77, 2016-05-12:
Can you leave the console active in a window during the entire process?  It is odd that you see istep16 first.  I don't have any guesses on how the FW in the PNOR image would be able to affect this at all.  The BMC is in total control of all boot operations so you will probably need to investigate at that level.

pridhiviraj, 2016-05-12:
Hi
I got the clue.
Due to my earlier ipmi command executions, executed the ipmitool chassis policy always-on. So i am thinking this command causing the overall system to boot automatically when issuing BMC Cold reset. Then I tried issuing chassis policy to always-off, Now issuing cold reset is not causing the overall system to boot.
I think it is behaving correctly.

Thanks

dcrowell77, 2016-05-12:
Yep, that would do it.  Please close this issue.

closed 2016-05-12 by williamspatrick