open-power/op-build #475: ipmitool usb interface is not working on habanero machine

pridhiviraj, 2016-05-12:
Install a habanero system with OpenPower Firmware which is built from master-next branch of op-build
Problem statement:
ipmitool usb interface is not working.
./ipmitool -I usb chassis status
Error in USB session setup 

Unable to setup interface usb
Error loading interface usb
echo $?
1

Firmware Levels:
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-3b7310d
 Product Extra         : 	buildroot-bc248fa
 Product Extra         : 	skiboot-5.2.2
 Product Extra         : 	hostboot-15bb028-9824b9f
 Product Extra         : 	linux-4.4.9-openpower1-cbbfd90
 Product Extra         : 	petitboot-v1.0.1-8f13d5f
 Product Extra         : 	habanero-xml-5a2a541-cb4e251
 Product Extra         : 	occ-19a3ebc
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	c

Op-build commit id:
3b7310db3c507c1058daf41bcfc1798145909457

usb interface is working fine on OP810.30 builds.

dcrowell77, 2016-05-12:
From where are you running ipmitool - the host OS, the BMC or an external system?

pridhiviraj, 2016-05-12:
i am running through in-band i.e on Host OS, and also tried with not only distro provided ipmitool , but with upstream ipmitool.

pridhiviraj, 2016-05-12:
Distro provided ipmitool.
 ipmitool -V
ipmitool version 1.8.15
ipmitool -I usb chassis status
Error in USB session setup 

Unable to setup interface usb
Error loading interface usb
echo $?
1

Upstream ipmitool
./ipmitool -V
ipmitool version 1.8.17-cvs
./ipmitool -I usb chassis status
Error in USB session setup 

Unable to setup interface usb
Error loading interface usb
echo $?
1

This is tested on PowerKVM 3.1.0 and RHEL 7.2 LE. On both the Os's usb interface is failing.

shenki, 2016-05-12:
USB is broken in this version. Please update to the latest master. You need to be at least https://github.com/open-power/op-build/commit/66133f351d2f5ca2689814a67327b5f2c442a06c which has kernel `v4.4.9-openpower2`

closed 2016-05-12 by shenki