open-power/op-build #476: Move to buildroot 2016.05

shenki, 2016-05-12:
WIP is at

  https://github.com/shenki/op-build/tree/buildroot-2016.05
  https://github.com/shenki/buildroot/tree/2016.05-op-build

This is based on Buildroot's 2016.05-rc1. It builds a working firmware that runs on a palmetto.

TODO:
 - update config options (fragments? https://github.com/open-power/op-build/issues/457)
 - try to upstream Stewart's fakeroot fix (they didn't take it: http://lists.busybox.net/pipermail/buildroot/2016-March/154911.html)

stewart-ibm, 2016-05-13:
We may want to do the board defconfig cleanups in two passes, one rearranging things the other updating for new buildroot. my guess is that we haven't done a savedefconfig for a very long time.

shenki, 2016-05-13:
Yes, they should be separate patches.

We are up to date wrt savedefconfig - whenever I update them I do a savedefconfig.

stewart-ibm, 2016-06-27:
Solved by https://github.com/open-power/op-build/pull/504

closed 2016-06-27 by stewart-ibm