open-power/op-build #482: when system in runtime, issuing of mc reset cold results in xscom write/read errors for sensors command.

pridhiviraj, 2016-05-19:
Install a OpenPower Habanero system with habenaro_update.pnor which is built from op-build master-next branch having commit id: f70946e5cd47929fd1e32ee2a3915d42809459b0.
Problem statement:
when system is in runtime, bmc cold reset resulting sensors command failure and makes the system crash one time and also results in lot of xscom write and read errors.

System is in below firmware level:
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-f70946e
 Product Extra         : 	buildroot-bc248fa
 Product Extra         : 	skiboot-5.2.2
 Product Extra         : 	hostboot-b31723b-ccd5752
 Product Extra         : 	linux-4.4.9-openpower2-8b0d756
 Product Extra         : 	petitboot-v1.0.1-1c3bb54
 Product Extra         : 	habanero-xml-5a2a541-0bc2709
 Product Extra         : 	occ-19a3ebc
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	c

Reproduction steps:
1. Install a habanero system with habanero_update.pnor which is built from op-build master-next branch having above commit id.
2. During system is in runtime, execute sensors command.
# sensors
No sensors found!
Make sure you loaded all the kernel drivers you need.
Try sensors-detect to find out which these are.
# modprobe ibmpowernv
# lsmod | grep -i powernv
ibmpowernv              6123  0 
powernv_flash           5610  0 
ipmi_powernv            5611  0 
mtd                    65409  3 powernv_flash,ofpart
ipmi_msghandler        66801  2 ipmi_powernv,ipmi_devintf
powernv_rng             2617  0 
 # sensors
ibmpowernv-isa-0000
Adapter: ISA adapter
Core 0-7:     +53.0°C  
Core 8-15:    +53.0°C  
Core 16-23:   +51.0°C  
Core 24-31:   +53.0°C  
Core 32-39:   +53.0°C  
Core 40-47:   +56.0°C  
Core 48-55:   +56.0°C  
Core 56-63:   +57.0°C  
Core 64-71:   +56.0°C  
Core 72-79:   +55.0°C  
Centaur 0:    +40.0°C  
Centaur 1:    +44.0°C  
Centaur 4:    +38.0°C  
Centaur 5:    +40.0°C  

Here sensors command will work fine as expected.
3. When system is in runtime, issue BMC cold reset.
4. After BMC comes up execute sensors command, it is failed to read centaur memory buffer temperatures and also resulted in system hang one time.
[SOL Session operational.  Use ~? for help]

[root@ite037 ~]# 
[root@ite037 ~]# 
[root@ite037 ~]# sensors
ibmpowernv-isa-0000
Adapter: ISA adapter
Core 0-7:     +51.0°C  
Core 8-15:    +51.0°C  
Core 16-23:   +50.0°[1164863065631,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[1164863069433,6] IPMI: dropping non severe PEL event
C  
Core 24-31:[1164865357782,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
   +[1164866382370,3] MFSI 0:0: Access error on port 3, stat=000100000000
51.0°C  [1164868396591,3] CENTAUR 80000000: MFSI write error -6 writing CMD

Core 32-39:   +53.0°C  
[... 1959 lines cut ...]
[ 2465.052495] swapper/62      R  running task        0     0      1 0x00000800
[ 2465.052534] Call Trace:
[ 2465.052550] [c000003c5cb23b80] [000001128d5664f7] 0x1128d5664f7 (unreliable)
[ 2465.052596] [c000003c5cb23d50] [c0000000000164f8] __switch_to+0x268/0x450
[ 2465.052634] [c000003c5cb23d70] [c0000000001234f4] finish_task_switch+0x74/0x210
[ 2465.052679] [c000003c5cb23db0] [c0000000009425c8] __schedule+0x338/0xa40
[ 2465.052718] [c000003c5cb23e80] [c0000000009442c4] schedule_preempt_disabled+0x34/0xb0
[ 2465.052762] [c000003c5cb23ea0] [c00000000015d560] cpu_startup_entry+0x180/0x300
[ 2465.052807] [c000003c5cb23f20] [c00000000004be00] start_secondary+0x310/0x340
[ 2465.052846] [c000003c5cb23f90] [c000000000009a6c] start_secondary_prolog+0x10/0x14

[ 2645.102172] INFO: rcu_sched detected stalls on CPUs/tasks: { 62} (detected by 24, t=42015 jiffies, g=6274, c=6273, q=0)
[ 2645.102522] Task dump for CPU 62:
[ 2645.102574] swapper/62      R  running task        0     0      1 0x00000800
[ 2645.102633] Call Trace:
[ 2645.102657] [c000003c5cb23b80] [000001128d5664f7] 0x1128d5664f7 (unreliable)
[ 2645.102727] [c000003c5cb23d50] [c0000000000164f8] __switch_to+0x268/0x450
[ 2645.102787] [c000003c5cb23d70] [c0000000001234f4] finish_task_switch+0x74/0x210
[ 2645.102858] [c000003c5cb23db0] [c0000000009425c8] __schedule+0x338/0xa40
[ 2645.102918] [c000003c5cb23e80] [c0000000009442c4] schedule_preempt_disabled+0x34/0xb0
[ 2645.102989] [c000003c5cb23ea0] [c00000000015d560] cpu_startup_entry+0x180/0x300
[ 2645.103058] [c000003c5cb23f20] [c00000000004be00] start_secondary+0x310/0x340
[ 2645.103119] [c000003c5cb23f90] [c000000000009a6c] start_secondary_prolog+0x10/0x14

Not sure , this issue persists with OP810.30 GA'ed FW, will try on it

pridhiviraj, 2016-05-19:
After issuing BMC reset cold not able to get the list of FSI centaur chips using getscom tool.
cd skiboot/external/xscom-utils/
 ./getscom -l
Chip ID  | Rev   | Chip type
---------|-------|--------
[282128895146,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[282128898670,6] IPMI: dropping non severe PEL event
[282130255551,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
[282132296049,3] MFSI 0:0: Access error on port 6, stat=000100000000
[282133308226,3] CENTAUR 80000005: MFSI write error -6 writing CMD
[282134807793,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[282134810879,6] IPMI: dropping non severe PEL event
[282136528548,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
[282137550111,3] MFSI 0:0: Access error on port 7, stat=000100000000
[282153042125,3] CENTAUR 80000004: MFSI write error -6 writing CMD
[282154131167,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[282154134218,6] IPMI: dropping non severe PEL event
[282156171350,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
[282162682598,3] MFSI 0:0: Access error on port 2, stat=000100000000
[282163719169,3] CENTAUR 80000001: MFSI write error -6 writing CMD
[282164778294,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[282164781349,6] IPMI: dropping non severe PEL event
[282166923135,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
[282168027095,3] MFSI 0:0: Access error on port 3, stat=000100000000
[282169053991,3] CENTAUR 80000000: MFSI write error -6 writing CMD
00000000 | DD2.0 | P8 (Venice) processor

pridhiviraj, 2016-05-19:
Reproduced with OP810.30 as well. Below are my findings.
ipmitool fru print 43
 Product Name          : OpenPOWER Firmware
 Product Version       : IBM-habanero-ibm-OP8_v1.7_1.62
 Product Extra         : 	hostboot-bc98d0b-1a29dff
 Product Extra         : 	occ-0362706-16fdfa7
 Product Extra         : 	skiboot-5.1.13
 Product Extra         : 	hostboot-binaries-43d5a59
 Product Extra         : 	habanero-xml-a71550e-cdd3b31
 Product Extra         : 	capp-ucode-105cb8f
cd skiboot/external/xscom-utils/
 ./getscom -l
Chip ID  | Rev   | Chip type
---------|-------|--------
80000005 | DD2.0 | Centaur memory buffer
80000004 | DD2.0 | Centaur memory buffer
80000001 | DD2.0 | Centaur memory buffer
80000000 | DD2.0 | Centaur memory buffer
00000000 | DD2.0 | P8 (Venice) processor
sensors
No sensors found!
Make sure you loaded all the kernel drivers you need.
Try sensors-detect to find out which these are.
# modprobe ibmpowernv
# sensors
ibmpowernv-isa-0000
Adapter: ISA adapter
Core 0-7:     +53.0°C  
Core 8-15:    +53.0°C  
Core 16-23:   +53.0°C  
Core 24-31:   +53.0°C  
Core 32-39:   +56.0°C  
Core 40-47:   +57.0°C  
Core 48-55:   +56.0°C  
Core 56-63:   +59.0°C  
Core 64-71:   +57.0°C  
Core 72-79:   +56.0°C  
Centaur 0:    +41.0°C  
Centaur 1:    +45.0°C  
Centaur 4:    +39.0°C  
Centaur 5:    +41.0°C  

 ipmitool mc reset cold
Sent cold reset command to MC
# [  277.749623] usb 1-1.1: USB disconnect, device number 3
[  278.773611] usb 1-1.2: USB disconnect, device number 4
[  279.799243] usb 1-1.3: USB disconnect, device number 5
[  280.769250] usb 1-1.4: USB disconnect, device number 6
Error: No response to keepalive - Terminating session
Error: No response de-activating SOL payload
[SOL Session operational.  Use ~? for help]

# sensors
ibmpowernv-isa-0000
Adapter: ISA adapter
Core 0-7:     +48.0°C  
Core 8-15:    +48.0°C  
Core 16-23:   +47.0°[236280079001,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[236280082957,6] IPMI: dropping non severe PEL event
C [236281611180,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
 
Co[236282608104,3] MFSI 0:0: Access error on port 3, stat=000100000000
re 24-31: [236283679855,3] CENTAUR 80000000: MFSI write error -6 writing CMD
  +48.0°C  
Core 32-39:   +50.0°C  
Core 40-47:   +51.0�[236285804646,3] XSCOM: write error gcid=0x0 pcb_addr=0x20000 stat=0x7
[236285807880,6] IPMI: dropping non severe PEL event
�C[236287731791,3] MFSI 0:0: XSCOM error -6 writing OPB CMD
  
C[236288765173,3] MFSI 0:0: Access error on port 2, stat=000100000000
[... 2393 lines cut ...]
[  979.754470] [c000003c21eb7870] [c0000000007d6acc] rtnetlink_rcv+0x3c/0x60
[  979.758675] [c000003c21eb78a0] [c000000000803708] netlink_unicast+0x548/0x610
[  979.760770] [c000003c21eb7950] [c000000000804c40] netlink_sendmsg+0x3c0/0x970
[  979.762826] [c000003c21eb7a40] [c00000000078d00c] sock_sendmsg+0xec/0x140
[  979.765045] [c000003c21eb7bb0] [c00000000078e72c] ___sys_sendmsg+0x36c/0x380
[  979.767149] [c000003c21eb7d50] [c000000000793ad4] SyS_sendmsg+0xc4/0x130
[  979.771103] [c000003c21eb7dd0] [c00000000079485c] SyS_socketcall+0x3ac/0x430
[  979.773190] [c000003c21eb7e30] [c00000000000a17c] system_call+0x38/0xb4
[  979.775918] INFO: task postfix:5818 blocked for more than 120 seconds.
[  979.778010] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  979.782284] postfix         D 00003fff951a8c5c     0  5818      1 0x00040080
[  979.784386] Call Trace:
[  979.784493] [c000003c21ee3990] [c0000000000164f8] __switch_to+0x268/0x450
[  979.786609] [c000003c21ee39f0] [c0000000009425b4] __schedule+0x324/0xa40
[  979.791316] [c000003c21ee3ac0] [c0000000009442c4] schedule_preempt_disabled+0x34/0xb0
[  979.793631] [c000003c21ee3ae0] [c00000000094094c] __mutex_lock_slowpath+0x10c/0x2b0
[  979.804522] [c000003c21ee3b60] [c000000000940b54] mutex_lock+0x64/0x70
[  979.807693] [c000003c21ee3b90] [c00000000080121c] netlink_autobind.isra.26+0x29c/0x3c0
[  979.811820] [c000003c21ee3c60] [c000000000803fc4] netlink_bind+0x414/0x600
[  979.814028] [c000003c21ee3cf0] [c000000000792620] SyS_bind+0x1a0/0x1e0
[  979.816187] [c000003c21ee3dd0] [c00000000079464c] SyS_socketcall+0x19c/0x430
[  979.818284] [c000003c21ee3e30] [c00000000000a17c] system_call+0x38/0xb4
[  OK  ] Unmounted RPC Pipe File System.
[  OK  ] Started Restore /run/initramfs.
[ !!  ] Forcibly rebooting as result of failure.

dcrowell77, 2016-05-19:
This looks like a dupe of https://github.com/open-power/hostboot/issues/43

williamspatrick, 2016-05-19:
Agree, this is a dupe and I have about 99% confidence this is a BMC problem.  @dcrowell77 made the following comment in the other issue:

"One possibility came up in discussions today. The BMC controls a pin to the processor that tells it where to get its FSI clocks. If this pin toggles during the reset, the processor would lose the clocks and everything downstream would go wacky. @amiviswa may be able to provide an answer."

This seems like the most probable cause.

Is this something we should get the system owner to take over and resolve with AMI?  Or @spinler , is it something you can take?

spinler, 2016-05-19:
Sertac had volunteered awhile ago to drive this, which is why he owns the defect.  Until someone puts a scope on the system and proves the glitch, nothing will get resolved.

stewart-ibm, 2018-08-07:
Realistically, nothing is ever going to change to even possibly fix it with an AMI BMC and P8. It unlikely makes any sense to keep this issue open unless it's reproduced with an OpenBMC P8 platform or a P9 platform.

closed 2018-08-07 by stewart-ibm