open-power/op-build #483: VPD options in config file

MengLiMarkham, 2016-05-20:
We seem to find a VPD option bug in openpower/configs/hostboot/palmetto.config
Also in firestone.config, habanero.config and so on.
```
# VPD options.
set MVPD_READ_FROM_HW
set MVPD_WRITE_TO_HW
set MVPD_READ_FROM_PNOR
set MVPD_WRITE_FROM_PNOR
```
But in hostboot code, we use the CONFIG_MVPD_WRITE_**TO**_PNOR. So that means we did not set the MVPD_WRITE_TO_PNOR in our config file. Then we can not cache the MVPD to PNOR.

dcrowell77, 2016-05-20:
This is wrong, but it turns out we're saved by the default logic in src/usr/vpd/HBConfig.
```
config MVPD_WRITE_TO_PNOR
    default y if MVPD_READ_FROM_PNOR
    help
        Write Module VPD data to PNOR cache
```

To confirm in your environment, look at output/build/hostboot-*/obj/genfiles/config.h, you should see an entry like this.
`#define CONFIG_MVPD_WRITE_TO_PNOR 1`

We'll still get a fix in though.

williamspatrick, 2016-05-21:
Fixed.

closed 2016-05-21 by williamspatrick