open-power/op-build #493: Ensure MRW_ENHANCED_GROUPING_NO_MIRRORING attribute is correct for all systems.

williamspatrick, 2016-06-13:
Hostboot commit open-power/hostboot@784d1b2e made a change to the meaning of the attribute MRW_ENHANCED_GROUPING_NO_MIRRORING.  The statement from the memory team is as follows:

> Users moving to mss_eff_grouping.C v1.37 or newer must default the machine readable workbook attribute ATTR_MRW_ENHANCED_GROUPING_NO_MIRRORING to FALSE for all system types.  Note a TRUE setting will prohibit the use of memory mirroring mode and MCS grouping of 4. Failure to properly set the attribute ATTR_MRW_ENHANCED_GROUPING_NO_MIRRORING may result in either degraded memory performance or boot failures.

stewart-ibm, 2016-07-08:
Is this something we need for v1.10 ?

williamspatrick, 2016-07-20:
Just waiting on Firestone.  I suspect that one is wrong.

williamspatrick, 2016-07-22:
All set now!  Thanks.

closed 2016-07-22 by williamspatrick