open-power/op-build #500: repository per platform for MRW XML doesn't scale

stewart-ibm, 2016-06-21:
If we look into the future with 50-100 platforms, that many git repositories may be getting a bit hard to handle, and we then have a *lot* of packages in our buildroot overlay to maintain, all of which will be roughly identical.

One possibility would be to have a single MRW repository and merge everything into it.

This would also make the process of upstreaming a new platform a lot simpler.

williamspatrick, 2016-06-21:
The more important step in my mind is to reduce the duplication per machine. 

We have:
1. The xml package itself. 
2. Buildroot defconfig. 
3. Hostboot config. 

The XML package can be simplified with a common .mk file to reduce that duplication. 

The defconfig we have the work from Sam that is awaiting upstream acceptance. 

The Hostboot config we should have someone from that team take on. 

I don't honestly see separate repositories as being a deal breaker, but I do see these duplications as painful. 

We are planning to use the XML to feed into openbmc as well, but we will obviously need more data. I'm not sure what the appropriate repository breakdown would be with that in mind. 

There is also no reason why XML repositories need to be in the open-power project. I would be fine pointing to company owned / maintained repositories elsewhere.

stewart-ibm, 2016-06-21:
I don't disagree with any of the above. Perhaps a change to our buildroot overlay so that it's just PLATFORM_XML rather than (HABANERO|FIRESTONE|PALMETTO)_XML would de-duplicate a lot.

stewart-ibm, 2017-12-01:
Considering that we've gone to one set of mk files for the XML, I'll close this issue for now.

closed 2017-12-01 by stewart-ibm