open-power/op-build #507: op-build master-next broken on rhel6.4 pool systems , missing 'makeinfo'

jazurin, 2016-06-27:
Our op-build CI is failing on rhel6.4 nodes.    I believe it started happening after https://github.com/open-power/op-build/pull/504 was merged.  The reason the pull request CI didn't catch this is because I had turned off testing on rhel6.4 machines temporarily due to the github download bug affecting only our rhel6.4 nodes.

```
>>> host-binutils 2.26 Building
PATH="/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/bin:/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/sbin:/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/bin:/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/sbin:/opt/tools/bin:/afs/rchland.ibm.com/projects/esw/dvt/tools:/gsa/ausgsa/home/h/o/hostboot/private/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/bin:/usr/bin:/opt/ibm/cmvc/bin:/afs/austin.ibm.com/projects/esw/bin:/usr/ode/bin:/usr/lpp/cmvc/bin:/home/db2lin95/sqllib/bin:/home/db2lin95/sqllib/adm:/home/db2lin95/sqllib/misc:/usr/local/sbin:/usr/sbin:/sbin:/opt/ibm/cmvc/bin" PKG_CONFIG="/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_LIBDIR="/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib/pkgconfig"  /usr/bin/make -j25  -C /tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/build/host-binutils-2.26/
make[3]: Nothing to be done for `all-target'.
make[4]: Nothing to be done for `all'.
true "AR_FLAGS=rc" "CC_FOR_BUILD=/usr/bin/gcc" "CFLAGS=-O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include" "CXXFLAGS=-O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include" "CFLAGS_FOR_BUILD=-O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include" "CFLAGS_FOR_TARGET=-g -O2" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=-L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/lib -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -Wl,-rpath,/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib" "LIBCFLAGS=-O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include" "LIBCFLAGS_FOR_TARGET=-g -O2" "MAKE=/usr/bin/make" "MAKEINFO=/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/build/host-binutils-2.26/missing makeinfo --split-size=5000000 --split-size=5000000 " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/sh" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr" "infodir=/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/share/info" "libdir=/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib" "prefix=/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr" "tooldir=/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/powerpc64le-buildroot-linux-gnu" "AR=/usr/bin/ar" "AS=/usr/bin/as" "CC=/usr/bin/gcc" "CXX=/usr/bin/g++" "LD=/usr/bin/ld" "LIBCFLAGS=-O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include" "NM=/usr/bin/nm" "PICFLAG=" "RANLIB=/usr/bin/ranlib" "DESTDIR=" DO=all multi-do # /usr/bin/make
make[4]: Nothing to be done for `all'.
make[5]: Nothing to be done for `all'.
Making info in doc
Making info in po
make[5]: Nothing to be done for `info'.
make[5]: Nothing to be done for `info-am'.
/usr/bin/make  all-recursive
Making all in doc
make[6]: Nothing to be done for `all'.
Making all in po
make[6]: Nothing to be done for `all'.
/usr/bin/make  all-recursive
Making all in .
Making all in po
make[6]: Nothing to be done for `all'.
/usr/bin/make  all-recursive
Making all in doc
make[6]: Nothing to be done for `all'.
Making all in po
make[6]: Nothing to be done for `all'.
/usr/bin/make  all-recursive
Making info in doc
make[5]: Nothing to be done for `info'.
Making info in po
make[5]: Nothing to be done for `info'.
Making all in po
make[6]: Nothing to be done for `all'.
make[5]: Nothing to be done for `info-am'.
/usr/bin/make  all-recursive
Making all in doc
make[6]: Nothing to be done for `all'.
Making all in po
make[6]: Nothing to be done for `all'.
/usr/bin/make  all-recursive
Making all in po
make[6]: Nothing to be done for `all'.
restore=: && backupdir=".am$$" && \
	rm -rf $backupdir && mkdir $backupdir && \
	if (/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/build/host-binutils-2.26/missing makeinfo --split-size=5000000 --split-size=5000000 --version) >/dev/null 2>&1; then \
	  for f in ld.info ld.info-[0-9] ld.info-[0-9][0-9] ld.i[0-9] ld.i[0-9][0-9]; do \
	    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
	  done; \
	else :; fi && \
	if /tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/build/host-binutils-2.26/missing makeinfo --split-size=5000000 --split-size=5000000 -I . -I ./../bfd/doc -I ../bfd/doc -I ./../libiberty  -I . \
	 -o ld.info `test -f 'ld.texinfo' || echo './'`ld.texinfo; \
	then \
	  rc=0; \
	else \
	  rc=$?; \
	  $restore $backupdir/* `echo "./ld.info" | sed 's|[^/]*$||'`; \
	fi; \
	rm -rf $backupdir; exit $rc
/bin/sh ./libtool --tag=CC   --mode=link /usr/bin/gcc -g -O2 -O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include -no-undefined -rpath /nowhere -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/lib -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -Wl,-rpath,/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -o libldtestplug.la  libldtestplug_la-testplug.lo  -ldl -ldl 
/bin/sh ./libtool --tag=CC   --mode=link /usr/bin/gcc -g -O2 -O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include -no-undefined -rpath /nowhere -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/lib -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -Wl,-rpath,/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -o libldtestplug2.la  libldtestplug2_la-testplug2.lo  -ldl -ldl 
/bin/sh ./libtool --tag=CC   --mode=link /usr/bin/gcc -g -O2 -O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include -no-undefined -rpath /nowhere -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/lib -L/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -Wl,-rpath,/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/lib -o libldtestplug3.la  libldtestplug3_la-testplug3.lo  -ldl -ldl 
/usr/bin/gcc -DHAVE_CONFIG_H -I.  -I. -I. -I../bfd -I./../bfd -I./../include  -O2 -I/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/host/usr/include -DENABLE_PLUGINS -DLOCALEDIR="\"/tmp/hostboot/j [... cut ...]
libtool: link: ( cd ".libs" && rm -f "libldtestplug.la" && ln -s "../libldtestplug.la" "libldtestplug.la" )
libtool: link: ( cd ".libs" && rm -f "libldtestplug2.la" && ln -s "../libldtestplug2.la" "libldtestplug2.la" )
make[5]: *** [all-recursive] Error 1
make[4]: *** [all] Error 2
make[3]: *** [all-ld] Error 2
make[2]: *** [all] Error 2
make[1]: *** [/tmp/hostboot/jenkins-slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/op-build/output/build/host-binutils-2.26/.stamp_built] Error 2
make: *** [_all] Error 2
```

jazurin, 2016-06-27:
@shenki could you please take a look at this?
 thank you.

shenki, 2016-06-28:
@williamspatrick wrote:
>Anything needed to build op-build, beyond a very small set, should be
> self-built by Buildroot itself.  I don't see 'makeinfo' as one of
> these standard requirements.
> 
> From the console it appears like there is a problem with the binutils
> where it is calling 'makeinfo' and it doesn't need to.  We should
> either get that fixed or get HOST_MAKEINFO added as a build
> dependency.
> 
> It looks like there is code in binutils.mk that already tries to
> avoid building this:
> # Don't build documentation. It takes up extra space / build time,
> # and sometimes needs specific makeinfo versions to work
> BINUTILS_CONF_ENV += ac_cv_prog_MAKEINFO=missing
> HOST_BINUTILS_CONF_ENV += ac_cv_prog_MAKEINFO=missing
> 
> That would also explain this:
> /tmp/hostboot/jenkins-
> slave/workspace/op_build_gerrit_trigger_multi/NODE_OS/rhel6.4/build/o
> p-build/output/build/host-binutils-2.26/missing: line 81: makeinfo:
> command not found

sammj, 2016-06-30:
Stared at this a bit today, it sounds like this can happen (even if it's specified that we don't want to build them) if the timestamps on the _*.info_ and _*.texi_ / _*.texinfo_ files are out of sync in the binutils tarball.

Some discussion:
http://sourceware-org.1504.n7.nabble.com/PATCH-Docs-Prevent-build-failures-when-quot-makeinfo-quot-is-missing-td398598.html
https://sourceware.org/bugzilla/show_bug.cgi?id=17817
https://sourceware.org/bugzilla/show_bug.cgi?id=15345

I don't have a RHEL6 install handy at the moment to test this, but I suspect this is the notable difference between binutils 2.24 and 2.36.

Assuming that's the problem, we _could_ see if it's fixed in a newer version, otherwise:

- Setting MAKEINFO=true in HOST_BINUTILS_CONF_ENV appears to be a workaround (in binutils.mk), or
- Just adding host-makeinfo as a build dependency.

shenki, 2016-07-05:
The team that maintain the arc toolchain in upstream buildroot have come across this, and written a patch[1].

I suggest we apply that to the 2.26 release.

[1] https://git.buildroot.net/buildroot/tree/package/binutils/arc-2016.09-eng005/0800-Docs-Prevent-build-failures-when-makeinfo-is-missing.patch

jazurin, 2016-07-07:
Looks like the problem is fixed now. 
Thanks @shenki

PR #516 (2016-07-05): buildroot: fixes for wget and makeinfo build break
wget
 - security bump to version 1.18

binutils
 - Fix 2.26 build failures if makeinfo is missing
 - bump 2.26.x series to 2.26.1

This fixes #507.

---
This change is [[image]](https://reviewable.io/reviews/open-power/op-build/516)
closed 2016-07-07 by jazurin