open-power/op-build #526: kernel oops when mounting [corrupted?] file system.

williamspatrick, 2016-07-12:
We had a power outage on site and afterwards we seem to have corrupted a hard-drive.  The skiroot partition has a kernel oops when scanning the disk.

```
[    8.624025] EXT4-fs (dm-2): error loading journal
[    8.636469] ------------[ cut here ]------------
[    8.636472] WARNING: at fs/block_dev.c:57
[    8.636473] Modules linked in:
[    8.636476] CPU: 9 PID: 3007 Comm: pb-discover Not tainted 4.2.1-openpower2 #0
[    8.636478] task: c000003faed80000 ti: c000003faee00000 task.ti: c000003faee00000
[    8.636480] NIP: c000000000142190 LR: c00000000014217c CTR: c0000000000db9d4
[    8.636482] REGS: c000003faee03840 TRAP: 0700   Not tainted  (4.2.1-openpower2)
[    8.636483] MSR: 9000000000029033 <SF,HV,EE,ME,IR,DR,RI,LE>  CR: 24004828  XER: 20000000
[    8.636489] CFAR: c000000000b4ffb8 SOFTE: 1 
[    8.636489] GPR00: c00000000014217c c000003faee03ac0 c0000000016ffd00 fffffffffffffffb 
[    8.636489] GPR04: c000003ff3020c08 0000000000000007 000000000000000e c000003faf620f38 
[    8.636489] GPR08: c000003fff4aeee0 0000000000000001 c0000000016ace80 0000000000000020 
[    8.636489] GPR12: 0000000000000001 c00000000fe81b00 0000000000000000 0000000000000000 
[    8.636489] GPR16: 0000000000000000 0000000000000000 0000000000000000 0000000000000000 
[    8.636489] GPR20: 0000000000000000 c0000000016afb9c c000003ff3020bf0 0000000000000000 
[    8.636489] GPR24: 0000000000000081 0000000000000001 0000000000000000 c000003ff3020a98 
[    8.636489] GPR28: 0000000000000000 c000003fbb3e4800 c000003ff3020b70 c000003ff3020a80 
[    8.636508] NIP [c000000000142190] __blkdev_put+0x108/0x208
[    8.636510] LR [c00000000014217c] __blkdev_put+0xf4/0x208
[    8.636512] Call Trace:
[    8.636514] [c000003faee03ac0] [c00000000014217c] __blkdev_put+0xf4/0x208 (unreliable)
[    8.636517] [c000003faee03b40] [c00000000010e4f4] kill_block_super+0x74/0x80
[    8.636519] [c000003faee03b70] [c00000000010e83c] deactivate_locked_super+0x6c/0xc4
[    8.636521] [c000003faee03ba0] [c00000000010f4a8] mount_bdev+0x188/0x1dc
[    8.636524] [c000003faee03c30] [c0000000001903f0] ext4_mount+0x20/0x34
[    8.636526] [c000003faee03c50] [c000000000110104] mount_fs+0x2c/0xac
[    8.636528] [c000003faee03cd0] [c00000000012a82c] vfs_kern_mount+0x64/0x138
[    8.636531] [c000003faee03d20] [c00000000012e74c] do_mount+0xb90/0xca4
[    8.636533] [c000003faee03dd0] [c00000000012eac8] SyS_mount+0x90/0xc8
[    8.636535] [c000003faee03e30] [c000000000009198] system_call+0x38/0xd0
[    8.636536] Instruction dump:
[    8.636537] 7c0004ac 9aed0254 7c2004ac 935e0080 38800001 7fc3f378 4bff30f9 60000000 
[    8.636541] 2fa30000 419e001c 89350003 69290001 <0b090000> 2fa90000 41fe0008 9b350003 
[    8.636545] ---[ end trace c8e0699410b39661 ]---
```

```
/ # cat /etc/os-release 
NAME=Buildroot
VERSION=2015.05-g40ff8b4
ID=buildroot
VERSION_ID=2015.05
PRETTY_NAME="Buildroot 2015.05"
```

We seem to be running something based on top of v1.8 because the skiboot version is 5.1.6.

We are going to flash on a v1.9 release and see how it goes.

anoo1, 2016-07-12:
Habanero system updated to v1.9 and petitboot still unable to see the harddrive where ubuntu was running.

# cat /etc/os-release 
NAME=Buildroot
VERSION=2016.02-00014-gbc248fa
ID=buildroot
VERSION_ID=2016.02
PRETTY_NAME="Buildroot 2016.02"

williamspatrick, 2016-07-12:
With v1.9 we seem to get something better.

```
[ 1983.670191] device-mapper: snapshots: Snapshot is marked invalid.
[ 1983.670655] EXT4-fs (dm-2): unable to read superblock
[ 1983.670977] EXT4-fs (dm-2): unable to read superblock
```

williamspatrick, 2016-07-12:
Manually mounted the filesystem and then petitboot was able to rescan.  `fsck` found nothing once we got all the way up.  After reboot petitboot finds the drive just fine.

shenki, 2016-07-13:
The old firmware looks very ancient. Good to hear that the new skiroot was able to recover your fs; it means we're doing the correct thing with bringing in newer kernels with bugfixes.

Are you happy if we close this?

@anoo1 thanks for pointing out the `/etc/os-release` info. That's good to have; but it could contain better information. @sammj will take a look at populating it with the version info.

stewart-ibm, 2016-07-13:
The "Snapshot is marked invalid" is kind of interesting though... and that a mount fixed things.

williamspatrick, 2016-07-14:
I'm fine closing it.

closed 2016-07-14 by williamspatrick