open-power/op-build #538: mlx5_core driver doesn't survive kexec, IB cards not accessable

stewart-ibm, 2016-07-19:
Without suitable kernel patch, IB card doesn't work after kexec.

Solution is we either bring in a patch that has been discussed internally to IBM into our kernel tree, or we disable mlx5_core for v1.10

shenki, 2016-07-19:
I propose we add the patch that was sent to the mailing list:

 https://lists.ozlabs.org/pipermail/openpower-firmware/2016-July/000047.html

It's a backport of the upstream patch that adds kexec. Carol has tested it and it resolves the issue.

I have added the patch to my queue, but a release is blocked on getting the usb situation sorted for v4.4.15.

If we want to rush this through, I could release v4.4.14-openpower2 with the queued up fixes.

stewart-ibm, 2016-07-20:
Since this has gone into 4.4.15-openpower1 which is now sitting in master-next, closing this bug.

stewart-ibm, 2016-08-01:
So, it turns out it doesn't work, so as discussed on phone with @shenki and @sammj  - we're going to disable it for this release.

CC: @DEADBEEFh

sammj, 2016-08-01:
For reference, this is what I saw with the 4.4.15-openpower1 kernel on a machine with a Mellanox CX-4 card.

```
The system is going down NOW!t
Sent SIGTERM to all processes
Sent SIGKILL to all processes
cpu 0x0: Vector: 300 (Data Access) at [c000001fd73436a0]
    pc: d0000000047f6ff8: reclaim_pages+0x164/0x198 [mlx5_core]
    lr: d0000000047f7004: reclaim_pages+0x170/0x198 [mlx5_core]
    sp: c000001fd7343920
   msr: 9000000000009033
   dar: c000002000000000
 dsisr: 40000000
  current = 0xc000001fd728d900
  paca    = 0xc00000000fe80000   softe: 0        irq_happened: 0x01
    pid   = 2336, comm = kexec
enter ? for help
[c000001fd73439c0] d0000000047f79e8 mlx5_reclaim_startup_pages+0x8c/0x144 [mlx5_core]
[c000001fd7343a70] d0000000047f0bf8 mlx5_unload_one+0x1cc/0x234 [mlx5_core]
[c000001fd7343b10] d0000000047f0d38 mlx5_pci_err_detected+0xd8/0x180 [mlx5_core]
[c000001fd7343b80] c000000000209cac pci_device_shutdown+0x50/0xa0
[c000001fd7343bb0] c0000000003d8be0 device_shutdown+0x170/0x1e8
[c000001fd7343c50] c000000000073cac kernel_restart_prepare+0x4c/0x60
[c000001fd7343c70] c0000000000be318 kernel_kexec+0x6c/0xbc
[c000001fd7343ce0] c000000000074210 SyS_reboot+0x1b4/0x1f4
[c000001fd7343e30] c000000000009198 system_call+0x38/0xd0
--- Exception: c01 (System Call) at 00003fffa2cd08b4
SP (3ffffd95dd20) is in userspace
0:mon>
```

closed 2016-08-01 by stewart-ibm