open-power/op-build #581: Move to Linux 4.9

shenki, 2016-08-13:
Linux 4.9 will be the next LTS kernel

https://plus.google.com/+gregkroahhartman/posts/DjCWwSo7kqY

We should have a plan for moving to this. We might want to update to an intermediate release to shake out bugs that may have appeared since 4.4.

stewart-ibm, 2016-08-15:
Trusted boot is going to be the big thing needing a kernel bump. It seems that nearly everything needed made it into 4.8, and the rest is heading into 4.9.

So... going to 4.8 in the meantime is probably a good idea to iron things out, how much we backport for Trusted Boot will likely depend on what platforms that ships with.

stewart-ibm, 2016-11-10:
At the same time, enable BR2_LINUX_KERNEL_XZ in buildroot.

closed 2017-01-30 by stewart-ibm