open-power/op-build #594: Some Test scenarios causing OCC cpu freq states power management non-functional on OpenPower platforms.

pridhiviraj, 2016-08-23:
Below test scenarios causing occ cpu frequency states Power management non-functional on OpenPower platforms.

1. Observed after an IPMI PNOR re-provisioning test
       Issue an ipmi pnor reprovision request to erase certain pnor partitions.
       reboot the system after reprovision completes
       Check occ cpu freq states functionality
2. Boot the system from golden to primary and primary to golden pnor sides and observe the occ cpu frequency states functionality.
3. Flash only the pnor image and boot the system, observe the occ cpu freq states functionality.

On a failure case:
dmesg
[   24.908553] powernv-cpufreq: ibm,pstate-min node not found
[   24.908555] powernv-cpufreq: powernv-cpufreq disabled. System does not support PState control
[   24.908797] usbcore: registered new interface driver usbhid
[   24.908798] usbhid: USB HID core driver

cat /sys/firmware/opal/msglog | grep -i occ
[71112025061,7]   OCC Common Area at 0x1fff800000 size 8MB
[71112049161,7]   OCC Common Area at 0x1fff800000 size 8MB
[7773882678,5]   0x001fff800000..001fffffffff : ibm,occ-common-area@1fff800000
[10976128831,7] OCC: Chip 00 Data (0000001ffd9f8000) = 010200d597979700
[10976131984,7] OCC: Chip 08 Data (0000001ffddf8000) = 010200d5ef000000
[10976135103,5] OCC: All Chip Rdy after 0 ms
[11012001999,7] OCC: CPU pstate state device tree init
[11012003835,7] OCC: Data (      1ffd9f8000) =  10200d597979700                0
[11012007021,7] OCC: Min -43 Nom -105 Max -105 Nr States 195
[11012009421,3] OCC: OCC range is not valid
[11012060597,3] Skiping core cpufreq init due to OCC error

On a working case:
cat /sys/firmware/opal/msglog | grep -i occ
[35633957498,7]   OCC Common Area at 0x1fff800000 size 8MB
[35633982419,7]   OCC Common Area at 0x1fff800000 size 8MB
[9238926975,5]   0x001fff800000..001fffffffff : ibm,occ-common-area@1fff800000
[9833630735,7] OCC: Chip 00 Data (0000001ffd9f8000) = 010200d5ef000000
[9833633913,7] OCC: Chip 08 Data (0000001ffddf8000) = 010200d5ef000000
[9833637103,5] OCC: All Chip Rdy after 0 ms
[9833689750,7] OCC: CPU pstate state device tree init
[9833691624,7] OCC: Data (      1ffd9f8000) =  10200d5ef000000                0
[9833694938,7] OCC: Min -43 Nom -17 Max 0 Nr States 44
[9834220204,7] OCC: Chip 0 Core 1 PPMSR dcdcd50008000000
[9834736754,7] OCC: Chip 0 Core 2 PPMSR e5e5d50008000000
[9835253179,7] OCC: Chip 0 Core 3 PPMSR eeeed50008000000
[9835769791,7] OCC: Chip 0 Core 5 PPMSR efefd50008000000
[9836286207,7] OCC: Chip 0 Core 6 PPMSR efefd50008000000
[9836802904,7] OCC: Chip 0 Core 9 PPMSR efefd50008000000
[9837319390,7] OCC: Chip 0 Core a PPMSR efefd50008000000
[9837835996,7] OCC: Chip 0 Core c PPMSR efefd50008000000
[9838352360,7] OCC: Chip 0 Core d PPMSR efefd50008000000
[9838868779,7] OCC: Chip 0 Core e PPMSR efefd50008000000
[9839423086,7] OCC: Chip 8 Core 1 PPMSR dcdcd50008000000
[9839939409,7] OCC: Chip 8 Core 2 PPMSR e5e5d50008000000
[9840455719,7] OCC: Chip 8 Core 4 PPMSR eeeed50008000000
[9840971935,7] OCC: Chip 8 Core 5 PPMSR efefd50008000000
[9841488111,7] OCC: Chip 8 Core 6 PPMSR efefd50008000000
[9842004591,7] OCC: Chip 8 Core 9 PPMSR efefd50008000000
[9842521036,7] OCC: Chip 8 Core b PPMSR efefd50008000000
[9843037236,7] OCC: Chip 8 Core c PPMSR efefd50008000000
[9843553419,7] OCC: Chip 8 Core d PPMSR efefd50008000000
[9844069419,7] OCC: Chip 8 Core e PPMSR efefd50008000000

OPAL is getting in-correct number of pstates(195) from occ-opal shared memory interface data.

Firmware version: PNOR level
System Firmware (ID 47)
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-firestone-23bdc73
 Product Extra         : 	buildroot-81b8d98
 Product Extra         : 	skiboot-5.3.1
 Product Extra         : 	hostboot-d9ee1a1-be2f697
 Product Extra         : 	linux-4.4.18-openpower1-5906f7a
 Product Extra         : 	petitboot-v1.2.2-7fc085b
 Product Extra         : 	firestone-xml-783df1f
 Product Extra         : 	occ-bb092e9
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	capp-u

BMC FW level:
cat /proc/ractrends/Helper/FwInfo 
FW_VERSION=2.13.98117
FW_DATE=Jun 15 2016
FW_BUILDTIME=11:26:40 CDT
FW_DESC=8335 SRC BUILD RR9 06152016
FW_PRODUCTID=1
FW_RELEASEID=RR9
FW_CODEBASEVERSION=2.X
#

dcrowell77, 2016-08-23:
Going to need more data to get anywhere here:
- complete SEL list
- SOL output from boot
- dump of all sensor values (particularly the OCC active sensor)

Adding @cjcain

pridhiviraj, 2016-08-23:
IPMI PNOR Reprovisioning test result:
nvram --print-config
"common" Partition
---------------------
petitboot,bootdevs=network any 

nvram --update-config "test-date=test-value"
nvram --print-config
"common" Partition
---------------------
petitboot,bootdevs=network any 
test-date=test-value

ipmitool  raw 0x3A 0x1C; echo $?

[155969472798,5] IPMI: PNOR access requested
0

ipmitool  raw 0x3A 0x1D; echo $?
 03
0
 [175086404684,5] IPMI: PNOR access released
ipmitool  raw 0x3A 0x1D; echo $?
 00
0

After reboot console output:
[-1;-1f[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[  626.675588] reboot: Restarting system
[354257474447,5] OPAL: Reboot request...
[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .  3.45177|Ignoring boot flags, incorrect version 0x0
  3.58600|ISTEP  6. 3
  4.04189|ISTEP  6. 4
  4.04252|ISTEP  6. 5
 38.78120|HWAS|PRESENT> DIMM[03]=AAAAAAAAAAAAAAAA
 38.78121|HWAS|PRESENT> Membuf[04]=CCCC000000000000
 38.78121|HWAS|PRESENT> Proc[05]=C000000000000000
 38.86892|ISTEP  6. 6
 39.13836|ISTEP  6. 7
 43.00754|ISTEP  6. 8
 43.12093|ISTEP  6. 9
 46.05615|ISTEP  6.10
 46.08494|ISTEP  6.11
 47.87057|ISTEP  6.12
 47.87197|ISTEP  6.13
 47.87262|ISTEP  7. 1
 48.03176|ISTEP  7. 2
 48.16387|ISTEP  7. 3
 48.19346|ISTEP  7. 4
 48.23332|ISTEP  7. 5
 48.48714|ISTEP  7. 6
 48.53446|ISTEP  7. 7
 48.74082|ISTEP  7. 8
 48.89561|ISTEP  7. 9
 48.89612|ISTEP  8. 1
 49.13282|ISTEP  8. 2
 49.95623|ISTEP  8. 3
 49.96626|ISTEP  8. 4
 50.21267|ISTEP  8. 5
 50.21351|ISTEP  8. 6
 51.05977|ISTEP  8. 7
 51.06041|ISTEP  8. 8
 51.12749|ISTEP  9. 1
 51.52489|ISTEP  9. 2
 52.07862|ISTEP 10. 1
 52.29691|ISTEP 10. 2
 53.58958|ISTEP 10. 3
 53.59027|ISTEP 10. 4
 53.59084|ISTEP 10. 5
 53.59164|ISTEP 10. 6
 53.69196|ISTEP 10. 7
[... 804 lines cut ...]
[   30.916656] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[   30.919510] tg3 0004:01:00.0 enP4p1s0f0: renamed from eth0
[   30.961713]  sdb: sdb1 sdb2 sdb3
[   30.962490] sd 1:0:0:0: [sdb] Attached SCSI removable disk
[   30.969092]  sda:
[   30.969471] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   31.003028] usb-storage 1-3.2:1.0: USB Mass Storage device detected
[   31.003139] usb-storage 1-3.2:1.0: Quirks match for vid 046b pid ff40: 200
[   31.003197] scsi host5: usb-storage 1-3.2:1.0
[   31.088406] usb 1-3.3: new high-speed USB device number 5 using xhci_hcd
[   31.193031] usb-storage 1-3.3:1.0: USB Mass Storage device detected
[   31.193147] scsi host6: usb-storage 1-3.3:1.0
[   31.268400] usb 1-3.4: new low-speed USB device number 6 using xhci_hcd
[   31.421548] input: American Megatrends Inc. Virtual Keyboard and Mouse as /devices/pci0003:00/0003:00:00.0/0003:01:00.0/0003:02:09.0/0003:0d:00.0/usb1/1-3/1-3.4/1-3.4:1.0/0003:046B:FF10.0001/input/input0
[   31.478445] hid-generic 0003:046B:FF10.0001: input: USB HID v1.10 Keyboard [American Megatrends Inc. Virtual Keyboard and Mouse] on usb-0003:0d:00.0-3.4/input0
[   31.487215] input: American Megatrends Inc. Virtual Keyboard and Mouse as /devices/pci0003:00/0003:00:00.0/0003:01:00.0/0003:02:09.0/0003:0d:00.0/usb1/1-3/1-3.4/1-3.4:1.1/0003:046B:FF10.0002/input/input1
[   31.487292] hid-generic 0003:046B:FF10.0002: input: USB HID v1.10 Mouse [American Megatrends Inc. Virtual Keyboard and Mouse] on usb-0003:0d:00.0-3.4/input1
[   31.800300] scsi 4:0:0:0: CD-ROM            AMI      Virtual CDROM0   1.00 PQ: 0 ANSI: 0 CCS
[   31.801796] scsi 4:0:0:1: CD-ROM            AMI      Virtual CDROM1   1.00 PQ: 0 ANSI: 0 CCS
[   31.802915] scsi 4:0:0:2: CD-ROM            AMI      Virtual CDROM2   1.00 PQ: 0 ANSI: 0 CCS
[   31.804164] scsi 4:0:0:3: CD-ROM            AMI      Virtual CDROM3   1.00 PQ: 0 ANSI: 0 CCS
[   32.000299] scsi 5:0:0:0: Direct-Access     AMI      Virtual Floppy0  1.00 PQ: 0 ANSI: 0 CCS
[   32.001549] scsi 5:0:0:1: Direct-Access     AMI      Virtual Floppy1  1.00 PQ: 0 ANSI: 0 CCS
[   32.002673] scsi 5:0:0:2: Direct-Access     AMI      Virtual Floppy2  1.00 PQ: 0 ANSI: 0 CCS
 Petitboot (dev.20160614)                     8335-GTA         0000000000000000

pridhiviraj, 2016-08-23:
ipmitool  sel list
   1 | 01/01/2000 | 00:15:20 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
   2 | 01/01/2000 | 00:16:02 | System Firmware Progress #0x05 | Memory initialization | Asserted
   3 | 01/01/2000 | 00:17:04 | System Firmware Progress #0x05 | Secondary CPU Initialization | Asserted
   4 | 01/01/2000 | 00:17:05 | System Firmware Progress #0x05 | System boot initiated | Asserted
   5 | 01/01/2000 | 00:17:42 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
   6 | 01/01/2000 | 00:17:51 | System Firmware Progress #0x05 | PCI resource configuration | Asserted
   7 | 01/01/2000 | 00:17:54 | System Firmware Progress #0x05 | System boot initiated | Asserted
   8 | 01/01/2000 | 00:30:40 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
   9 | 01/01/2000 | 00:31:01 | System Firmware Progress #0x05 | Memory initialization | Asserted
   a | 01/01/2000 | 00:31:18 | System Firmware Progress #0x05 | Secondary CPU Initialization | Asserted
   b | 01/01/2000 | 00:31:19 | System Firmware Progress #0x05 | System boot initiated | Asserted
   c | 01/01/2000 | 00:31:48 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
   d | 01/01/2000 | 00:31:57 | System Firmware Progress #0x05 | PCI resource configuration | Asserted
   e | 01/01/2000 | 00:31:59 | System Firmware Progress #0x05 | System boot initiated | Asserted
   f | 01/01/2000 | 00:43:24 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
  10 | 01/01/2000 | 00:44:05 | System Firmware Progress #0x05 | Memory initialization | Asserted
  11 | 01/01/2000 | 00:45:08 | System Firmware Progress #0x05 | Secondary CPU Initialization | Asserted
  12 | 01/01/2000 | 00:45:09 | System Firmware Progress #0x05 | System boot initiated | Asserted
  13 | 01/01/2000 | 00:45:35 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
  14 | 01/01/2000 | 00:45:41 | System Firmware Progress #0x05 | PCI resource configuration | Asserted
  15 | 01/01/2000 | 00:45:48 | System Firmware Progress #0x05 | System boot initiated | Asserted
root@ltc-fire4:~# ipmitool sel time get
01/01/2000 00:51:08
ipmitool  sel list -v
Running Get PICMG Properties my_addr 0x20, transit 0, target 0
Error response 0xc1 from Get PICMG Properities
Running Get VSO Capabilities my_addr 0x20, transit 0, target 0
Invalid completion code received: Invalid command
Discovered IPMB address 0x0
SEL Record ID          : 0001
 Record Type           : 02
 Timestamp             : 01/01/2000 00:15:20
 Generator ID          : 0020
 EvM Revision          : 04
 Sensor Type           : System Firmware Progress
 Sensor Number         : 05
 Event Type            : Sensor-specific Discrete
 Event Direction       : Assertion Event
 Event Data            : c21400
 Description           : Motherboard initialization

SEL Record ID          : 0002
 Record Type           : 02
 Timestamp             : 01/01/2000 00:16:02
 Generator ID          : 0020
 EvM Revision          : 04
 Sensor Type           : System Firmware Progress
 Sensor Number         : 05
 Event Type            : Sensor-specific Discrete
 Event Direction       : Assertion Event
 Event Data            : c20100
 Description           : Memory initialization

SEL Record ID          : 0003
 Record Type           : 02
 Timestamp             : 01/01/2000 00:17:04
 Generator ID          : 0020
 EvM Revision          : 04
 Sensor Type           : System Firmware Progress
 Sensor Number         : 05
 Event Type            : Sensor-specific Discrete
 Event Direction       : Assertion Event
 Event Data            : c20300
 Description           : Secondary CPU Initialization

SEL Record ID          : 0004
 Record Type           : 02
 Timestamp             : 01/01/2000 00:17:05
 Generator ID          : 0020
[... 557 lines cut ...]
APSS Fault       | 0x00              | ok
CPU VDD Volt     | 0.93 Volts        | ok
CPU VDD Curr     | 12 Amps           | ok
PS Derating Fac  | 0x00              | ok
Proc1 Power      | 21 Watts          | ok
PCIE Proc1 Power | 47 Watts          | ok
GPU Sense        | 141 Watts         | ok
GPU Temp 1       | 30 degrees C      | ok
GPU Temp 2       | 34 degrees C      | ok
GPU Temp 3       | no reading        | ns
GPU Temp 4       | no reading        | ns
BMC Golden Side  | 0x00              | ok
GPU Func 1       | 0x00              | ok
GPU Func 2       | 0x00              | ok
GPU Func 3       | 0x00              | ok
GPU Func 4       | 0x00              | ok
PSU Fault 1      | 0x00              | ok
PSU Fault 2      | 0x00              | ok
Fan 1            | 3100 RPM          | ok
Fan 2            | 3000 RPM          | ok
Fan 3            | 3000 RPM          | ok
Fan 4            | 3100 RPM          | ok
NxtPwr Redundant | 0x00              | ok
CurPwr Redundant | 0x00              | ok
Turbo Allowed    | 0x00              | ok

pridhiviraj, 2016-08-23:
is there a way to attach logs here instead of pasting it.

dcrowell77, 2016-08-23:
No obvious signs of error in there so we'll probably need help from the Opal team to explain the error they spit out. @stewart-ibm 

One more chunk of data to grab would be the devtree
dtc -I fs -O dts /proc/device-tree

Re: attaching files - Look at the bottom of the comment window.  You can drag them in or select them.

shilpasri, 2016-08-24:
Hi,
[11012003835,7] OCC: Data ( 1ffd9f8000) = 10200d597979700 0
[11012007021,7] OCC: Min -43 Nom -105 Max -105 Nr States 195
[11012009421,3] OCC: OCC range is not valid

Valid | Version | Throttle | Min  | Nominal | Turbo | Ultra-turbo | Reserved
    1        02           00         d5          97         97            97                00
Opal is correctly parsing the min and max pstates as OCC has provided it. We read the pstate table only for Chip 0 as pstates and frequencies are common across all chips. Here the min pstate is less than turbo  pstate, ultra turbo pstate and nominal pstate which is the wrong data. We bail out here with "OCC range is not valid" because the no of pstates is greater than 128.

[10976128831,7] OCC: Chip 00 Data (0000001ffd9f8000) = 010200d597979700
[10976131984,7] OCC: Chip 08 Data (0000001ffddf8000) = 010200d5ef000000

If we see Chip 8 has sane values for min, nominal and turbo pstates.

dcrowell77, 2016-08-24:
@mzipse  - Can't find anyone from occ on github so please route appropriately.

shilpasri, 2016-08-25:
I was able to reproduce the bug on Pridhviraj's test machine. I am attaching the opal logs which dumps the entire pstate table. 
[opal_out.txt](https://github.com/open-power/op-build/files/436733/opal_out.txt)
From the logs I could verify that Opal has not corrupted the memory, as I have dumped the first 8 bytes of the pstate table right in the beginning of the OPAL init path.
[59096655891,5] SkiBoot skiboot-5.3.1-shilpa-dirty-e75ee6d starting...
[59096662393,5] initial console log level: memory 7, driver 5
[59096665185,5] OCC: Data (      1ffd9f8000) =  10200d597979700                0
[59096669211,5] OCC: Data (      1ffddf8000) =  10200d5ef000000                0

closed 2016-08-25 by williamspatrick