open-power/op-build #601: continuous ipmi workload, causes hostboot to fail and booted from golden side pnor

pridhiviraj, 2016-08-26:
Install a latest firestone_update.pnor on a firestone system which is built from op-build master-next branch. And try to boot the system when in parallel executing a ipmi commands. While system is booting hostboot fails and started booting from golden side pnor.

for i in {1..100}; do ipmitool -I lanplus -H 9.40.193.20 -U ADMIN -P admin raw 0x3a 0x0b 0x56 0x45 0x52 0x53 0x49 0x4f 0x4e 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 | xxd -r -p; done

This is a read only pnor VERSION partition command.
and also execute two more for loops with commands "power status" and "sensor list".

Console Output:
ousb systemd.unit=kdump-tools.service elfcorehdr=156480K
[    9.152302] kdump-tools[6629]:  * loaded kdump kernel

Ubuntu 16.04.1 LTS ltc-fire4 hvc0

ltc-fire4 login: [469943710066,5] IPMI: PNOR access requested
[469975490207,5] IPMI: PNOR access released
[471478598831,5] IPMI: PNOR access requested
[474038828202,5] IPMI: PNOR access released
[474073283415,5] IPMI: PNOR access requested
[474127627871,5] IPMI: PNOR access released
[474169688441,5] IPMI: PNOR access requested
[474200215166,5] IPMI: PNOR access released
[475590305911,5] IPMI: PNOR access requested
[478143320973,5] IPMI: PNOR access released
[478168342478,5] IPMI: PNOR access requested
[478198955633,5] IPMI: PNOR access released
[478275713823,5] IPMI: PNOR access requested
[478285093391,5] IPMI: PNOR access released
[479650089306,5] IPMI: PNOR access requested
[479687916042,5] IPMI: PNOR access released
[481565000372,5] IPMI: PNOR access requested
[484132845656,5] IPMI: PNOR access released
[484161023336,5] IPMI: PNOR access requested
[484186390351,5] IPMI: PNOR access released
[484208466927,5] IPMI: PNOR access requested
[486786719403,5] IPMI: PNOR access released
[488224956049,5] IPMI: PNOR access requested
[490804298507,5] IPMI: PNOR access released
[490832505388,5] IPMI: PNOR access requested
[490849846230,5] IPMI: PNOR access released
[490883315202,5] IPMI: PNOR access requested
[493451126383,5] IPMI: PNOR access released
[494916418063,5] IPMI: PNOR access requested
[497490866847,5] IPMI: PNOR access released
[497526325687,5] IPMI: PNOR access requested
[500094243439,5] IPMI: PNOR access released
[500121420463,5] IPMI: PNOR access requested
[500130194863,5] IPMI: PNOR access released
[501486757247,5] IPMI: PNOR access requested
[504059855816,5] IPMI: PNOR access released
[504123211386,5] IPMI: PNOR access requested
[506696242498,5] IPMI: PNOR access released
[506725799367,5] IPMI: PNOR access requested
[506737358027,5] IPMI: PNOR access released
[508347736262,5] IPMI: PNOR access requested
[508369828729,5] IPMI: PNOR access released
[509965059239,5] IPMI: PNOR access requested
[509997197935,5] IPMI: PNOR access released
[510567130271,5] IPMI: Soft shutdown requested

[-1;-1f[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[511555270636,5] IPMI: PNOR access requested
[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[  962.713437] reboot: Power down
[-1;-1fUbuntu 16.04[-1[512326578869,5] OPAL: Shutdown request type 0x0...
;-1f.  .  .  .[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[513364680882,3] BT: seq 0x99 netfn 0x00 cmd 0x02: Retry sending message
[-1;-1fUbuntu 16.04[-1;-1f.  .  .  .[514128815742,5] IPMI: PNOR access released
  3.58649|Ignoring boot flags, incorrect version 0x0
 16.98259|ISTEP  6. 3
 17.43731|ISTEP  6. 4
 17.43775|ISTEP  6. 5
 29.31895|HWAS|PRESENT> DIMM[03]=AAAAAAAAAAAAAAAA
 29.31896|HWAS|PRESENT> Membuf[04]=CCCC000000000000
[... 611 lines cut ...]
[541552627770,5] OPAL table: 0x300a0840 .. 0x300a0cb0, branch table: 0x30002000
[541552632950,5] FDT: Parsing fdt @0xff00000
[541558291179,5] XSCOM: chip 0x0 at 0x3fc0000000000 [P8 DD2.0]
[541558300589,5] XSCOM: chip 0x8 at 0x3fc4000000000 [P8 DD2.0]
[541558308050,6] XSTOP: XSCOM addr = 0x2010c82, FIR bit = 31
[541558311214,6] MFSI 0:0: Initialized
[541558313223,6] MFSI 0:2: Initialized
[541558315263,6] MFSI 0:1: Initialized
[541558317646,6] MFSI 8:0: Initialized
[541558319573,6] MFSI 8:2: Initialized
[541558321513,6] MFSI 8:1: Initialized
[541558654902,5] LPC: Bus on chip 0 PCB_Addr=0xb0020
[541558668244,5] LPC: Default bus on chip 0
[541558821066,6] MEM: parsing reserved memory from node /ibm,hostboot/reserved-memory
[541558831762,7] HOMER: Init chip 0
[541558834361,7]   PBA BAR0 : 0x0000001ffd800000
[541558837001,7]   PBA MASK0: 0x0000000000300000
[541558839533,7]   HOMER Image at 0x1ffd800000 size 4MB

We can observe hostboot reached ISTEP 21.1. and fails to boot, and switch back to golden side.

# ipmitool -H <BMC IP> -I lanplus -U ADMIN -P admin sensor list | grep -i golden
BIOS Golden Side | 0x0        | discrete   | 0x0180| na        | na        | na        | na        | na        | na        
BMC Golden Side  | 0x0        | discrete   | 0x0080| na        | na        | na        | na        | na        | na        
#

dcrowell77, 2016-08-26:
I believe you are hitting a known limitation in the BMC implementation of the PNOR read command.  Even though the command is just reading the PNOR, there is contention with Hostboot's use of the device.  There is supposed to be a handshake that Hostboot will reject to prevent any BMC access to the PNOR while we are booting.

stewart-ibm, 2016-08-29:
It's quite possible that the BMC gets it wrong (I've seen this happen, although it's rather hard to reproduce). Although the IPMISENSOR messages from HB are kind of interesting (are they timing out? That would fit with BMC doing things wrong).

Basically, the BMC has everything running inside this giant IPMIMain process, which is fairly easy to get it to eat 100% of CPU on the BMC, so it timing out of some IPMI messages is not unlikely. It's possible the HB code needs to be a bit more resilient to a badly behaved BMC.

dcrowell77, 2016-08-29:
There isn't any way to handle the PNOR getting whacked while we're using it.  From previous experience it has required a power cycle to recover the LPC logic if there is a collision.

closed 2016-08-29 by williamspatrick