open-power/op-build #607: Continuous HMI TFMR Error injection(Recoverable errors) causes the system to boot.

pridhiviraj, 2016-08-31:
Install a latest pnor image which is built from op-build master-next branch onto a firestone system, and do a continuous HMI TFMR Error injection. which results in system reboot. I am doing injection only Recoverable Error(TFMR_PARITY_ERROR).

Command used to inject the error is:
for i in {1..200};do ./putscom -c 00000000 15013281 0001080000000000;echo $?; done

./getscom -l
Chip ID  | Rev   | Chip type
---------|-------|--------
80000085 | DD2.0 | Centaur memory buffer
80000084 | DD2.0 | Centaur memory buffer
80000081 | DD2.0 | Centaur memory buffer
80000080 | DD2.0 | Centaur memory buffer
80000005 | DD2.0 | Centaur memory buffer
80000004 | DD2.0 | Centaur memory buffer
80000001 | DD2.0 | Centaur memory buffer
80000000 | DD2.0 | Centaur memory buffer
00000008 | DD2.0 | P8 (Venice) processor
00000000 | DD2.0 | P8 (Venice) processor

Disabled cpu idle states:
for i in /sys/devices/system/cpu/cpu*/cpuidle/state1/disable; do echo 1 > $i; done
for i in /sys/devices/system/cpu/cpu*/cpuidle/state2/disable; do echo 1 > $i; done

cat /sys/firmware/opal/msglog | grep -i core
[35393747145,6] CPU: P8 generation processor(max 8 threads/core)
[2913337288,7] PHB#0000: Core revision 0xa30005
[2977135688,7] PHB#0001: Core revision 0xa30005
[3040820248,7] PHB#0020: Core revision 0xa30005
[3104494031,7] PHB#0021: Core revision 0xa30005
[3168090008,7] PHB#0022: Core revision 0xa30005
[9469631251,7] OCC: Chip 0 Core 1 PPMSR dcdcd50008000000
[9470148477,7] OCC: Chip 0 Core 2 PPMSR e5e5d50008000000
[9470665566,7] OCC: Chip 0 Core 3 PPMSR eeeed50008000000
[9471183389,7] OCC: Chip 0 Core 5 PPMSR efefd50008000000
[9471700337,7] OCC: Chip 0 Core 6 PPMSR efefd50008000000
[9472219095,7] OCC: Chip 0 Core 9 PPMSR efefd50008000000
[9472736120,7] OCC: Chip 0 Core a PPMSR efefd50008000000
[9473254061,7] OCC: Chip 0 Core c PPMSR efefd50008000000
[9473771101,7] OCC: Chip 0 Core d PPMSR efefd50008000000
[9474288129,7] OCC: Chip 0 Core e PPMSR efefd50008000000
[9474929092,7] OCC: Chip 8 Core 1 PPMSR dcdcd50008000000
[9475445423,7] OCC: Chip 8 Core 2 PPMSR e5e5d50008000000
[9475961847,7] OCC: Chip 8 Core 4 PPMSR eeeed50008000000
[9476478046,7] OCC: Chip 8 Core 5 PPMSR efefd50008000000
[9476994095,7] OCC: Chip 8 Core 6 PPMSR efefd50008000000
[9477509471,7] OCC: Chip 8 Core 9 PPMSR efefd50008000000
[9478024759,7] OCC: Chip 8 Core b PPMSR efefd50008000000
[9478539945,7] OCC: Chip 8 Core c PPMSR efefd50008000000
[9479055115,7] OCC: Chip 8 Core d PPMSR efefd50008000000
[9479570316,7] OCC: Chip 8 Core e PPMSR efefd50008000000

console messages during failure.

for i in {1..200};do ./putscom -c 00000000 15013281 0001080000000000;echo $?; done
0000080000000000
[12057.133834] Severe Hypervisor Maintenance interrupt [Recovere0d]

[12057.134086]  Error detail: Timer facility experienced an error
[12057.134160] 	HMER: 0c40000000000000
[12057.134202] 	TFMR: b612000980a14008
0000080000000000[12057.135803] S
evere Hypervisor Maintenance interrupt [Recovered]
[12057.135915]  Error detail: Timer facility experienced an error
[12057.135981] 	HMER: 0c40000000000000
[12057.136021] 	TFMR: b612000980a04008
0
0000080000000000
[12057.137924] Severe Hypervisor Maintenance interrupt [Recovered]
[12057.138011]  Error detail: Timer facility experienced an error
[12057.138076] 	HMER: 0c40000000000000
[... 1468 lines cut ...]
 Generator ID          : 0020
 EvM Revision          : 04
 Sensor Type           : System Firmware Progress
 Sensor Number         : 05
 Event Type            : Sensor-specific Discrete
 Event Direction       : Assertion Event
 Event Data            : c21300
 Description           : System boot initiated

FW Info:
ipmitool fru print 47
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-firestone-3c3e4a5
 Product Extra         : 	buildroot-81b8d98
 Product Extra         : 	skiboot-5.3.2
 Product Extra         : 	hostboot-7658eab-1d63f96
 Product Extra         : 	linux-4.4.18-openpower1-a3a126e
 Product Extra         : 	petitboot-v1.2.3-2d20552
 Product Extra         : 	firestone-xml-783df1f
 Product Extra         : 	occ-bb092e9
 Product Extra         : 	hostboot-binaries-7f593a3
 Product Extra         : 	capp-u

uname -a
Linux ltc-fire4 4.4.0-34-generic #53-Ubuntu SMP Wed Jul 27 16:04:07 UTC 2016 ppc64le ppc64le ppc64le GNU/Linux

williamspatrick, 2016-08-31:
```
17.56682|Error reported by unknown (0xE500)
17.56682| 
17.56683| ModuleId 0x0b unknown
17.56683| ReasonCode 0xe540 unknown
17.56683| UserData1 unknown : 0x0005000000000101
17.56683| UserData2 unknown : 0x4241000500000000
```

This indicates that we hit a checkstop due to PBCENTFIR[5] "Combined-Response Address Error".  That would be either skiboot or Linux accessing a bad address.

stewart-ibm, 2016-09-01:
Patrick Williams <notifications@github.com> writes:
> ```
> 17.56682|Error reported by unknown (0xE500)
> 17.56682|
> 17.56683| ModuleId 0x0b unknown
> 17.56683| ReasonCode 0xe540 unknown
> 17.56683| UserData1 unknown : 0x0005000000000101
> 17.56683| UserData2 [...]

I wish Hostboot spat out those words rather than 5 occurances of the
word 'unknown'.

-- 
Stewart Smith
OPAL Architect, IBM.

stewart-ibm, 2018-08-07:
So much has changed since this original bug report, I don't *think* it applies anymore. If it does, and we have an `op-test` test that reproduces, let's reopen and investigate.

closed 2018-08-07 by stewart-ibm