open-power/op-build #619: ipmitool usb interface(BT) execution causes fru and sdrr data erase

pridhiviraj, 2016-09-02:
Ran ipmitool usb interface(BT) parallel on 3 to 4 terminals at a time, it causes the fru and sdr data erase.

for i in {1..100};do ipmitool -I usb sensor list; done
for i in {1..100};do ipmitool -I usb fru print; done
 Even BMC cold reset did not get back the vpd and sensor data.

After BMC Cold reset:
ipmitool -I lanplus -H BMC-IP -U ADMIN -P admin fru print
FRU Device Description : Builtin FRU Device (ID 0)
 Device not present (Requested sensor, data, or record not found)

SDRR successfully erased
Err in cmd get sensor sdr info
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
 ipmitool -I lanplus -H BMC-IP -U ADMIN -P admin sensor list
SDRR successfully erased
Err in cmd get sensor sdr info
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found

These data(FRU and SDR) got after a firmware upgrade.

And one more thing observed here is USB interface is working fine only when ipmitool commands executed in one terminal. If we try from multiple terminals all the commands are failing.

williamspatrick, 2016-09-02:
Why do you think this is a host firmware bug and not a BMC bug?

pridhiviraj, 2016-09-03:
Not sure which one exactly causes this, due to skiboot involved in BT interface commands execution. If it is BMC bug, i will raise it in internal bugzilla.

Thanks

stewart-ibm, 2018-08-07:
This is probably a BMC bug that'll never be fixed. If we can reproduce with one of the `op-test` tests, let's go and do that and maybe do something horrific in restricting what the host can do in parallel on such BMC implementations.

Hopefully no P9 platforms are affected....

closed 2018-08-07 by stewart-ibm

hexicans, 2019-02-11:
Hello,

Same problem ..

`
root@debian9:~# ipmitool fru print
FRU Device Description : Builtin FRU Device (ID 0)
 Chassis Type          : Unspecified
 Board Mfg Date        : Fri Jul 23 17:00:00 2010
 Board Mfg             : Dell
 Board Product         : PowerEdge
 Board Serial          : CN0WT5R37170303P0304
 Board Part Number     : 31S99MB0060
 Product Manufacturer  : Dell
 Product Name          : PowerEdge C1100
 Product Version       : A00
 Product Serial        : 9RLMT4J
 Product Asset Tag     : 1234567890
 Product Extra         : 12345678901234567890123

SDRR successfully erased
Unable to obtain SDR reservation
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
root@debian9:~# 5. Show the sensor output
-bash: 5.: command not found
root@debian9:~# pmitool sdr list
-bash: pmitool: command not found
root@debian9:~# pmitool sdr list ^C
root@debian9:~# ipmitool sdr list
SDRR successfully erased
Unable to obtain SDR reservation
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
root@debian9:~# ipmitool bmc reset cold
Sent cold reset command to MC
root@debian9:~# ipmitool sdr list
SDRR successfully erased
Unable to obtain SDR reservation
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
Get SDR 0000 command failed: Requested sensor, data, or record not found
root@debian9:~# `

hexicans, 2019-02-11:
root@debian9:~# ipmitool sdr fill sensors
SDRR successfully erased
Unable to obtain SDR reservation
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xff Unspecified error
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xc3 Timeout
Get Device ID command failed: 0xc3 Timeout
^C
SIGN INT: Close Interface Linux OpenIPMI Interface
root@debian9:~#