open-power/op-build #644: In latest CI hostboot is failed to boot(lot of i2c and fsi timeout errors.

pridhiviraj, 2016-09-15:
Installed a OpenPower habanero system with latest pnor image which is built from op-build master-next branch at this commit id.
Opbuild_CID=b94cea17075e45b3a4648fb34d8860a3ae32bf8e.
 After PNOR flash, system is failed to boot with below errors.
[SOL Session operational.  Use ~? for help]
  2.30519|ERRL|Dumping errors reported prior to registration
  2.34235|================================================
  2.34235|Error reported by i2c (0x0700)
  2.36094|  Timed out waiting for command complete.
  2.36094|  ModuleId   0x05 I2C_WAIT_FOR_CMD_COMP
  2.36094|  ReasonCode 0x0705 I2C_CMD_COMP_TIMEOUT
  2.37027|  UserData1  Status Register Value : 0x0002000000000000
  2.37027|  UserData2  Master Target : 0x0000000000000001
  2.37027|User Data Section 0, type UD
  2.37027|  Subsection type 0x06
  2.37027|  ComponentId errl (0x0100)
  2.37028|  CALLOUT
  2.37028|  HW CALLOUT
  2.37028|  Reporting CPU ID: 13
  2.37028|  Called out entity:
  2.37029|User Data Section 1, type UD
  2.37029|  Subsection type 0x06
  2.37029|  ComponentId errl (0x0100)
  2.37029|  CALLOUT
  2.37030|  PROCEDURE ERROR
  2.37030|  Procedure: 85
  2.37030|User Data Section 2, type UD
  2.37030|  Subsection type 0x04
  2.37031|  ComponentId errl (0x0100)
  2.37031|User Data Section 3, type UD
  2.37031|  Subsection type 0x03
  2.37031|  ComponentId errl (0x0100)
  2.37032|User Data Section 4, type UD
  2.37032|  Subsection type 0x01
  2.37032|  ComponentId errl (0x0100)
  2.37032|  STRING
  2.37033|  Hostboot Build ID: hostboot-70b5e31-opdirty-b69e321/hbicore.bin/genie
  2.37033|User Data Section 5, type UD
  2.37034|  Subsection type 0x0c
  2.37034|  ComponentId hb-trace (0x3100)
  2.37034|================================================
  3.39698|================================================
  3.39698|Error reported by i2c (0x0700)
  3.39699|  Timed out waiting for command complete.
  3.39699|  ModuleId   0x05 I2C_WAIT_FOR_CMD_COMP
  3.39699|  ReasonCode 0x0705 I2C_CMD_COMP_TIMEOUT
  3.39699|  UserData1  Status Register Value : 0x0002000000000000
  3.39700|  UserData2  Master Target : 0x0000000000000001
  3.39700|User Data Section 0, type UD
  3.39700|  Subsection type 0x06
  3.39700|  ComponentId errl (0x0100)
  3.39701|  CALLOUT
  3.39701|  HW CALLOUT
  3.39701|  Reporting CPU ID: 9
  3.39701|  Called out entity:
  3.39701|User Data Section 1, type UD
  3.39702|  Subsection type 0x06
  3.39702|  ComponentId errl (0x0100)
  3.39702|  CALLOUT
  3.39702|  PROCEDURE ERROR
  3.39703|  Procedure: 85
  3.39703|User Data Section 2, type UD
  3.39703|  Subsection type 0x04
  3.39703|  ComponentId errl (0x0100)
  3.39703|User Data Section 3, type UD
  3.39704|  Subsection type 0x03
  3.39704|  ComponentId errl (0x0100)
  3.39704|User Data Section 4, type UD
  3.39705|  Subsection type 0x01
  3.39705|  ComponentId errl (0x0100)
  3.39705|  STRING
[... 4993 lines cut ...]
181.15343|  presenceDetect> FSI and CVPD do not agree
181.15343|  ModuleId   0x21 FSI::MOD_FSIPRES_MEMBPRESENCEDETECT
181.15343|  ReasonCode 0x0a10 FSI::RC_FSI_CVPD_MISMATCH
181.15344|  UserData1  HUID of membuffer : 0x0000000000040001
181.15344|  UserData2  FSI Presence : 0x0000000100000000
181.15344|User Data Section 0, type UD
181.15344|  Subsection type 0x04
181.15345|  ComponentId errl (0x0100)
181.15345|User Data Section 1, type UD
181.15345|  Subsection type 0x06
181.15345|  ComponentId errl (0x0100)
181.15345|  CALLOUT
181.15346|  HW CALLOUT
181.15346|  Reporting CPU ID: 11
181.15346|  Called out entity:
181.15346|User Data Section 2, type UD
181.15347|  Subsection type 0x02
181.15347|  ComponentId errl (0x0100)
181.15347|  TARGET
181.15347|  HUID: 00040001
181.15349|  Phys path: Physical:/Sys0/Node0/Membuf1
181.15349|  Affinity path: Logical:/Sys0/Node0/Proc0/MCS1/Membuf1
181.15349|User Data Section 3, type UD
181.15350|  Subsection type 0x01
181.15350|  ComponentId

pridhiviraj, 2016-09-15:
SEL Entries  

d1 | 09/15/2016 | 05:02:24 | System Firmware Progress #0x05 | Motherboard initialization | Asserted
  d3 | 09/15/2016 | 05:02:39 | Processor #0x3e | Presence detected | Asserted
  d4 | 09/15/2016 | 05:02:39 | Processor #0x40 | Presence detected | Asserted
  d5 | 09/15/2016 | 05:02:39 | Processor #0x42 | Presence detected | Asserted
  d6 | 09/15/2016 | 05:02:39 | Processor #0x43 | Presence detected | Asserted
  d7 | 09/15/2016 | 05:02:39 | Processor #0x44 | Presence detected | Asserted
  d8 | 09/15/2016 | 05:02:40 | Processor #0x46 | Presence detected | Asserted
  d9 | 09/15/2016 | 05:02:40 | Processor #0x47 | Presence detected | Asserted
  da | 09/15/2016 | 05:02:40 | Processor #0x48 | Presence detected | Asserted
  db | 09/15/2016 | 05:02:40 | Processor #0x4e | Presence detected | Asserted
  ea | 09/15/2016 | 05:03:16 | Processor #0x4e | Transition to Non-recoverable
  ec | 09/15/2016 | 05:03:17 | Processor #0x4e | Transition to Non-recoverable
  ee | 09/15/2016 | 05:03:19 | Processor #0x4e | Transition to Non-recoverable
  f0 | 09/15/2016 | 05:03:22 | Processor #0x4e | Transition to Non-recoverable
  f2 | 09/15/2016 | 05:03:24 | Processor #0x4e | Transition to Non-recoverable
  f4 | 09/15/2016 | 05:03:26 | Memory #0x4b | Transition to Non-recoverable
  f6 | 09/15/2016 | 05:03:28 | Processor #0x4e | Transition to Non-recoverable
  f8 | 09/15/2016 | 05:03:31 | Memory #0x4b | Transition to Non-recoverable
  fa | 09/15/2016 | 05:03:33 | Processor #0x4e | Transition to Non-recoverable
  fc | 09/15/2016 | 05:03:37 | Memory #0x4b | Transition to Non-recoverable
  fe | 09/15/2016 | 05:03:38 | Processor #0x4e | Transition to Non-recoverable
 100 | 09/15/2016 | 05:03:41 | Memory #0x4b | Transition to Non-recoverable
 102 | 09/15/2016 | 05:03:43 | Processor #0x4e | Transition to Non-recoverable
 104 | 09/15/2016 | 05:03:46 | Memory #0x4b | Transition to Non-recoverable
 106 | 09/15/2016 | 05:03:48 | Processor #0x4e | Transition to Non-recoverable
 108 | 09/15/2016 | 05:03:52 | Memory #0x4b | Transition to Non-recoverable
 10a | 09/15/2016 | 05:03:53 | Processor #0x4e | Transition to Non-recoverable
 10c | 09/15/2016 | 05:03:57 | Memory #0x4b | Transition to Non-recoverable
 10e | 09/15/2016 | 05:03:59 | Processor #0x4e | Transition to Non-recoverable
 110 | 09/15/2016 | 05:04:03 | Memory #0x4b | Transition to Non-recoverable
 112 | 09/15/2016 | 05:04:04 | Processor #0x4e | Transition to Non-recoverable
 114 | 09/15/2016 | 05:04:07 | Memory #0x4a | Transition to Non-recoverable
 116 | 09/15/2016 | 05:04:09 | Processor #0x4e | Transition to Non-recoverable
 118 | 09/15/2016 | 05:04:13 | Memory #0x4a | Transition to Non-recoverable
 11a | 09/15/2016 | 05:04:15 | Processor #0x4e | Transition to Non-recoverable
 11c | 09/15/2016 | 05:04:19 | Memory #0x4a | Transition to Non-recoverable
 11e | 09/15/2016 | 05:04:20 | Processor #0x4e | Transition to Non-recoverable
 120 | 09/15/2016 | 05:04:24 | Memory #0x4a | Transition to Non-recoverable
 122 | 09/15/2016 | 05:04:26 | Processor #0x4e | Transition to Non-recoverable
 124 | 09/15/2016 | 05:04:29 | Memory #0x4a | Transition to Non-recoverable
 126 | 09/15/2016 | 05:04:32 | Processor #0x4e | Transition to Non-recoverable
 127 | 09/15/2016 | 05:04:34 | Watchdog 2 | Hard reset | Asserted
 128 | 09/15/2016 | 05:04:35 | Watchdog 2 | Hard reset | Asserted
 13c | 09/15/2016 | 05:06:56 | OEM record df | 702cda57040020ffff6faaaaaa
 13d | 09/15/2016 | 05:07:00 | OEM record df | 742cda57040020ffff6faaaaaa
 13e | 09/15/2016 | 05:07:01 | OEM record df | 752cda57040020ffff6faaaaaa
 13f | 09/15/2016 | 05:07:05 | OEM record df | 792cda57040020ffff6faaaaaa
 140 | 09/15/2016 | 05:07:06 | OEM record df | 7a2cda57040020ffff6faaaaaa
 141 | 09/15/2016 | 05:07:10 | OEM record df | 7e2cda57040020ffff6faaaaaa
 142 | 09/15/2016 | 05:07:12 | OEM record df | 802cda57040020ffff6faaaaaa
 143 | 09/15/2016 | 05:07:16 | OEM record df | 842cda57040020ffff6faaaaaa
 144 | 09/15/2016 | 05:07:17 | OEM record df | 852cda57040020ffff6faaaaaa
 145 | 09/15/2016 | 05:07:21 | OEM record df | 892cda57040020ffff6faaaaaa
 146 | 09/15/2016 | 05:07:23 | OEM record df | 8b2cda57040020ffff6faaaaaa
 147 | 09/15/2016 | 05:07:26 | OEM record df | 8e2cda57040020ffff6faaaaaa
 148 | 09/15/2016 | 05:07:28 | OEM record df | 902cda57040020ffff6faaaaaa
 149 | 09/15/2016 | 05:07:30 | OEM record df | 922cda57040020074e07aaaaaa
 14a | 09/15/2016 | 05:07:30 | Processor #0x4e | Transition to Non-recoverable
 14b | 09/15/2016 | 05:07:31 | OEM record df | 932cda57040020074e07aaaaaa
 14c | 09/15/2016 | 05:07:31 | Processor #0x4e | Transition to Non-recoverable
 14d | 09/15/2016 | 05:07:33 | OEM record df | 952cda57040020074e07aaaaaa
 14e | 09/15/2016 | 05:07:34 | Processor #0x4e | Transition to Non-recoverable
 14f | 09/15/2016 | 05:07:36 | OEM record df | 982cda57040020074e07aaaaaa
 150 | 09/15/2016 | 05:07:36 | Processor #0x4e | Transition to Non-recoverable
 151 | 09/15/2016 | 05:07:37 | OEM record df | 992cda57040020074e07aaaaaa
 152 | 09/15/2016 | 05:07:38 | Processor #0x4e | Transition to Non-recoverable
 153 | 09/15/2016 | 05:07:40 | OEM record df | 9c2cda570400200c4b07aaaaaa
 154 | 09/15/2016 | 05:07:40 | Memory #0x4b | Transition to Non-recoverable
[... 60 lines cut ...]
 191 | 09/15/2016 | 05:08:31 | Memory #0x34 | Memory Device Disabled | Asserted
 192 | 09/15/2016 | 05:08:31 | Memory #0x35 | Memory Device Disabled | Asserted
 193 | 09/15/2016 | 05:08:32 | Memory #0x4b | Transition to Non-recoverable
[... cut ...]
 1a0 | 09/15/2016 | 05:08:44 | Processor #0x4e | Transition to Non-recoverable
 1a1 | 09/15/2016 | 05:08:47 | OEM record df | df2cda570400200c4b07aaaaaa
 1a2 | 09/15/2016 | 05:08:47 | Memory #0x4b | Transition to Non-recoverable
 1a3 | 09/15/2016 | 05:08:49 | OEM record df | e12cda57040020074e07aaaaaa
 1a4 | 09/15/2016 | 05:08:49 | Processor #0x4e | Transition to Non-recoverable
 1a5 | 09/15/2016 | 05:10:37 | Watchdog 2 | Power cycle | Asserted
 1a6 | 09/15/2016 | 05:10:40 | Watchdog 2 | Power cycle | Asserted

ipmitool sel time get
09/15/2016 05:16:53

pridhiviraj, 2016-09-15:
As per @stewart-ibm suggestion, dumped the GUARD partition for checking any hardware guards.
Looks like there are no hardware is garded.
Dump:
/tmp/pflash -P GUARD -r gard.dump
Reading to "gard.dump" from 0x0002c000..0x00005000 !
[==================================================] 100%

Check the gard list:
./gard -e -f ../../../gard.dump -p list
No GARD entries to display

dcrowell77, 2016-09-15:
Is this a brand-new system?  Those i2c errors imply some kind of issue accessing the planar vpd.  I have a vague recollection of some early (unreleased) boards having i2c issues.  I'd suggest doing a complete AC pull to make sure all of the logic is reset and try again.  If that doesn't work it might need some hardware debug.

pridhiviraj, 2016-09-15:
yes, i requested lab admin to do AC power cycle. After that i will try boot the system and update here.
Thanks

pridhiviraj, 2016-09-15:
Hi @dcrowell77  
After AC power cycle also still system is not booting, and hostboot failed with above i2c errors.

dcrowell77, 2016-09-15:
What pass of board is this?

pridhiviraj, 2016-09-15:
i have no idea, let me check with my team. And let me know if you need any further data.

pridhiviraj, 2016-09-16:
AC power cycle didn't make system came up, but after did a flashing of full firmware HPM file brings the system up.

dcrowell77, 2016-09-16:
That makes me wonder if you might have had an incorrect image on there to begin with.  Please close the issue.

closed 2016-09-17 by williamspatrick