open-power/op-build #654: [v1.11] Firestone does not IPL under OpenBMC

madscientist159, 2016-09-22:
When using OpenBMC and v1.11 of op-build Firestone does not IPL.  There is no host serial console output at all, and the boot watchdog times out repeatedly.

Using the latest IBM binary host image (the version bundled for an AMI BMC) with OpenBMC works fine.

dcrowell77, 2016-09-22:
@causten I heard a rumor you were trying to get this setup going internally...

madscientist159, 2016-09-26:
@causten Is this something you are actively working on?  I was considering looking into this but don't want to duplicate effort.

Thanks!

williamspatrick, 2016-10-19:
@causten was able to narrow the commit down to the commit in Hostboot that introduced the SIO driver.

@shenki pointed me to the scratch register configuration code for openbmc.  https://github.com/openbmc/linux/blob/dev-4.7/arch/arm/mach-aspeed/aspeed.c#L62

	writel(0x00000042, AST_IO(AST_BASE_LPC | 0x170));
	writel(0x00008000, AST_IO(AST_BASE_LPC | 0x174));

This is in a common code path for all machines so there should not be anything Firestone specific there.

@dcrowell77 can you "parse" these scratch register values and confirm that they look sane?  I highly suspect we need some focus on this before it surfaces in Witherspoon.

dcrowell77, 2016-10-19:
Those addresses (0x170, 0x174) don't seem familiar to me.  How do they correspond with the scratch regs that start at 0x28?

dcrowell77, 2016-10-19:
Guessing... 
0x170->0x28 = Magic value that says the rest are valid, 0x42 is correct
0x174->0x29 = Boot flags
   0:1 = 00 = Normal boot
   2:3 = 00 = Do not clear gard or nvram
   4 = 0 = Do not use memboot
   5:7 = 000 = Normal log level
 
The regs are only 8 bits as we define them so I don't understand the 32-bit value in the call.

williamspatrick, 2016-10-19:
@dcrowell77 - Isn't the SUART vs VUART selection done via scratch registers?

shenki, 2016-10-19:
```cpp
    class Ast2400Uart : public Uart
    {
     public:
        enum consoleConfig_t
        {
            NONE            = 0x00,  // No output selected
            SELECT_SUART    = 0x40,  // SIO Uart
            SELECT_VUART    = 0x80,  // SOL virtual uart
            RESERVED        = 0xc0,  // Reserved
        };
```

and

https://github.com/open-power/hostboot/blob/master-p8/src/usr/console/ast2400.C#L311

So 0x80 tells hostboot to use the vuart.

williamspatrick, 2016-10-19:
So as best as I can tell, we believe we are programming the scratch registers correctly.

Need to get more debug on the hostboot side to understand why the SIO driver commit appears to break support.

dcrowell77, 2016-10-19:
I'm still confused by the address translation.  Where do you get Scratch reg 0x2D=0x80 out of the data above?  0x2D is where the UART selection comes from.

dcrowell77, 2016-10-19:
From the BMC spec:
SIO Scratch regs
0x28: Magic value 0x42. This is a "random" bit pattern used to identify
that the scratch regs have been populated by this version. Will change if a
different layout is used
0x29: Boot Flags:
[0:1] Boot mode:
00 : Normal
01 : Manufacturing
10 : istep mode
[2:3] Boot options
01 : Clear NVRAM
10 : Clear gard
[ 4 ] Memboot – replace PNOR with memory image (no PNOR
erase/write)
[5:7] Log level:
000 : Normal
001 : Verbose
0x2a: Istep control
0x2b: Istep major number:
0x2c: Istep minor number:
0x2d: Platform flags:
[0:1] IO ports at 3f8 config
00 : No uart
01 : SUART1
10 : VUART
11 : Reserved
[1:7] Reserved
0x2e: Reserved
0x2f: Reserved

madscientist159, 2016-10-19:
@williamspatrick One thing I noticed when working on the original support is that the BMC hardware does strange things when both LPC and AHB -based configuration of the serial hardware is attempted.

As it stands, there are essentially two kernels attempting to "own" the serial hardware on the AST2400 -- OpenBMC (as expected) and, for a short time, hostboot.  The patch in question may introduce a race condition due to explicitly reconfiguring hardware that is under the direct control of the Open BMC kernel (the VUART and the SUART) regardless of scratch mode register setting.

madscientist159, 2016-12-07:
@williamspatrick I haven't heard much activity on this issue.  Is there work still being done on it?

Thanks!

causten, 2016-12-07:
We still are.

madscientist159, 2016-12-07:
@causten Good to hear.  Last I heard there was an LPC stall occurring somewhere around the serial setup, is this still the case?

stewart-ibm, 2017-10-04:
Any further news here?

Is it worth keeping this bug open?

madscientist159, 2017-10-06:
Would love to get this fixed; AFAIK it's not fixed.