open-power/op-build #659: Linux patches fail to apply for linux-4.8-rc5.tar.xz

ploetzma, 2016-09-27:
Building witherspoon from a clean master next. Witherspoon defconfig has this linux release:

BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="https://cdn.kernel.org/pub/linux/kernel/v4.x/testing/linux-4.8-rc5.tar.xz"

The patches in openpower/linux/ from 0003 onward fail to apply.

>>> linux custom Extracting
xzcat /esw/san2/maploetz/p9github/op-build/dl/linux-4.8-rc5.tar.xz | tar --strip-components=1 -C /esw/san2/maploetz/p9github/op-build/output/build/linux-custom   -xf -
>>> linux custom Patching
for p in /esw/san2/maploetz/p9github/op-build/openpower/linux ; do if test -d $p ; then support/scripts/apply-patches.sh  /esw/san2/maploetz/p9github/op-build/output/build/linux-custom $p \*.patch || exit 1 ; else support/scripts/apply-p
atches.sh  /esw/san2/maploetz/p9github/op-build/output/build/linux-custom `dirname $p` `basename $p` || exit 1; fi done

Applying 0001-xhci-Use-xhci_pci_remove-for-xhci-device-shutdown.patch using patch:
patching file drivers/usb/host/xhci-pci.c
Hunk #1 succeeded at 462 (offset 15 lines).

Applying 0002-drivers-drm-ast-Switch-SCU-to-VGA-output-on-POST.patch using patch:
patching file drivers/gpu/drm/ast/ast_post.c

Applying 0003-net-mlx4_core-Set-UAR-page-size-to-4KB-regardless-of.patch using patch:
patching file drivers/infiniband/hw/mlx4/qp.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/infiniband/hw/mlx4/qp.c.rej
patching file drivers/net/ethernet/mellanox/mlx4/cq.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/net/ethernet/mellanox/mlx4/cq.c.rej
patching file drivers/net/ethernet/mellanox/mlx4/en_resources.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/net/ethernet/mellanox/mlx4/en_resources.c.rej
patching file drivers/net/ethernet/mellanox/mlx4/en_tx.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/net/ethernet/mellanox/mlx4/en_tx.c.rej
patching file drivers/net/ethernet/mellanox/mlx4/eq.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/net/ethernet/mellanox/mlx4/eq.c.rej
patching file drivers/net/ethernet/mellanox/mlx4/main.c
Reversed (or previously applied) patch detected!  Skipping patch.
5 out of 5 hunks ignored -- saving rejects to file drivers/net/ethernet/mellanox/mlx4/main.c.rej
patching file drivers/net/ethernet/mellanox/mlx4/pd.c
Reversed (or previously applied) patch detected!  Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/net/ethernet/mellanox/mlx4/pd.c.rej
patching file include/linux/mlx4/device.h
Reversed (or previously applied) patch detected!  Skipping patch.
3 out of 3 hunks ignored -- saving rejects to file include/linux/mlx4/device.h.rej
make[1]: *** [/esw/san2/maploetz/p9github/op-build/output/build/linux-custom/.stamp_patched] Error 1
make: *** [_all] Error 2
make: Leaving directory `/esw/san2/maploetz/p9github/op-build/buildroot'

stewart-ibm, 2016-09-27:
Likely also the case for POWER9 Mambo.

Solution is to not use the 4.4.XX patches and instead apply no patches to the kernel we're building. We may have to revisit this in the future, but for P9 platforms, having USB not survive kexec isn't really an issue yet.

williamspatrick, 2016-09-28:
I believe this is resolved by #661 .

closed 2016-09-28 by williamspatrick