open-power/op-build #672: Incorrect setting of Boot status sensors even actually boot is not happening.

pridhiviraj, 2016-10-03:
Install a P8 OpenPower Habanero hardware with habanero_update.pnor which is build from op-build master-next branch. Then executed the op-test-framework tests(opal tests). Then at some point in time, boot is not happening.

Then did the below steps.
1.manually powered off the system.
2. check the boot status sensors.
3. Power on the system.
4. Check the sol console for booting status and boot status sensors.

1. Power Off
ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin power status
Chassis Power is on
# ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin power off   
Chassis Power Control: Down/Off

2. system is in standby state(soft-off state)
# ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist
Host Status      | 04h | ok  | 35.0 | S5/G2: soft-off

3. Power on the system
ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin power status
Chassis Power is off
# ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin power on    
Chassis Power Control: Up/On
# ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin power status
Chassis Power is on

4. SOL console:
ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sol activate
[SOL Session operational.  Use ~? for help]

hostboot is not started isteps and then there is no skiboot started.

5. Check the sensors again
 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist all
Host Status      | 04h | ok  | 35.0 | S0/G0: working
OCC Active       | 08h | ok  | 210.0 | Device Disabled
All PGood        | 1Ch | ok  | 19.0 | Power off/down

We can observe here, system is actually not started booting, but Host Status sensor is set to "S0/G0: working" state.

As per my understanding Host status sensor should set as per below statements.
Host status--> Soft Off      --> Before Power ON.(i.e when system is in off state)
	       Legacy ON     --> When Host Boot boots,(i.e when Hostboot FW is able to communicate to BMC)
	       Working state --> When OPAL starts Linux kernel.

This issue from last two CI execution's observing.

Tests are exactly failing at Test IPMI System Power Control.6.

13:35:50 # Running test for tyan,habanero on habanero
13:35:51 # Starting Test Init.1
13:35:51 # 0
13:35:51 # Starting Test Skiroot Dropbear Safety Functionality.2
13:46:50 # 0
13:46:50 # Starting Test BMC Cold reset vs Host FW Functionality.3
13:58:42 # 0
13:58:42 # Starting Test Inband IPMI USB Interface Functionality.4
16:08:35 # 0
16:08:35 # Starting Test Out-of-band IPMI.5
16:29:50 # 0
16:29:50 # Starting Test IPMI System Power Control.6
16:44:20 # 16:44:20 :: ERROR: command failed with exit code 256
16:44:20 # 256
16:44:20 
16:44:20 #   Failed test 'Test IPMI System Power Control.6'
16:44:20 #   at ./run-op-it line 436.

Testcase result:
16:51:56 :: Running command: op-ci-bmc-run  "op_fwts_fvt.test_system_reboot()"
17:14:00 spawn ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sol activate
17:14:00 [SOL Session operational.  Use ~? for help]
17:14:00 /home/jenkins/workspace/FVTOPAL@3/op-test-framework/ci/source/op_ci_tools.cfg
17:14:00 /home/jenkins/workspace/FVTOPAL@3/op-test-framework/ci/source/op_ci_tools.cfg
17:14:00 Testing FWTS: Booting system to OS
17:14:00 Performing a IPMI Power OFF Operation
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin chassis power off
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'Host Status'
17:14:00 Host Status is S5/G2: soft-off, system reached standby
17:14:00 System is in standby/Soft-off state
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin chassis power on
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sol deactivate
17:14:00 Time: (0, '20161003_1653'), logfile: /home/jenkins/workspace/FVTOPAL@3/op-test-framework/host_sol_20161003_1653.log
17:14:00 /home/jenkins/workspace/FVTOPAL@3/op-test-framework/bvt/../common/sol_logger.exp /home/jenkins/workspace/FVTOPAL@3/op-test-framework/host_sol_20161003_1653.log ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin 
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'Host Status'
17:14:00 Host Status is S0/G0: working, IPL finished
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sol deactivate
17:14:00 System booted to working state
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 ipmitool -H bmc-ip -I lanplus -U ADMIN -P admin sdr elist |grep 'OS Boot'
17:14:00 IPL timeout
17:14:00 Traceback (most recent call last):
17:14:00   File "<string>", line 16, in <module>
17:14:00   File "/home/jenkins/workspace/FVTOPAL@3/op-test-framework/ci/source/op_fwts_fvt.py", line 88, in test_system_reboot
17:14:00     return opTestFWTS.test_system_reboot()
17:14:00   File "/home/jenkins/workspace/FVTOPAL@3/op-test-framework/testcases/OpTestFWTS.py", line 125, in test_system_reboot
17:14:00     self.cv_SYSTEM.sys_check_host_status()
17:14:00   File "/home/jenkins/workspace/FVTOPAL@3/op-test-framework/common/OpTestSystem.py", line 298, in sys_check_host_status
17:14:00     raise OpTestError(l_msg)
17:14:00 common.OpTestError.OpTestError: System failed to boot Host OS

So my understanding is this testcase failing due to incorrectly setting Host status sensor even system is not actually booting. After Host Status sensor is setting incorrectly test case looks for Os Boot sensor to set to "boot completed - device not specified". So testcase is failing.

PNOR Version:
FRU Device Description : System Firmware (ID 43)
 Product Name          : OpenPOWER Firmware
 Product Version       : open-power-habanero-53fbbf2
 Product Extra         : 	buildroot-211bd05
 Product Extra         : 	skiboot-5.3.6
 Product Extra         : 	hostboot-19ef62a
 Product Extra         : 	linux-4.4.22-openpower1-89b41af
 Product Extra         : 	petitboot-v1.3.0-dfa1106
 Product Extra         : 	habanero-xml-5565b8f-c658c41
 Product Extra         : 	occ-bb092e9
 Product Extra         : 	hostboot-binaries-61c5c1a
 Product Extra         : 	capp-uco

dcrowell77, 2016-10-03:
I'm a little confused by what is being reported.  Is it a or b?
a) System is not booting during some kind of IPMI test, also status sensor seems wrong
b) System is expected to not boot, but status sensor is not reflecting the proper value

I think in both cases these seem like BMC issues.

pridhiviraj, 2016-10-03:
yes, a is correct.  so here we have two issues, one is system is not booting at that point and 2nd one is "Host status" sensor set incorrectly.
But how come "Host status" sensor is set without booting. This sensor is set to the state "S0/G0: working" by opal when it really finishes it's booting process.

dcrowell77, 2016-10-03:
Hostboot never actually touches that sensor from what I can see so I have no guesses on why it is set incorrectly.

For the 'not booting' issue, going to probably need some hardware debug with Cronus + FSP Box to see where things are stuck.  What does the failing testcase actually do?  I know we've seen issues in the past where there were problems with IPMI traffic while we booted.

pridhiviraj, 2016-10-04:
Below are the testcases.
3:35:51 # Starting Test Init.1
13:35:51 # 0
13:35:51 # Starting Test Skiroot Dropbear Safety Functionality.2
13:46:50 # 0
13:46:50 # Starting Test BMC Cold reset vs Host FW Functionality.3
13:58:42 # 0
13:58:42 # Starting Test Inband IPMI USB Interface Functionality.4
16:08:35 # 0
16:08:35 # Starting Test Out-of-band IPMI.5
16:29:50 # 0
16:29:50 # Starting Test IPMI System Power Control.6
16:44:20 # 16:44:20 :: ERROR: command failed with exit code 256

And it is failing at test Test IPMI System Power Control.6.

Test Skiroot Dropbear Safety Functionality.2
it will check for dropbear is running or not in petitboot.

Test BMC Cold reset vs Host FW Functionality.3
This testcase when system is in runtime issue BMC cold reset and check host FW status. At the end reboot the system.

 Test Inband IPMI USB Interface Functionality.4
This test case will run in-band ipmi usb interface commands.

 Test Out-of-band IPMI.5
This testcase will run out-of-band ipmi commands.(only serial execution, command by command, there is no parallel execution).

Test IPMI System Power Control.6
IPMI Power control operations
Power off
Power ON-----------> At this point system is not booting
Power soft off
Power ON
Power Cycle
Power Reset

In today's run also test failed exactly at that point only.
And this behaviour seeing only on habanero not on firestone. On firestone these tests are executing fine

For more info about these tests, you need to refer https://github.com/open-power/op-test-framework

stewart-ibm, 2018-08-07:
I *think* we're all okay with this as either we updated the test cases to do something different, or we can't run them anymore due to AMI BMCs being not particularly solid. So, probably good to close this unless we find something with modern firmware and test suite.

closed 2018-08-07 by stewart-ibm